OpenCPN/OpenCPN #1481: Request: Replace Included Dashboard with Dash-T

rgleason, 2019-11-12:
[Dashboard-Tactics](https://github.com/canne/dashboard_tactics_pi) has been improved and extensively tested and offers greater accuracy, stability and better history graphs than the current Dashboard. It also includes the recently improved [Tactics_pi](https://github.com/tom-r/tactics_pi) instruments and thus has a larger set of instruments and offers 

    Tactics and performance functions
    Influx DB 2.0 ouput streaming and export
    Signal K data input streaming
    all docs - (HTML/PDF formats)

Each instrument has been carefully tested, improved and confirmed using TestQuality for Test Management and Codacy for code quality.

I would like to suggest that perhaps [Dash-T](https://github.com/canne/dashboard_tactics_pi) should replace Dashboard as the Included plugin.  

This effort was a highly collaborative one, Petri  [Canne] creating Dash-T and testing extensively and Thomas Rausch creating and maintaining Tactics_pi to extend the instruments for sailors.  

There are some important questions to ask  before doing this however:
1. Ask Petri and Thomas  ([Tactics_pi](https://github.com/tom-r/tactics_pi)) if they are willing to have [Dash-T](https://github.com/canne/dashboard_tactics_pi) become an "Included Plugin".
2. Ask them both if they are willing to continue maintaining their portions of the plugin.
3. Resolve how to document the instruments, since the Tactics manual is excellent and the Dashboard Instruments need some updating.
4. Resolve how the use of the repository works. (Perhaps Petri could allow Thomas as a collaborator.) or perhaps Thomas keeps maintaining Tactics and pulls are made to Dash-T.

But perhaps the first question to ask is Dave as head developer, willing to make this change?

transmitterdan, 2019-11-24:
Rick,

I have tried to build and integrate this new dashboard_tactics_pi but it seems futile.  It uses the text name dashboard which conflicts with the existing dashboard.  Also, it's JSON code seems to be old and buggy.  I cannot get it to run at all in debug mode because of ASSERTS and null pointer dereference errors.  If I run the WMM plugin and this dashboard_tactics_pi then O crashes.

Why does it use a different set of JSON source files than the original dashboard?  Also, it should not name itself dashboard but rather something unique like dashboard_t perhaps.

rgleason, 2019-11-25:
> "I have tried to build and integrate this new dashboard_tactics_pi but it seems futile. "
    --I have built it, but not recently. I have been using the Release versions where Petri has provided 9 different assets, for various OS.  I find these releases to be quite convenient and I have not had a problem with them. https://github.com/canne/dashboard_tactics_pi/releases

> "It uses the text name dashboard which conflicts with the existing dashboard."
  --  I think the plugin name is pretty clear  "Dashboard-Tactics". I guess I don't understand the conflict problem, what it is precisely?

> Also, it's JSON code seems to be old and buggy. I cannot get it to run at all in debug mode because of ASSERTS and null pointer dereference errors. 
  -- I don't believe it is old and buggy. Perhaps the build is missing something?

> If I run the WMM plugin and this dashboard_tactics_pi then O crashes.
  -- I have been running Dashboard-Tactics with WMM and had no problems.

> Why does it use a different set of JSON source files than the original dashboard? Also, it should not name itself dashboard but rather something unique like dashboard_t perhaps.
 -- It is named Dashboard-Tactics, not Dashboard. I think you need to discuss this with Petri further, particularly about the technical aspects. 

It is an enhanced version of Dashboard, that I will be using and I believe it should be used in place of Dashboard.

If you read the [documentation, here in the Readme file](https://github.com/canne/dashboard_tactics_pi) you will start to understand its purpose.

We need to get you able to build this properly, so I am going to ask Petri if he will respond here.

I assume you have cloned his repository and tried to build it normally. I will try that using windows in the next day or so.

transmitterdan, 2019-11-25:
I submitted a PR that I believe fixed the crashing problem.  I am looking at the code now.  There are a lot of little things that others users may not like.  Example: The preferences dialog box opens off the screen in many cases. This is because of the odd decision to open the dialog in relation to wherever the cursor was when right clicking to get preferences open. Why that was done isn't clear.  Regular dashboard always opens preferences well inside the main window.

The installer tries to add a shortcut to the Windows start menu.  I don't know why that is or how to stop it.  I don't see what good a shortcut is for a plugin.

When we include a plugin in the base build it needs to not do "strange" things like crash or open dialogs that are partially off the screen. This plugin needs a lot of testing.

rgleason, 2019-11-25:
Thanks Dan.  I guess you were able to build it successfully.  
Here is a codacy review score of "A" for Dashboard-Tactics for your information
https://app.codacy.com/manual/petri38-github/dashboard_tactics_pi/dashboard

Also Petri has done a whole series of testing of each and every instrument, and while there have recently been some small issues, the calculations and reliability of the instruments is much better.
I am going to try to find that link for you.  First look at the list of Closed Issues here
https://github.com/canne/dashboard_tactics_pi/issues?q=is%3Aissue+is%3Aclosed
The Testquality website is https://canne.testquality.com  but I believe you must have an account and use a new email to use it.  Petri has done an extensive amount of testing of all the instruments, very methodically. I am going to add some screenshots of the process since the testquality project is not available easily.

[image]
[image]
[image]

Incidentally, Petri advises that there is a switch inside the plugin, so that it can be built with only the Dashboard Instruments, leaving out Tactics, if that is the version preferred.

rgleason, 2019-12-21:
It has been extensively tested and is more accurate and reliable.
https://github.com/canne/dashboard_tactics_pi/blob/pr151/docs/developers/README.md

Or are you all in the NIH (Not invented here) mode?

transmitterdan, 2019-12-21:
Rick,

There is a lot of other work going on. It’s a bit of work to switch out old dashboard for this one. Just not enough resources right now.

rgleason, 2019-12-21:
OK, for the future.
In cmakelists.txt Line 26

# ON = performance instruments, OFF = exclude all tactics_pi originated code from compilation _and_ build
SET(TACTICS_INSTRUMENTS ON)

rgleason, 2019-12-21:
[Information for developers, contributors and collaborators
v1.5.11 and onwards](https://github.com/canne/dashboard_tactics_pi/blob/pr151/docs/developers/README.md)

Discusses which branches to post PR's to and explains the "long term maintenance" phase.

closed 2020-05-05 by rgleason