OpenCPN/OpenCPN #1482: Option menu -> Treat WPL as APRS reports

RooieDirk, 2019-11-14:
[image]
A WPL sentence is **always** treated as an APRS report never mind the setting of the checkbox.
Found under linux.

rgleason, 2021-04-14:
How can we get this closed?  Is a PR needed to remove Checkbox "Treat WPL...."? (and a change in the manual?

Hakansv, 2021-04-15:
Changed in current master.
[image]

rgleason, 2021-04-15:
Added to manual here https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:toolbar_buttons:ais#treat_wpl_sentences_as_aprs_position_reports

Can this be closed now?

closed 2021-04-17 by RooieDirk