OpenCPN/OpenCPN #1483: Option Menu -> Display Chart information objects

RooieDirk, 2019-11-14:
[image]
The checkbox in the Option menu for chart information objects is not working. However in the menu it is working fine
[image]

tai-fun, 2019-11-14:
The "Chart Information Objects" in options-charts works as it is supposed
to for me on Linux.
Were do you find "Chart Information Objects" in "menu"? The attributes
listed there is not what "Chart Information Objects" is all about.

rgleason, 2021-04-14:
Can't this be closed now?

rgleason, 2021-04-23:
Chart inform Objects works in Windows too.  @RooieDirk  can this be closed?

nohal, 2021-04-24:
Seems OK with current code

closed 2021-04-24 by nohal