OpenCPN/OpenCPN #1484: Bug: Grib_pi does not display Oscar current files

rgleason, 2019-11-16:
Grib_pi no longer displays OSCAR grib files.

1. The files load and the filename shows in the lower left.
2. The current vectors do not display.
3. The current at the cursor no longer shows.

These files used to work in v4.2 through v4.8.8

[image]

Most convenient way to obtain an OSCAR file is by sending email with details as indicated in this link.  http://www.svsarana.com/oscar/index.php
Find  "How to get the data"

These files used to work properly in grib_pi!

soggygeek, 2019-11-16:
Actually, the GribPi does load ONE OSCAR file, fine for me, in both 5.0.0 and 4.8.6.  But in both of those versions, if you try to load both a wind GRIB and an OSCAR GRIB, the OSCAR doesn't appear to load.
[image]

soggygeek, 2019-11-16:
More info:  The problem appears to be that OSCAR current data is historical data, and typically the wind data is a forecast covering today and into the future.  With my grib_pi settings set to "start at the nearest forecast to the current time" (this is set in Preferences in Settings/Plugins/Grib_Pi), when you load both a wind GRIB and the OSCAR current GRIB, the current file is there, but doesn't show because it's date is (typically) for yesterday or before.  Clicking on the drop-down showing the dates of the "forecast" and selecting the oldest one, the current info shows up.  The problem is informing wxroute that there is (older) current data, in addition to newer wind data, I think.

rgleason, 2019-11-16:
Exactly, I am just one step behind you and have also determined that.
OSCAR is historical
GFS is predictive.
Time period does not match.

rgleason, 2019-11-16:
Also this website, http://www.svsarana.com/oscar/index.php, which I found wonderful many many years ago, only seems to provide one single frame. It is not 5 days as it states in the description, and it is in the past.  

With normal weather grib files being predictive, we cannot use this data for weather routing.

rgleason, 2019-11-16:
OK, I am closing this now. Thanks. Sorry for the static.

closed 2019-11-16 by rgleason