OpenCPN/OpenCPN #1488: GetArch does not take into account arm compilation on Arch

gromain, 2019-11-20:
Hello,

I stumbled on a small bug while working to build OpenCPN for Manjaro ARM.
I realized that the GetArch function doesn't take into account the possibility of compiling for arm on a distro other than debian, red hat or suse.
The function I'm talking about is hosted here: https://github.com/OpenCPN/OpenCPN/blob/master/cmake/GetArch.cmake
This would ease building for raspberry's and the likes.

Sincerely,
Romain

gromain, 2019-11-20:
Additionally, maybe a way of providing this information directly to the command line would be nice. If I'm not mistaken, setting directly the correct flags (ARCH or PACKAGE_FORMAT for example) in the command line will not work correctly since those settings will be overriden by the call to this function.

leamas, 2019-11-20:
hm...  this actually looks like a can of worms. 

First question is why not just check for arm in all cases, before the different distro checking. After all, arm is arm and the cmake code to identify it is probably OK.

Secondly many of the variables here are just fpr the debian package setup. The support for RPM is dropped, but it still lingers here and there. Linux packaging is basically deb or none.

In short: this needs a general cleanup.

gromain, 2019-11-21:
Sorry I opened that can!

But yes, I agree with you. I did not follow all the recent changes, but this looks very redondant.

What about giving the user the opportunity to chose the packaging they want directly from the command line, instead of detecting the platform?

leamas, 2019-11-21:
There is actually not much too choose. If you are on Debian, you can build a .deb package. No other linux packaging is available besides flatpak, which is quite different.

And, there are already too many options. What is your usecase?

gromain, 2019-11-21:
I'm building directly on a Raspberry Pi 4 running an Arch Linux derivative. Arch uses its own package building system (basically using a description of the steps to build the package).
In short, it doesn't like automatic detection stuff, since I need to create a patch to be able to build without intervention nor manual changes to the checked out repo or downloaded archive.

leamas, 2019-11-21:
Well, whether you aim to package it for Arch (yes, I know it...) or to just use opencpn, it basically boils down the same thing: a working *make install*, right?

gromain, 2019-11-21:
Absolutely!

But in its current form, GetArch is preventing this, since I have to edit it to return the right architecture when compiling on something else than Debian!

leamas, 2019-11-21:
In that case we could leave all the packaging variables as they are, they are just not involved.

Perhaps you could try to make a patch where  the `if (CMAKE_SYSTEM_PROCESSOR MATCHES "arm*")`  test is done once and for all  before checking for the different releases. This test is basically generic and should work on any distro. 

Note that on linux CMAKE_SYSTEM_PROCESSOR is just the output of _uname -p_.

Things like this needs to be tested on more than one platform. If you make a patch which looks sound and works on your yet to be revelealed Arch derivative I could send it through our CI setups which builds on the platforms we care about.

leamas, 2019-11-21:
Or, you could try this. It's more or less untested. Does it work for you?

[getarch.patch.gz](https://github.com/OpenCPN/OpenCPN/files/3874177/getarch.patch.gz)

gromain, 2019-11-21:
This sounds like a sensible patch. I'll try that this afternoon and report back.

gromain, 2019-11-21:
Looks like it's working both on the Raspberry and on my computer from the latest commit on the master branch. This looks good on my side.

leamas, 2019-11-21:
OK, I will try to make a ci test. Stay tuned, it may take a while.

gromain, 2019-11-21:
No worries, there is no hurry!

PR #1801 (2020-10-03): build: Clean up cmake architecture detection (#1488).
Clean up current mess in cmake/GetArch, making *make install* work better one more platforms.

Closes: #1488
closed 2020-10-03 by bdbcat