OpenCPN/OpenCPN #1493: Invalid check for "OpenCPN requires access to serial ports to use serial NMEA data"

hreuver0183, 2019-11-24:
Hello,
The ARM-based computer I use has some USB-ports and a serial port for a remote terminal.
I use OpenCPN on this board with some NMEA input via USB and a USB key for oesENC.

OpenCPN checks for permissions in the order:
/dev/ttyS0
/dev/ttyUSB0
/dev/ttyACM0

Permissions:
$ ls -al /dev/tty{S,USB}0
crw--w---- 1 root tty     239, 0 Nov 24 11:11 /dev/ttyS0
crw-rw---- 1 root dialout 188, 0 Jan 28  2018 /dev/ttyUSB0

The OpenCPN-user is member of the dialout group.

By changing up the order I can prevent a popup-screen, but the correct way should be first checking which port you need and then checking for permissions.

If you don't have much time I hope you can at least change the order of the checks, since on most computers ttyS0 isn't present anymore and it is not likely to break anything.

With regards,
Huub

The quick and especially dirty fix is below:
--- orig/OpenCPN/src/chart1.cpp 2019-11-24 12:11:27.494414347 +0100
+++ OpenCPN/src/chart1.cpp      2019-11-24 11:27:52.343275722 +0100
@@ -11053,9 +11053,10 @@
     bret = false;
 
     wxArrayString result1;
-    wxExecute(_T("stat -c %G /dev/ttyS0"), result1);
+    wxExecute(_T("stat -c %G /dev/ttyUSB0"), result1);
+
     if(!result1.size())
-        wxExecute(_T("stat -c %G /dev/ttyUSB0"), result1);
+        wxExecute(_T("stat -c %G /dev/ttyS0"), result1);
 
     if(!result1.size())
         wxExecute(_T("stat -c %G /dev/ttyACM0"), result1);

leamas, 2019-11-25:
hm... The dirty fix is certainly doable. Still, if you edit our comment and add four spaces before each line in the patch it becomes sort of readable ;)

The main issue is IMHO something which should be handled downstream in the Debian packaging. I assume that you are using Raspbian? That users in the dialout group cannot access /dev/ttyS? strikes me as a strange configuration.

leamas, 2019-11-25:
Hm, again... Looking at a debian host (stretch) /dev/ttyS0 is group-owned by dialout, not tty.  This seems like the correct setup to me. 

What distro are you on?

hreuver0183, 2019-11-25:
Hi,
It is not a question of "what distribution", but "how did you configure your system". The system is Ubuntu 18.04, odroid N2 (arm64).

The ttyS0 is a connector on the board for a hardwired rs232, only use i can think of is logging the booting process if a normal system start fails for some reason. That is not something opencpn needs access to. Every normal connection goes through /dev/ttyUSBx first.

Therefore root.tty for /dev/ttyS0 without access for the opencpn user is absolutely correct.

With regards,
Huub

leamas, 2019-11-25:
Looks like Fedora has the same setup as  you: dialout for USB devices and tty for /dev/ttyS?

I might make a PR. If you are motivated, it might be better if you submit it. My plate is sort of full, so *if* I make it's definitely not in a near future.

That said, I agree that at least the dirty fix proposed by you should fix this problem without breaking too many eggs.

hreuver0183, 2019-11-26:
I will do sometime this week.
We will see if it gets to pass.

With regards,
Huub

Hakansv, 2019-12-19:
Just more info and comment.
On my arm, RPi, I also got this message and it's totally irrelevant since I use only Network connections. If we need the message at all we better check for available networks as well.

leamas, 2019-12-19:
Indeed. Still, this is tricky: on a fresh installation this is one of the most common problems, and we cannot assume any configuration whatsoever... perhaps better do just not assume anything and wait with the checks until user actually tries to configure a device?!

Hakansv, 2019-12-19:
Yes, very good. If that's possible I do agree and maybe complete:  "....until user actually tries to configure a SERIAL device?"!

leamas, 2019-12-19:
I don't what's possible, for who or when... but you are of course right.

closed 2020-03-15 by hreuver0183