OpenCPN/OpenCPN #1494: 5.0.0 build fail in Mageia Cauldron src/glChartCanvas.cpp:287:36: error: 'glXGetProcAddress' was not declared in this scope

barjac, 2019-11-25:
5.0.0 was building until (I'm guessing) update of one of it's dependancies.

/home/baz/rpmbuild/BUILD/OpenCPN-5.0.0/src/glChartCanvas.cpp:287:36: error: 'glXGetProcAddress' was not declared in this scope; did you mean 'ocpnGetProcAddress'?
  287 | #define systemGetProcAddress(ADDR) glXGetProcAddress((const GLubyte*)ADDR)
      |                                    ^~~~~~~~~~~~~~~~~
/home/baz/rpmbuild/BUILD/OpenCPN-5.0.0/src/glChartCanvas.cpp:287:36: note: in definition of macro 'systemGetProcAddress'
  287 | #define systemGetProcAddress(ADDR) glXGetProcAddress((const GLubyte*)ADDR)
      |                                    ^~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/opencpn.dir/build.make:1431: CMakeFiles/opencpn.dir/src/glChartCanvas.cpp.o] Error 1
make[2]: Leaving directory '/home/baz/rpmbuild/BUILD/OpenCPN-5.0.0/build'
make[2]: *** Waiting for unfinished jobs....`

barjac, 2020-02-01:
So why close without any comment?

bdbcat, 2020-02-01:
Closed due to inactivity, and lots of new commits in the meantime.
And also to get the attention of the OP.
Are you still seeing this issue?
Dave

barjac, 2020-02-02:
Hi Dave,
Thanks for your quick reply.
Yes, I have patched the sources to git master to test and still see the same issue.
Here is a link to the full build log which may be of use to you: 
http://mtf.duckdns.org/pub/linux/barjac/distrib/cauldron/x86_64/log/opencpn-5.0.0-3.1.mga8_bcj.src.rpm/build.0.20200202133056.log
Barry.

bdbcat, 2020-02-04:
Barry...
Funny, we build for Fedora regularly.
You say it was building sometime in the past?
If so, perhaps using the git bisect command will help isolate the point where it stopped building correctly.

bdbcat, 2020-02-04:
Barry..
Do you happen to have nVidia graphics card?

barjac, 2020-02-04:
This did build back in March 2019 in Mageia 7, but a lot has changed in Mageia 8 (aka Cauldron) development branch. http://mtf.duckdns.org/pub/linux/barjac/distrib/cauldron/x86_64/log/opencpn-5.0.0-1.mga7.src.rpm/build.0.20190327125828.log
There is no graphics active in the build chroot so I doubt that would be relevant, however no, I don't have nvidia on the host machine.
I will test the Mga7 build again using the release tarball to see if that still builds.
Bisecting at the moment would not help as neither release nor master will build ;)

barjac, 2020-02-05:
The original 5.0.0 from Mageia 7 still builds fine: http://mtf.duckdns.org/pub/linux/barjac/distrib/7/x86_64/log/opencpn-5.0.0-1.mga7_bcj.src.rpm/build.0.20200205001846.log
But it fails in Mageia 8 with the same error:  http://mtf.duckdns.org/pub/linux/barjac/distrib/cauldron/x86_64/log/opencpn-5.0.0-3.1.mga8_bcj.src.rpm/build.0.20200205003720.log
It looks like glXGetProcAddress is no longer available.

bdbcat, 2020-02-05:
Barry...
There is some confusion about this function definition in some GL libraries.
As a test, please try to edit the source of v5.0.0 glChartCanvas.cpp, and make the following change at line #319 or thereabouts.

currently is:
#define systemGetProcAddress(ADDR) glXGetProcAddress((const GLubyte*)ADDR)

should be:
#define systemGetProcAddress(ADDR) glXGetProcAddressARB((const GLubyte*)ADDR)

Good Luck
Dave

barjac, 2020-02-05:
/home/baz/rpmbuild/BUILD/OpenCPN-5.0.0/src/glChartCanvas.cpp:287:36: error: 'glXGetProcAddressARB' was not declared in this scope; did you mean 'ocpnGetProcAddress'?
  287 | #define systemGetProcAddress(ADDR) glXGetProcAddressARB((const GLubyte*)ADDR)
      |                                    ^~~~~~~~~~~~~~~~~~~~
/home/baz/rpmbuild/BUILD/OpenCPN-5.0.0/src/glChartCanvas.cpp:287:36: note: in definition of macro 'systemGetProcAddress'
  287 | #define systemGetProcAddress(ADDR) glXGetProcAddressARB((const GLubyte*)ADDR)
      |                                    ^~~~~~~~~~~~~~~~~~~~
Failed sadly.
I did find some references to glXGetProcAddress vs. glXGetProcAddressARB but they were 15 years old!
Here is a link to all packages (versioned) that are installed in the build system that fails, which you may find useful. http://mtf.duckdns.org/pub/linux/barjac/distrib/cauldron/x86_64/log/opencpn-5.0.0-3.2.mga8_bcj.src.rpm/rpm_qa.0.20200205122016.log

barjac, 2020-02-05:
Also tested with 5.0.0 patched to master - same result :(

bdbcat, 2020-02-06:
Barry...
You could add a line like this:
`extern void (*glXGetProcAddress(const GLubyte *procname))( void );
`
before the offending instruction.  This will probably allow the compile to succeed, but link may fail.
Anyway, more information, maybe...

barjac, 2020-02-06:
Yes link fails:
/usr/bin/ld: CMakeFiles/opencpn.dir/src/glChartCanvas.cpp.o: in function `ocpnGetProcAddress(char const*, char const*)':
/home/baz/rpmbuild/BUILD/OpenCPN-5.0.0/src/glChartCanvas.cpp:350: undefined reference to `glXGetProcAddress(unsigned char const*)'

http://mtf.duckdns.org/pub/linux/barjac/distrib/cauldron/x86_64/log/opencpn-5.0.0-3.3.mga8_bcj.src.rpm/build.0.20200206130431.log

barjac, 2020-02-17:
Any news, and can this be re-opened?

bdbcat, 2020-02-18:
No news.  Re-opened.
I suppose I will need to provision a virtual box image of this distro, and dig into it.

barjac, 2020-02-18:
This is not in a current release of the distro (Mga7), but in the development branch  (Mga8 a.k.a Cauldron). As I said in the original post, this must be related to newer versions of libs or tools than you are currently using. I did post a link above to a file containing all the versions of the installed packages in the build environment in the hope that you may be able to isolate the culprit.
You could install Mageia 7 in a VM and then switch media to Cauldron and update it. I can help if you need to do that. Or maybe I could make a VirtualBox image available?

david-geiger, 2020-02-19:
We have on Mageia Cauldron the same build issue for the plugin climatology_pi [https://github.com/seandepagnier/climatology_pi](url)

david-geiger, 2020-02-19:
So apparently added missing include "#include <GL/glx.h>" fixes build issue.

http://svnweb.mageia.org/packages?view=revision&revision=1544121

barjac, 2020-02-19:
Thanks David, that does also clear this particular issue in opencpn, however we then have a whole raft of new undefined. I have reverted to wxgtk3.0 from wkgtk3.1 with no improvement.
http://mtf.duckdns.org/pub/linux/barjac/distrib/cauldron/x86_64/log/opencpn-5.0.0-3.5.mga8_bcj.src.rpm/build.0.20200219155743.log

nohal, 2020-02-19:
Is WXGTK in Cauldron built with support for AUI and OpenGL enabled at all? Looking at http://svnweb.mageia.org/packages/cauldron/wxgtk/current/SPECS/wxgtk.spec?revision=1537786&view=markup#l165 I'm not really convinced and would much prefer to see `--enable-aui` and `--with-opengl`...

barjac, 2020-02-19:
Yes it is, but I have passed your comment to our wxgtk maintainer.
We have already patched opencpn for this issue in the past, and due to the long delay in finding the missing #include I had forgotten that it had been temporarily removed. Why it was not upstreamed in the past I don't know but here it is:

[opencpn-5.0.0-mga-link_wxgtk.patch.txt](https://github.com/OpenCPN/OpenCPN/files/4227481/opencpn-5.0.0-mga-link_wxgtk.patch.txt)
 We also build for aarch64 and need to add this patch:
[opencpn-5.0.0-aarch64-plugindir.patch.txt](https://github.com/OpenCPN/OpenCPN/files/4227486/opencpn-5.0.0-aarch64-plugindir.patch.txt)
And lastly the one this bug was about:
[opencpn-5.0.0-mga-missing_glx_include.patch.txt](https://github.com/OpenCPN/OpenCPN/files/4227491/opencpn-5.0.0-mga-missing_glx_include.patch.txt)
If the above could be merged upstream that would be great!
Closing this now as our issue is resolved.
Thanks for your help.

bdbcat, 2020-02-19:
Slow down, too many moving parts.
Let's start with the missing glx.h.  That seems safe.  We can do that now.

The other patches, not so sure.  Changes to CMakeLists.txt are notorious for having unintended consequences for other platforms.  So these need testing on all possible affected platforms.
Please submit PR's as desired, and we will get them into wider alpha test now.

Of course, none of these patches will appear as Releases before the next Beta test starts.  So if you need them now, patch locally as required.

Dave

barjac, 2020-02-19:
HI Dave,
Well looking back at our history I got that first patch from upstream, i.e. you :) 
I honestly don't remember, but it was when I updated the package to 5.0.0 on 27th March last year. The comments were:
- new version 5.0.0
- patch from upstream to fix wxgtk link issues

Ah found it, that patch was from nohal, here, in my bug report  #1329.
I assumed it would be merged.
The other patch about aarch64 lib location is minor and we could continue to patch it.

bdbcat, 2020-02-20:
barjac...
Breaking this down...
AFICT, the first patch to CMakeLists.txt is already in place on current master branch, I think has been for some time.
Second patch for aarch64 needs wider test.
third (glx.h) patch is OK to add now.

So, to summarize:
I think you need the third patch right now, and the second whenever it appears.
OK?

bdbcat, 2020-02-20:
Or maybe I am mistaken, and you are not working from current master?

barjac, 2020-02-20:
I am using the release tarball of 5.0.0 as this is for our next stable distro release. We will update this if you make a new release before we finally freeze versions.

as365n4, 2020-08-15:
I just tried to compile V5.2.0 and have the same problem, I am using Arch Linux Arm on aarch64/Kernel 5.7.8.
`[ 56%] Building CXX object CMakeFiles/opencpn.dir/src/glChartCanvas.cpp.o
/usr/bin/c++ -DHAVE_WEBVIEW -DNDEBUG -DWXUSINGDLL -D_FILE_OFFSET_BITS=64 -D__WXGTK3__ -D__WXGTK__ -DocpnUSE_GL -DocpnUSE_SVG -Dopencpn_EXPORTS -DwxUSE_WEBVIEW=1 -I/home/pi/opencpn-git/src/opencpn-git/build/include -I/home/pi/opencpn-git/src/opencpn-git/include -I/usr/include/libexif -I/home/pi/opencpn-git/src/opencpn-git/libs/ssl_sha1/include -I/usr/include/gtk-3.0 -I/usr/include/pango-1.0 -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -I/usr/include/harfbuzz -I/usr/include/freetype2 -I/usr/include/libpng16 -I/usr/include/libmount -I/usr/include/blkid -I/usr/include/fribidi -I/usr/include/cairo -I/usr/include/pixman-1 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/gio-unix-2.0 -I/usr/include/atk-1.0 -I/usr/include/at-spi2-atk/2.0 -I/usr/include/dbus-1.0 -I/usr/lib/dbus-1.0/include -I/usr/include/at-spi-2.0 -I/home/pi/opencpn-git/src/opencpn-git/libs/SQLiteCpp/include -I/home/pi/opencpn-git/src/opencpn-git/libs/wxcurl/include -I/home/pi/opencpn-git/src/opencpn-git/libs/wxsvg/include -I/home/pi/opencpn-git/src/opencpn-git/libs/serial/include -I/home/pi/opencpn-git/src/opencpn-git/libs/libtess2/Include -I/home/pi/opencpn-git/src/opencpn-git/libs/sound/include -I/home/pi/opencpn-git/src/opencpn-git/libs/wxJSON/include -I/home/pi/opencpn-git/src/opencpn-git/libs/nmea0183/src -I/home/pi/opencpn-git/src/opencpn-git/libs/iso8211/include -I/home/pi/opencpn-git/src/opencpn-git/libs/gdal/include -I/home/pi/opencpn-git/src/opencpn-git/libs/garmin/jeeps -I/home/pi/opencpn-git/src/opencpn-git/libs/texcmp/squish -I/home/pi/opencpn-git/src/opencpn-git/libs/mipmap/include -I/home/pi/opencpn-git/src/opencpn-git/libs/easywsclient -I/home/pi/opencpn-git/src/opencpn-git/libs/wxservdisc -isystem /usr/lib/wx/include/gtk3-unicode-3.1 -isystem /usr/include/wx-3.1 -D_FORTIFY_SOURCE=2 -march=armv8-a -O2 -pipe -fstack-protector-strong -fno-plt -pthread -O3 -DNDEBUG    -DPREFIX=\"/usr\" -Wall -Wno-unused -fexceptions -rdynamic -fno-strict-aliasing -Wno-deprecated-declarations -pthread -std=gnu++11 -o CMakeFiles/opencpn.dir/src/glChartCanvas.cpp.o -c /home/pi/opencpn-git/src/opencpn-git/src/glChartCanvas.cpp
/home/pi/opencpn-git/src/opencpn-git/src/glChartCanvas.cpp: In function ‘void (* ocpnGetProcAddress(const char*, const char*))()’:
/home/pi/opencpn-git/src/opencpn-git/src/glChartCanvas.cpp:318:36: error: ‘glXGetProcAddress’ was not declared in this scope; did you mean ‘ocpnGetProcAddress’?
  318 | #define systemGetProcAddress(ADDR) glXGetProcAddress((const GLubyte*)ADDR)
      |                                    ^~~~~~~~~~~~~~~~~
/home/pi/opencpn-git/src/opencpn-git/src/glChartCanvas.cpp:318:36: note: in definition of macro ‘systemGetProcAddress’
  318 | #define systemGetProcAddress(ADDR) glXGetProcAddress((const GLubyte*)ADDR)
      |                                    ^~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/opencpn.dir/build.make:1606: CMakeFiles/opencpn.dir/src/glChartCanvas.cpp.o] Error 1
make[2]: Leaving directory '/home/pi/opencpn-git/src/opencpn-git/build'
make[1]: *** [CMakeFiles/Makefile2:812: CMakeFiles/opencpn.dir/all] Error 2
make[1]: Leaving directory '/home/pi/opencpn-git/src/opencpn-git/build'
make: *** [Makefile:174: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...`

bdbcat, 2021-05-02:
Closing due to inactivity

closed 2021-05-02 by bdbcat