OpenCPN/OpenCPN #1496: About? Where is it?  Can't determine what version is running mingw or msvc

rgleason, 2019-11-26:
Where is the "About" Key?
I would like to determine exactly what version is running mingw or msvc.

leamas, 2019-11-27:
Yes, you would really expect that info here. 

Sooner or later there will be 64-bit builds for both windows and macos. They will then also need this info for the same reason: a 64-bit build can most likely only run 64-bits plugins.  So this is not just about mingw

leamas, 2019-11-27:
BTW: this could probably be added to #1460 without much effort (the info required is available in existing macros). But I'm short of time.

rgleason, 2020-01-05:
Thank you!

[image]

closed 2020-01-05 by rgleason