OpenCPN/OpenCPN #1497: Plugin API:   CMake-CI Files Version Number needed

rgleason, 2019-12-04:
I believe we should implement a CMake-CI Version system

"CMake-CI Version Number: 1.0.0"
Change Notes: [Additional notes detailing particular changes made differing from the Version]

To ease understanding what CI-CMake is being used by a given plugin.
The number will enable easier updates of the CI-CMake system.

Plugin API
Jon Gough advises that this may require an addition to the Plugin API.
Also I think the plugin should perhaps show the Cmake-CI version.

rgleason, 2019-12-04:
**CMake-CI System**

**Files**
- CMakeLists.txt
- appveyor.yml

**Folders**
- circleci
- ci
- cmake
   - in-files
-debian

rgleason, 2019-12-04:
Also, I believe the Version number should be tied somehow, maybe directly, to a PI Installer version number. That to may need to be included in the Plugin API

leamas, 2019-12-07:
This is basically solved by #1460. From the plugins perspective, there is just the plugin API  version level, so the "Plugin installer version" seems a little odd.

rgleason, 2019-12-07:
True, but it would be most helpful to know exactly by a version number or some identifier, what version of CMake and CI Files is being used in each plugin!   
It will help with updating the Plugins rather than guessing!
Maybe we just start a numbering system and put it in the CMakkeLists file.

rgleason, 2019-12-08:
BTW Plugin API Version so far, has been completely independent of the CMake and CI files being used. The API Version establishes the earliest Plugin API that can be used to compile a working plugin and is independent of CMake files and Continuous Integration files used.

leamas, 2019-12-08:
As for the plugins, its covered by the PR which supports both pre-release info (like _alpha_) and build  info (for example _741.deadbee_, CI build number + git commit). The PR is agnostic as to what's goes into these fields.

Is this what you are looking for?

rgleason, 2019-12-14:
Alec, the problem is identifying what CMake and CI files are being used, as they get changed. Sometimes for an older plugin it is an ancient version, which is readily identified, however as they get more current it becomes more difficult.

So for Sean's plugins I think I am just going to put a version number in at the head of CmakeLists.txt and increment it, to ease identification of what is used, for the purpose of updating, etc. My own private way of keeping track that does not affect anyone else.

I had thought that would be useful in a broader way, but I am learning that some of the cmake files need to be adjusted for the specific plugiin sometimes.

rgleason, 2019-12-14:
I'm going to close this now.

closed 2019-12-14 by rgleason

leamas, 2019-12-14:
As long as you have the git commit as part of the build info, you can always track down exact versions used...

rgleason, 2019-12-14:
Got it. Good point.