OpenCPN/OpenCPN #1500: logging: Bogus error logging dialogs ats startup

leamas, 2019-12-07:
When  starting opencpn (v5.0.0) is displays two error boxes, initiated by wxWidgets after error returns from GTK3.

Error occurs  on Debian, and only with certain compilation  flags. In particular, it happens in the downstream 5.0.0 packaging.

This should not happen, the code initiates the logging so that it goes to a file. So it also does for later error messages, so this seems to be  about in which order things are initiated.

After stripping the first, wxWidgets parts the stacktrace from the first log (the second is similar):

```
BACKTRACE:
[1] wxWindow::GTKGetScrollEventType(_GtkRange*)
[2] g_signal_emit_valist
[3] g_signal_emit
[4] g_closure_invoke
[5] g_signal_emit_valist
[6] g_signal_emit
[7] gtk_adjustment_configure
[8] wxScrollHelper::DoAdjustScrollbar(_GtkRange*, int, int, int, int*, int*, int*)
[9] wxScrollHelper::AdjustScrollbars()
[10] wxGenericListCtrl::DoGetBestClientSize() const
[11] wxWindowBase::GetBestSize() const
[12] wxWindowBase::GetBestHeight(int) const
[13] wxBookCtrlBase::GetControllerSize() const
[14] wxBookCtrlBase::GetPageRect() const
[15] wxBookCtrlBase::DoSetSelection(unsigned long, int)
[16] wxListbook::DoRemovePage(unsigned long)
[17] wxBookCtrlBase::DeletePage(unsigned long)
[18] options::AddPage(unsigned long, wxString const&)
[19] options::CreatePanel_Display(unsigned long, int, int)
[20] options::CreateControls()
[21] options::options(MyFrame*, int, wxString const&, wxPoint const&, wxSize const&, long)
[22] MyFrame::OnInitTimer(wxTimerEvent&)

```
I have yet to understand why this happens

leamas, 2019-12-07:
SImple thing: it's about NDEBUG not being defined in release builds in Linux.

leamas, 2019-12-08:
OTOH, the underlying issue triggering the assert  is still there. It seems to be a bug in wxWidgets, file
[window.cpp](https://github.com/wxWidgets/wxWidgets/blob/master/src/gtk/window.cpp). Here, around line 6084 we have:

```
// Determine if increment is the same as +/-x, allowing for some small
//   difference due to possible inexactness in floating point arithmetic
static inline bool IsScrollIncrement(double increment, double x)
{
    wxASSERT(increment > 0);
    const double tolerance = 1.0 / 1024;
    return fabs(increment - fabs(x)) < tolerance;
}

```

This will trigger for increment == 0. There are posts out there recommending to change the condition to  `wxASSERT(increment > 0);`. We should file an upstream bug after searching their archive.

taruti, 2019-12-11:
I remember seeing this bug in Linux GTK3 builds.

leamas, 2019-12-11:
Yep, the downstream packaging is in the process of moving to gtk3 since Debian is about to drop the gtk2 support. So this is gtk3. And the good new is that it seems to work just fine if compiled with -DNDEBUG.

leamas, 2019-12-11:
hm... the downstream packaging also uses the upstream  wxsvg which has some patches related to this. I don't think the bundled wxsvg code has been updated with them, not sure...

leamas, 2020-03-02:
Filed upstream ticket: https://trac.wxwidgets.org/ticket/18688.

Until upstream bug is resolved we have to use -NDEBUG on gtk3. Keeping our own bug open  until it's verified that this is documented.

leamas, 2020-05-02:
Upstream bug seems to be fixed (knock, knock...) This should basically boil down to that once wxWidgets is updated, for linux to 3.2.x, problem should have been handled.

Until then, we have to live with -NDEBUG on gtk3 builds. Sigh. One possible route now when there is  a patch upstream might be file a bug against the Debian wxWidgets package, requesting them to apply this patch.

leamas, 2020-05-11:
Fixed in 5a10b0ab07d. Closing

closed 2020-05-11 by leamas