OpenCPN/OpenCPN #1504: Large commit

bdbcat, 2019-12-12:
https://github.com/OpenCPN/OpenCPN/commit/d155012248885d255855b94b11dd7f0b2501099f is a large-scale merge involving many modules.
This merge is important.  It brings the Android build platform fully into the master branch, and so improves overall maintainability and viability of that platform.
As you might expect, we may see some regression and merge/rebase problems as we digest this large update.
I apologize in advance for the required rebase efforts needed on the various upgrade projects currently being worked.
Dave

bdbcat, 2019-12-12:
And it is broken for MSW now.  Working on it...

leamas, 2019-12-12:
Following makes options.cpp compile under mingw

    diff --git a/src/OCPNPlatform.cpp b/src/OCPNPlatform.cpp
    index b0b89f156..4cd2fce15 100644
    --- a/src/OCPNPlatform.cpp
    +++ b/src/OCPNPlatform.cpp
    @@ -25,6 +25,12 @@
 
     #include "wx/wxprec.h"
 
    +#ifdef __MINGW32__
    +#undef IPV6STRICT    // mingw FTBS fix:  missing struct ip_mreq
    +#include <windows.h>
    +#endif
    +
    +

leamas, 2019-12-12:
I also have a MacOS error when trying to rebase:

```
/Users/travis/build/leamas/opencpn-tmp/src/chart1.cpp:3211:13: error: use of undeclared identifier 'applyDarkAppearanceToWindow'

            applyDarkAppearanceToWindow(MacGetTopLevelWindowRef(), true);

            ^

1 error generated.
```

No idea if "my" code is involved, but at a glance I don't see how it should cause this...

bdbcat, 2019-12-12:
All CI is building now, except mingw.  Looks like grib_pi is not finding wxWidgets.  Difficult to debug, as I have no mingw environment running yet.
Alec?

Thanks
Dave

leamas, 2019-12-12:
I can have a look, but it's late here. And tomorrow is Friday, wine and lamb... but later in the weekend.

Hakansv, 2019-12-12:
Dave..
Although not complete it builds in VS2017 after one error fix:
(Hopefully correct?)
```
@@ -1484,7 +1484,7 @@ static void AISDrawTarget( AIS_Target_Data *td, ocpnDC& dc, ViewPort& vp, ChartC
 
             wxBrush realtime_brush = wxBrush( GetGlobalColor( "GREY1" ) );
             dc.SetBrush( realtime_brush );
-            dc.StrokePolygon( nPoints, iconPoints, Point.x, Point.y, scale_factor );
+            dc.StrokePolygon( nPoints, iconPoints, Point.x, Point.y, AIS_scale_factor );
         }
         dc.SetBrush( target_brush );
```
Some notes, so far, if not to early:
O Crash when receiving UDP data to port 10110. 
The same data via COM (2) run wo crash.
Own ship real scaled vector also not good. See pict.
Real scaled raster and default ship is as usual.

[image]

leamas, 2019-12-12:
Whatever, cannot sleep anyway. Here is the mingw patch:

```
--- a/libs/wxJSON/CMakeLists.txt
+++ b/libs/wxJSON/CMakeLists.txt
@@ -31,4 +31,5 @@ add_library(ocpn::wxjson ALIAS WXJSON)
 
 set_property(TARGET WXJSON PROPERTY COMPILE_FLAGS "${OBJ_VISIBILITY}")
 target_include_directories(WXJSON PUBLIC ${CMAKE_CURRENT_SOURCE_DIR}/include)
+target_link_libraries(WXJSON PRIVATE ${wxWidgets_LIBRARIES})

```

bdbcat, 2019-12-13:
Alec...
Thanks.  I would not have looked there....

bdbcat, 2019-12-13:
Hakan...
I see no problem on UDP or TCP data reception with W7 client.
What type of NMEA data?  Any idea of the crash point, under debug mode?
I corrected the real-scaled vector ownship rendering.  Thanks for the catch.
Dave

Hakansv, 2019-12-13:
Dave..
Data is my old fashion: https://www.dropbox.com/s/jc872vqze5okbqr/Hakefjord.txt?dl=1
but sent via Dan's VDRPlayer like this: python %myFolder%VDRplayer.py %myFolder%Hakefjord.txt %ip%255 10110 0.15 UDP 
BUT- to brake it further down it's not about UDP and Network. That was a quick and false conclusion. It's the same crash when using VDR player instead of network UDP.
The O crash is when AIS has been received for a while. 10 sek? 

Debug:
I didn't so far succeed to debug. O hangs while init. 
Here on the, I think, second run (IsMaximized) in this function: (wxFrame:276 See last call stack for pos)
```
            ::SendMessage(GetHwnd(), WM_SIZE,
                          IsMaximized() ? SIZE_MAXIMIZED : SIZE_RESTORED,
                          MAKELPARAM(r.right - r.left, r.bottom - r.top));
        }
```
Last call stack before the hang: (VS debug says nothing about the hang. It's "dead waiting")
```
>	wxmsw312ud_core_vc_custom.dll!wxFrame::SendSizeEvent(int flags) Line 276	C++
 	opencpn.exe!MyFrame::ApplyGlobalSettings(bool bnewtoolbar) Line 5542	C++
 	opencpn.exe!MyApp::OnInit() Line 2313	C++
 	wxbase312ud_vc_custom.dll!wxAppConsoleBase::CallOnInit() Line 93	C++
 	wxbase312ud_vc_custom.dll!wxEntryReal(int & argc, wchar_t * * argv) Line 491	C++
 	wxbase312ud_vc_custom.dll!wxEntry(int & argc, wchar_t * * argv) Line 184	C++
 	wxmsw312ud_core_vc_custom.dll!wxEntry(HINSTANCE__ * hInstance, HINSTANCE__ * __formal, char * __formal, int nCmdShow) Line 305	C++
 	opencpn.exe!WinMain(HINSTANCE__ * hInstance, HINSTANCE__ * hPrevInstance, char * __formal, int nCmdShow) Line 1083	C++
 	[External Code]	
 	[Frames below may be incorrect and/or missing, no symbols loaded for kernel32.dll]	Unknown

```

Hakansv, 2019-12-13:
Dave.. Note 2
Real vector own ship is again fine after last commit. 
Thanks.

leamas, 2019-12-13:
Dave..
When you have some time: could you look into pluginmanager.cpp in #1460 around line 4400? I have made a sloppy rebasing here just to make it compile. However, we seem to have different ideas as to when two buttons should be displayed, and for now my idea just wins. It probably wrong.

Hakansv, 2019-12-13:
Dave..
AIS -> O Crash found:
ais.cpp: 1738
When TrackLength = 1 as i my case we're going further down and TrackPoints is uninitialized. See pict.

[image]

[image]

Hakansv, 2019-12-13:
Dave..
About VS2017 Debug hang:
Chart1.cpp : 5542
If I comment out the call for SendSizeEvent(); the Debug can run.
[image]

bdbcat, 2019-12-13:
Hakan...
Fix committed for zero-length track rendering.  Thanks for the detailed debug trace.
Dave

Hakansv, 2019-12-13:
Dave..
Thanks for the zero-length fix. Work fine!
One typo bug still there:  ("scale_factor" is not declared. typo?)
```
@@ -1484,7 +1484,7 @@ static void AISDrawTarget( AIS_Target_Data *td, ocpnDC& dc, ViewPort& vp, ChartC
 
             wxBrush realtime_brush = wxBrush( GetGlobalColor( "GREY1" ) );
             dc.SetBrush( realtime_brush );
-            dc.StrokePolygon( nPoints, iconPoints, Point.x, Point.y, scale_factor );
+            dc.StrokePolygon( nPoints, iconPoints, Point.x, Point.y, AIS_scale_factor);
         }
```

bdbcat, 2019-12-13:
Alec....
re: "When you have some time:..."
Done, tidied up somewhat, dropped some #ifdefs.  Only ~10,000 more to go :)
Dave

bdbcat, 2019-12-13:
Hakan...
I think you are having some merge problem with ais.cpp.  The current github codebase is correct.  In fact, I see no instance of "realtime_brush" in OCPN at all.  Try a fresh pull of master.

Hakansv, 2019-12-13:
Dave..
New "AIS" bug:
AIS target mouse rollover info is not shown. (All three options are set) 
With or without OpenGL don't matter. One or multi canvas don't matter.

Hakansv, 2019-12-13:
Dave.. 
Merge problem--
OK I'll make a new fetch and come back

Hakansv, 2019-12-13:
Dave..
So sorry for the lack of rebase of "myBranch". No build errors on master.

But still no AIS target rollover. Where can I debug?

tai-fun, 2019-12-13:
Håkan
Make sure that  Options->User Interface->" Enable Route/AIS info block" is
ticked  before you spend more time on the issue. This is a new option.

Thomas

Hakansv, 2019-12-13:
Thomas..
Ahaaa... Of course you're right. Then it's working! Thanks.
I may be curious about the reason for that new option. I do need some arguments since I can't found any needs and need some good ones for next training sessions. :)

bdbcat, 2019-12-14:
On reason for the new option relates to touch-screens.  Since there is no concept of "mouse hover" on a touch screen, the "rollover" can only be activated when the user touches the AIS target on screen.   On the other hand, the user also touches the screen to move the chart,zoom, etc.  Some find that having these target rollover boxes pop up willy-nilly can be distracting.  So, we add a method to turn them off.
Maybe it should default "on" with new installations and upgrades without touch-screens?
Dave

Hakansv, 2019-12-14:
Dave.. 
OK, thanks for the background. I understand. Good reasons for the new option.
Yes, I think the default "on" would still be suitable for most users.

bcn2, 2019-12-14:
> Maybe it should default "on" with new installations and upgrades without touch-screens?

Yes, and when toggling "Enable touch interface" perhaps a pop-up about the option.

Hakansv, 2019-12-15:
@bdbcat  @leamas 
Dave..
A continuous large commit today. Very nice!
Where do you want notes for this and probably coming commits for the, should we call it, Alfa test? 
1. General comments here or in a new issue or.....?
2. Specific comments about plugin loader in a new issue or here or...?

Here is one general for a start:
- On Win10: Waypoint properties window size: The window size is only default and is not saved when changed. Even if I drag it to a reasonable size and close. Next time it's back to default.
Håkan

leamas, 2019-12-15:
Speaking for no one besides me and my lice, I definitely prefer separate issues filed. They are just so much easier to track.

Hakansv, 2019-12-15:
Dave,,
About Waypoint properties dialog:
The issue is rather that the size when shown is to small. See pict. If all boxes would fit inside the main frame the need for resize is not there.  And here it's not because the canvas size is to small. 
I'm never into any of these if:  (chcanv.cpp: 8908)
     ```
  if(canvas_size.x < fitted_size.x){
            fitted_size.x = canvas_size.x - 40;
            if(canvas_size.y < fitted_size.y)
                fitted_size.y -= 40;                // scrollbar added
        }
        if(canvas_size.y < fitted_size.y){
            fitted_size.y = canvas_size.y - 40;
            if(canvas_size.x < fitted_size.x)
                fitted_size.x -= 40;                // scrollbar added
        }
```

![bild](https://user-images.githubusercontent.com/7202854/70870213-a6a7fc00-1f90-11ea-847d-b1875747f5ff.png)

Hakansv, 2019-12-15:
Picture to above post
[image]

bdbcat, 2019-12-16:
Hakan...
I agree with Alec.  Now that we are into implementation, and past philosophical discussion, lets take these issues separately, one-by-one.  Hopefully, each issue can be closed by specific SMALL commit.
Dave

leamas, 2019-12-18:
@bdbcat: Just a heads-up: #1527, needs some attention...

closed 2020-01-25 by bdbcat