OpenCPN/OpenCPN #1511: PI Installer: Advanced menus

rgleason, 2019-12-15:
I find there is something about the  PI Installer menus that could be improved, but I am not sure how or what to suggest, perhaps it is related to being clear about what the current URL does or can do.  For discussion.

Screen 1
[image]

Screen 2
[image]

Screen 3
[image]

Hakansv, 2019-12-15:
Rick..
Another example from OpenPlotter if you haven't seen it. Just to compare.

[image]

leamas, 2019-12-15:
Ack.

However, note that there is the "Update catalog" button which updates the catalog without any dialogs at all, besides some info.

Also note that  the OpenPlotter example is not really comparable to Rick's screenshot. Openplotter's is about updating plugins, Rick's is about updating the catalog i. e. the list of candidates.

leamas, 2019-12-15:
We need to move this to https://github.com/leamas/opencpn/issues.  Please copy-paste and close.

rgleason, 2019-12-15:
https://github.com/leamas/OpenCPN/issues/1

leamas, 2019-12-15:
Rick,

Honestly, I didn't see this coming, not now. Dave has merged the PR. This means that your two bugs here should be reopened, I'll close those at my fork. 

Sorry for the mess

rgleason, 2019-12-15:
No problem Alec, I don't think it is a bug, but I could be wrong, I just thought there was something funny about the way the menus work and wanted a little more clarity for users. 

I think what you have done is good, as it gives access to insert an metadata.xml URL for testing, while keeping the user out of that.

I also think the different levels "experiment", "release" is good. Most users will have it on "release", but beta testers will probably want to use "experiement" or "beta"  (Still not quite sure how this works in the background)

Showing the "Update" window is very instructive, and I used that to confirm I had the right URL loaded for testing.

Maybe the problem I have is, once you put the URL in the field there is no update confirmation that it took properly, as that is a separate action "update".

I hope this helps. Most users won't get into this part however, so if they do, we need to make it clear how to go back to the "default" "standard" setting I think, because some of us will certainly poke the black box to see how it works.

I will reopen, and close it when you say to.

rgleason, 2019-12-15:
BTW Congratulations to you on the Merge!  I hope this works out as it will be a big time saver once the kiinks are worked out, I think.

leamas, 2019-12-15:
I'm not sure this a day to celebrate. There are tons of things to do...

The windows: here is at least on bug compared to my initial ideas: The smallest status window should no be visible when the advanced dialog is shown.

That said, the advanced dialog is, well, advanced. Most users will just just the Update Catalog button.

And, finally: here is room for a completely new UI. However, this is a lot of work, both specification-wise and actual coding.

leamas, 2019-12-15:
As for the UI: discussion is nice. But we need concrete proposals to advance this, we have passed the "general ideas" stage. ;)

rgleason, 2020-01-03:
Alec's commits perhaps also fix this
https://github.com/OpenCPN/OpenCPN/issues/1612

rgleason, 2020-01-03:
Will have to wait to try these commits. No artifacts.

rgleason, 2020-01-06:
This is old and there is ongoing work. Closing

closed 2020-01-06 by rgleason