OpenCPN/OpenCPN #1517: PI Installer: Deployment - PR #1457   It would be really sweet

rgleason, 2019-12-15:
I know there is some coordination and effort needed to accomplish this, and it is certainly not expected immediately, but it would be really nice if Cloudsmith and/or github deployment or even appveyor artifacts,  was supported.
https://github.com/OpenCPN/OpenCPN/pull/1457

If I can help somehow?

leamas, 2019-12-15:
As of now, anything is supported which produces downloadable tarballs and XML metadata. Please feel free to initiate separate efforts to formalize the use of CI environments such as  circleci.com and cloudsmith.com. 

There will probably initiallly be different setups in different plugin projects. Time will tell if there will be a single setup which will rule them all...

rgleason, 2019-12-15:
Yes, well I guess I was not clear about this. I would like to be able to download an appveyor artifact or cloudsmith installation of this version of Opencpn.

leamas, 2019-12-15:
Right... seems reasonable. Take a look in my usual cloudsmith repo in a few hours.

leamas, 2019-12-16:
The builds at https://cloudsmith.io/~alec-leamas/repos/opencpn/packages/ now contains updated builds. See also #1522.

rgleason, 2019-12-17:
Thank you Alec.  I will close this now.

closed 2019-12-17 by rgleason