OpenCPN/OpenCPN #1520: glChartCanvas.cpp will not compile on some Linux (e.g. Raspberry Pi)

transmitterdan, 2019-12-15:
This code:

```
#if defined(__OCPN__ANDROID__)
#include "androidUTIL.h"
#elif defined(__WXQT__)
#include <GL/glx.h>
#endif
```
blocks include of glx.h on GTK based systems.  Is there any reason it should not be like this?

```
#if defined(__OCPN__ANDROID__)
#include "androidUTIL.h"
#elif defined(__WXQT__) || defined(__WXGTK__)
#include <GL/glx.h>
#endif
```

bdbcat, 2019-12-16:
Dan...
That seems safe, but the code looks funny.
What fails on rPI compile with glx.h missing?
Dave

transmitterdan, 2019-12-16:
Dave,

This is the error report from gcc.

```
/home/pi/opencpn/src/glChartCanvas.cpp: In function ‘void (* ocpnGetProcAddress(const char*, const char*))()’:
/home/pi/opencpn/src/glChartCanvas.cpp:287:74: error: ‘glXGetProcAddress’ was not declared in this scope
 #define systemGetProcAddress(ADDR) glXGetProcAddress((const GLubyte*)ADDR)
                                                                          ^
/home/pi/opencpn/src/glChartCanvas.cpp:317:29: note: in expansion of macro ‘systemGetProcAddress’
     return (GenericFunction)systemGetProcAddress(addrbuf);
```

bdbcat, 2019-12-16:
Dan...
OK, thanks.  I will investigate why this changed after the big merge.
Meanwhile, your patch looks safe to use until then.
Dave

transmitterdan, 2019-12-16:
Dave,

I checked out V5.0.x and it has the same problem. So probably not due to the big merge.

closed 2019-12-30 by transmitterdan

dominig, 2020-08-24:
Dave,
I have the same issue with 5.2 on OpenSUSE.
The patch allows to pass the issue but is certainly not the root cause.
Dominig