OpenCPN/OpenCPN #1521: Update to new plugin installer version causes plugins not to find thier icons

jongough, 2019-12-16:
I installed opencpn on Ubuntu 16.04 from the official repositories supplied by the download link. I then installed opencpn-plugin-draw from the same repository. Running OpenCPN worked as expected and the icons for opencpn-plugin-draw were found and displayed correctly on the main toolbar.

I downloaded the current build of opencpn from the cloudsmith repository, `https://cloudsmith.io/~alec-leamas/repos/opencpn/packages/`, that was built after the acceptance of pull 1457 into this repository. The file was `opencpn-xenial_5.0.0+1.4c8be5c-1_amd64.deb`. Running OpenCPN worked but the icons for opencpn-plugin-draw were not found. OpenCPN told the opencpn-plugin-draw that its icons were in `/usr/local/share/opencpn/plugins/ocpn_draw_pi/data/`, this information was returned by the `GetpSharedDataLocation()` API call which has been in existence since before API 1.9 was introduced. The API 1.16 call `GetPluginDataDir( const char* plugin_name)` used to call the same `GetpSharedDataLocation()` function at the start. This has now been changed with this new plugin installer version.

The location of the icon files for opencpn-plugin-draw for the previous version of opencpn was `/usr/share/opencpn/plugins/ocpn_draw_pi/data`. The icons are still in this location, however, this location is ignored by opencpn after accepting pull request 1457.

This issue was raised before this pull request was accepted but it was not fixed. It now needs to be fixed.
[Post pull 1457 opencpn log.txt](https://github.com/OpenCPN/OpenCPN/files/3966288/Post.pull.1457.opencpn.log.txt)
[Pre pull 1457 opencpn log.txt](https://github.com/OpenCPN/OpenCPN/files/3966289/Pre.pull.1457.opencpn.log.txt)

leamas, 2019-12-16:
This is probably a question how the package is built. IIRC, there is no specific /usr prefix setup in the build which makes it default to using /usr/local according to GNU standards. It might be a good idea to adjust  build scripts to use /usr to make the builds compatible with the official .deb packages.

leamas, 2019-12-19:
Is the problem resolved in the latest builds in https://cloudsmith.io/~david-register/repos/opencpn-unstable/packages/ ?

jongough, 2019-12-19:
This fixes the issue of plugins installed and upgrading to the newer version. I did the test as above on a clean Ubuntu 16.04 with newly added opencpn ppa then upgrading to the cloudsmith version. The installed plugins continue to be loaded and look in the original location for data files.

Looking through the logs I notice that `/usr/local/*` is no longer referenced being `/usr/*` instead. This used to be a build option that was supplied in CMAKE_INSTALL_PREFIX, is this still used?

leamas, 2019-12-19:
The patch applied is exactly this, you can inspect it above. Nothing has changed w r t this.

jongough, 2019-12-19:
But according to your own comments about standards you should still include `/usr/local/*` as a searched location. Whilst the fix works and is how the system used to work before the plugin installer was included you have broken your own standards to do so.

leamas, 2019-12-20:
First, this is not about me or my shortcomings. Let us focus on discussing the code instead.

Secondly: the support for dynamic plugin relocation and related freedesktop standards, the basic underpinnings for the installer, is only available in `GetPluginDataDir()`. The now legacy `GetpSharedDataLocation()` works as it always have, returning a hardcoded location.

jongough, 2019-12-20:
Who said anything about shortcomings. You have pointed out that OCPN and my plugins do not follow standards. So I was just commenting on the fact that your change didn't either and wondered why. I would have thought you would have added the dynamic execution stuff and co-related shared data directories to the search path in the process rather than just changing the CMAKE variable. There have been issues in the past with that approach, which is what I was assuming you were making the changes for.

There is always as transition period and in that transition there is a need to keep legacy code working until all the required changes have been made. Then deprecate the legacy methods and change the version numbers to inform all of what is happening.

I am just testing the environment out to ensure the best user experience. Sorry if this offends.

leamas, 2019-12-20:
Well, the question boils down  to: is this bug resolved?

jongough, 2019-12-20:
https://github.com/OpenCPN/OpenCPN/issues/1521#issuecomment-567678803
The fix works and the original plugin continues to work after the upgrade to the new version. This issue is fixed.

closed 2019-12-20 by jongough