OpenCPN/OpenCPN #1522: RFE: Set up the new CI deployment

leamas, 2019-12-16:
After merging #1457 there is setup to make test builds available for devs and early testers. The result is something like https://cloudsmith.io/~alec-leamas/repos/opencpn/packages/ which holds the latest builds as downloadable installers for msvc, mingw, macos and debian.

IMHO, this functionality should either be dropped or transferred to the project. Assuming the latter, the following needs to be done by project maintainers (specifically, owners of the github opencpn account).

  - Create a cloudsmith account (don't care about payment notices, open-source is free).
  - Setup up a cloudsmith repo . NOTE: define it as open-source, cannot be changed 
    after creation. It is probably best named opencpn-unstable or so since this is, well,
    unstable builds.
  - The cloudsmith repo has an API key available in the UI. This key should be made 
    available as an environment variable named CLOUDSMITH_API_KEY in the
    travis build setup.

Once this is done I can submit PR with minimal configuration changes to make this fly -- that is, the latest travis builds are always available to for testing.

bdbcat, 2019-12-16:
Alec...
cloudsmith account and repo built.

Repo:  david-register/opencpn-unstable
API key:  e58a442fefeb8aea47b94b7a910b79641cf8f379

Need anything else?

leamas, 2019-12-17:
I have sent some private messages on this. Please note the the API keys should be kept private.

bdbcat, 2019-12-17:
I got no PMs.  What channel?
Dave

rgleason, 2019-12-17:
Dave, you probably should delete the message above too.  Thank.s

bdbcat, 2019-12-17:
Rick..
I can generate a new key whenever I hear from Alec.
Dave

leamas, 2019-12-17:
> I got no PMs. What channel?

The yahoo account

leamas, 2019-12-18:
First builds available at https://cloudsmith.io/~david-register/repos/opencpn-unstable/packages/. All looks OK, despite that some platforms don't build (unrelated to this)

Thanks to all involved!

Closing

closed 2019-12-18 by leamas