OpenCPN/OpenCPN #1524: Rasbian stretch build error ChartDataInputStream.cpp

Hakansv, 2019-12-16:
Built on a RPi 3B+ stretch
I saw trusty and others have built fine. I got these errors some hours ago. 
Anything I need to do?
Håkan
```
/usr/bin/ld: CMakeFiles/opencpn.dir/src/ChartDataInputStream.cpp.o: undefined reference to symbol 'lzma_stream_decoder@@XZ_5.0'
//lib/arm-linux-gnueabihf/liblzma.so.5: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
CMakeFiles/opencpn.dir/build.make:2987: recipe for target 'opencpn' failed
make[2]: *** [opencpn] Error 1

CMakeFiles/Makefile2:120: recipe for target 'CMakeFiles/opencpn.dir/all' failed
make[1]: *** [CMakeFiles/opencpn.dir/all] Error 2
Makefile:141: recipe for target 'all' failed
make: *** [all] Error 2

From ChartDataInputStream.cpp: 118

void wxCompressedFFileInputStream::init_lzma()
{
    lzma_stream s = LZMA_STREAM_INIT;
    memcpy(&strm, &s, sizeof s);
    lzma_ret ret = lzma_stream_decoder(
        &strm, UINT64_MAX, LZMA_CONCATENATED);

    if(ret != LZMA_OK)
        m_lasterror = wxSTREAM_READ_ERROR;
```

leamas, 2019-12-16:
hm... dr Google recommends checking https://github.com/mwydmuch/ViZDoom/issues/75, where they checked and eventually handled a similar (same?) problem.

Hakansv, 2019-12-16:
My search:
pi@raspberrypi:~ $ ldconfig -p | grep lzma
	liblzma.so.5 (libc6,hard-float) => /lib/arm-linux-gnueabihf/liblzma.so.5
	liblzma.so (libc6,hard-float) => /usr/lib/arm-linux-gnueabihf/liblzma.so

pi@raspberrypi:~ $ ls /lib/arm-linux-gnueabihf/ | grep lzma
liblzma.so.5
liblzma.so.5.2.2

leamas, 2019-12-16:
hm.. is /usr/lib/arm-linux-gnueabihf/liblzma.so  missing?

Hakansv, 2019-12-16:
No, diden't searched correct I suppose:
pi@raspberrypi:~ $ ls /usr/lib/arm-linux-gnueabihf/ | grep lzma
liblzma.a
liblzma.so
pi@raspberrypi:~ $

leamas, 2019-12-16:
hm, again... you should have the symbols there, right?

    $ objdump -T /lib/arm-linux-gnueabihf/liblzma.so.5 | grep stream_decoder
    0000bb20 g    DF .text	00000078  XZ_5.0      lzma_stream_decoder

Assuming that one looks OK: could you please post the full build log? Question is if  the  `-llzma` flag is missing...

leamas, 2019-12-16:
I definitely need some care: the error is right in our face.

https://stackoverflow.com/questions/19901934/libpthread-so-0-error-adding-symbols-dso-missing-from-command-line

leamas, 2019-12-16:
You have some workaround there (these days cmake should respect LDFLAGS). That said, this looks like a bug, most likely in the cmake setup, only exposed on the pi.

Hakansv, 2019-12-16:
Built output and cmake log attached.  Is that what you need?

[RPi3_O_built.txt](https://github.com/OpenCPN/OpenCPN/files/3970058/RPi3_O_built.txt)
[CMakeOutput.log](https://github.com/OpenCPN/OpenCPN/files/3970071/CMakeOutput.log)

Hakansv, 2019-12-16:
And..
$ objdump -T /lib/arm-linux-gnueabihf/liblzma.so.5 | grep stream_decoder
0000baa0 g    DF .text	00000078  XZ_5.0      lzma_stream_decoder

Hakansv, 2019-12-16:
And sys info whether you need it or not:
pi@raspberrypi:~/builds/OpenCPN/build $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
VERSION_CODENAME=stretch
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@raspberrypi:~/builds/OpenCPN/build $

leamas, 2019-12-16:
If you indent those lines (four spaces) they are more readable... 

Yes, that should be all I need. A quick grep through it reveals that there is no -llzma flag, so something is indeed wrong in the cmake code.

Good catch! 

I have too much to do, but perhaps...

Hakansv, 2019-12-16:
Very good! I've no hurry at all. This was just a test on one of the "animals" on the desk.
I've a RPi 4 buster as well. But running Openplotter for the moment. If you can use a test from that as well please say and I'll fix.

leamas, 2019-12-16:
The build log has 
```
-- Could NOT find LibLZMA (missing:  LIBLZMA_HAS_AUTO_DECODER LIBLZMA_HAS_EASY_ENCODER LIBLZMA_HAS_LZMA_PRESET) (found version "5.2.2")
-- lzma library not found
```
So it starts there....  First: have installed all required deps? Here is the updated list from the debian packaging:

```
cmake,
 libarchive-dev,
 libbz2-dev,
 libcurl4-gnutls-dev,
 libelf-dev,
 libexif-dev,
 libgdk-pixbuf2.0-dev,
 libgtk-3-dev,
 liblz4-dev,
 liblzma-dev,
 libpango1.0-dev,
 libsqlite3-dev,
 libtinyxml-dev,
 libunarr-dev,
 libwxsvg-dev (>=  2:1.5.21),
 libwxgtk3.0-gtk3-dev,
 portaudio19-dev
```

Hakansv, 2019-12-17:
liblzma-dev is included in the official list so it's there as of the grep list above.

Now I haven't yet checked "on board"  but from your list these are missing in the official list. If there's another I've missed that. Sorry.
 ibarchive-dev,
 libgdk-pixbuf2.0-dev,
 libgtk-3-dev,
 liblz4-dev, (have 2)
 libpango1.0-dev,
 libunarr-dev,

What's installed is the list here: https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:developer_guide:compiling_linux

 sudo apt-get install build-essential cmake 
  gettext git-core gpsd gpsd-clients 
  libgps-dev 
  wx-common 
  libwxgtk3.0-dev 
  libglu1-mesa-dev 
  libgtk2.0-dev 
  wx3.0-headers 
  libbz2-dev 
  libtinyxml-dev 
  libportaudio2 
  portaudio19-dev 
  libcurl4-openssl-dev 
  libexpat1-dev 
  libcairo2-dev 
  libarchive-dev 
  liblzma-dev 
  libexif-dev 
  libelf-dev 
  libsqlite3-dev

leamas, 2019-12-17:
yep... I'm a little lost, but it should definitely build on stretch. Could you make a try after installing libarchive-dev (running cmake should be sufficient, look for if it finds liblzma)

Hakansv, 2019-12-17:
I checked your deps list above and the following was not installed:
libgtk-3-dev  - now OK    
liblz4-dev - now OK 
pi@raspberrypi:~ $ sudo apt-get install libunarr-dev  
E: Unable to locate package libunarr-dev
And...
pi@raspberrypi:~ $ sudo apt-get install liblzma-dev 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
liblzma-dev is already the newest version (5.2.2-1.2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
pi@raspberrypi:~ $ 
pi@raspberrypi:~/builds/OpenCPN/build $ ldconfig -p | grep liblzma   
	liblzma.so.5 (libc6,hard-float) => /lib/arm-linux-gnueabihf/liblzma.so.5
	liblzma.so (libc6,hard-float) => /usr/lib/arm-linux-gnueabihf/liblzma.so
But.. cmake ../
-- Could NOT find LibLZMA (missing:  LIBLZMA_HAS_AUTO_DECODER LIBLZMA_HAS_EASY_ENCODER LIBLZMA_HAS_LZMA_PRESET) (found version "5.2.2")
-- lzma library not found
And one more for next step..
-- *** Staging to build grib_pi ***
-- Could NOT find JPEG (missing:  JPEG_LIBRARY JPEG_INCLUDE_DIR) 
-- Could NOT find Jasper (missing:  JASPER_LIBRARIES JASPER_INCLUDE_DIR JPEG_LIBRARIES) 
-- Using bundled jasper library...
-- *** Added plugin: /home/pi/builds/OpenCPN/plugins/grib_pi

One note or observation: I have to use sudo in order to get cmake to work. This is what I know not usual but would not have any impact on this error?

leamas, 2019-12-17:
Thou shall not use sudo when running cmake. Ownership problems in the build/ subdir?

The cmake output is just weird. I'll try to flash a raspberry and try myself. Stay tuned.

Hakansv, 2019-12-17:
Alec..
Now cmake found everything. make is ongoing. Take some time.
I'm back in a couple of hours when make is eventually ready.

My changes to CMakeLists: ( I added the dir where LibLZMA resides! So sub-dirs are not included?)
```
if (NOT WIN32 AND NOT APPLE AND NOT QT_ANDROID)
    if (EXISTS "/usr/lib64")
        set(_LIBDIRS "~/.local/lib:/usr/local/lib64:/usr/lib64")
    else ()
        set(_LIBDIRS "~/.local/lib:/usr/local/lib:/usr/lib:/usr/lib/arm-linux-gnueabihf"
	)
    endif ()
    set(OCPN_LINUX_LOAD_PATH "${_LIBDIRS}"
        CACHE STRING "Plugin libraries directories (${_LIBDIRS})"
    )
endif ()
message(STATUS "HAKAN*** Build Architecture is ${_LIBDIRS}")
```
So...
-- Found TinyXML: /usr/include  
-- Looking for lzma_auto_decoder in /usr/lib/arm-linux-gnueabihf/liblzma.so
-- Looking for lzma_auto_decoder in /usr/lib/arm-linux-gnueabihf/liblzma.so - found
-- Looking for lzma_easy_encoder in /usr/lib/arm-linux-gnueabihf/liblzma.so
-- Looking for lzma_easy_encoder in /usr/lib/arm-linux-gnueabihf/liblzma.so - found
-- Looking for lzma_lzma_preset in /usr/lib/arm-linux-gnueabihf/liblzma.so
-- Looking for lzma_lzma_preset in /usr/lib/arm-linux-gnueabihf/liblzma.so - found
-- Found LibLZMA: /usr/include (found version "5.2.2") 
-- Building with lzma support
Next..
What about this one? Important?
-- *** Staging to build grib_pi ***
-- Could NOT find JPEG (missing:  JPEG_LIBRARY JPEG_INCLUDE_DIR) 
-- Could NOT find Jasper (missing:  JASPER_LIBRARIES JASPER_INCLUDE_DIR JPEG_LIBRARIES) 
-- Using bundled jasper library.

leamas, 2019-12-17:
You don't need to patch CMakeLists, it works if the invocation is correct. This is my line: 

    cmake  \
            -DCMAKE_LIBRARY_PATH=/usr/lib/arm-linux-gnueabihf \
            -DCMAKE_INSTALL_PREFIX=/usr \
            -DBUNDLE_DOCS=OFF \
            -DBUNDLE_TCDATA=ON \
            -DBUNDLE_GSHHS=CRUDE \
            -DBUILD_SHARED_LIBS=OFF \
            -DUSE_BUNDLED_LIBS=OFF \
            -DOCPN_NEW_SERIAL=OFF \
            -DOCPN_FORCE_GTK3=ON \
            ..

Also, here is the list of dependencies:

               cmake,
               gettext,
               libarchive-dev,
               libbz2-dev,
               libcurl4-gnutls-dev,
               libelf-dev,
               libexif-dev,
               libgdk-pixbuf2.0-dev,
               libgtk-3-dev,
               liblz4-dev,
               liblzma-dev,
               libpango1.0-dev,
               libsqlite3-dev,
               libtinyxml-dev,
               libunarr-dev,
               libwxgtk3.0-gtk3-dev,
               libwxsvg-dev (>= 2:1.5.21),
               portaudio19-dev

leamas, 2019-12-17:
Everything is from downstream packaging, verified on a clean stretch installation.

Hakansv, 2019-12-17:
Neither of your invocations worked. E.g. the use of: DCMAKE_LIBRARY_PATH=/usr/lib/arm-linux-gnueabihf makes all other libs not found.
And the -DBUNDLE... resulted in:
CMake Warning:
  Manually-specified variables were not used by the project:
    BUNDLE_GSHHS
    BUNDLE_TCDATA

But with the CMakeLists patch and only: cmake ../ it works. And the best would be with a CMakeLists taking care of a "normal" build?
The dependencies list may be correct and the lunix compiler manual then need some tweaking.

I'll check on RPi buster for the LibLZMA location later on.

leamas, 2019-12-17:
So, there are differences between our two stretch systems. We need some common ground. I havn't access to mine right now, complicating things.

The invocations are from the debian packaging, so they are not taken out of thin air. They are used to build the package on many platforms, on clean builders. So, we should give it a chance.

Patching CMakeLists with arm-specific paths is IMHO  the wrong way to do it. It creates platform dependencies that just shouldn't be required. 

The open question: what are the differences between out two system? "mystified"

leamas, 2019-12-17:
My bad... I'm using Buster, explains the differences.

Should we really support  2.5 year old Stretch on the master branch? Even older when next Ocpn release?

Hakansv, 2019-12-17:
> Should we really support 2.5 year old Stretch on the master branch? Even older when next Ocpn release?

Well, the Linux world is not my table and I've no opinion. but so far we still support WinXP and 2.5 years seems than comparatively short :) I think I bought that RPi3 device one year ago or so. It's often updated and still on stretch so I'm not alone? But it may be most will go for a RPi4 from now on so you be right when it comes to build from source. A note on the wiki may do?
The plugin installer would however support stretch and trusty I suppose?
A short plugin loader test today showed no result for stretch but that's maybe not a surprise?
Håkan

leamas, 2019-12-17:
The installer as such has no limitations. However, there need to be actual builds for all versions  used -- that is, different tarballs and XML metadata for Stretch and Buster. It's a major PITA, and one of the reasons flatpak is attractive.

Hakansv, 2019-12-17:
Yes, understood. Could pass a link or whatever so I can learn more about what Flatpack is. I've not followed this discussion in detail. And as you know I'm mainly dedicated to Windows.

leamas, 2019-12-17:
https://flathub.org/home

Hakansv, 2019-12-17:
OK, thanks. Very good. What's directly comes to my mind:
The Linux world is very complex and diverged. Windows four main issues, XP, 7, 8 and 10, seems comparable simple. And since MS bought Git it may be they head for open source?? (Hopefully not the opposite.)
Anyhow. For Raspbian they support from stretch and up, so I'm home with my RPi3B :)

leamas, 2019-12-17:
> The Linux world is very complex and diverged. 

Indeed. That's why I have headed for flatpak (works on all linuxes) and pure debian (mainly for resource-constrained arm devices like the rPi) With these two packages we cover the linux universe.

The installer is the tool that hopefully in the end will provide plugins to these platforms, avoiding the need for full-fledged packages, and for users to find them.

Hakansv, 2019-12-17:
Så är det möjligen herr Larsson vi har nöjet att...

leamas, 2019-12-17:
Nix,icke så. Vi har inte träffats...

Hakansv, 2019-12-19:
One last comment on this
I use this patch. Works. Would it hurt other system or can I PR?
CmakeList: 259  >>
```
if (NOT WIN32 AND NOT APPLE AND NOT QT_ANDROID)
    if (EXISTS "/usr/lib64")
        set(_LIBDIRS "~/.local/lib:/usr/local/lib64:/usr/lib64")
    else ()
>>	if (EXISTS "/usr/lib/arm-linux-gnueabihf") 
	    set(_LIBDIRS "~/.local/lib:/usr/local/lib:/usr/lib:/usr/lib/arm-linux-gnueabihf")
	else ()
            set(_LIBDIRS "~/.local/lib:/usr/local/lib:/usr/lib")
	endif ()	
    endif ()
    message(STATUS "*** Linux load path is ${_LIBDIRS}")
    set(OCPN_LINUX_LOAD_PATH "${_LIBDIRS}"
        CACHE STRING "Plugin libraries directories (${_LIBDIRS})"
    )
endif ()
```

leamas, 2019-12-19:
FWIW, I don't see any major problems with this. Nit-picking, it's IMHO better written using elseif, like

    if (EXISTS "/usr/lib64")
      set(_LIBDIRS "~/.local/lib:/usr/local/lib64:/usr/lib64")
    elseif (EXISTS "/usr/lib/arm-linux-gnueabihf") 
      set(_LIBDIRS "~/.local/lib:/usr/local/lib:/usr/lib:/usr/lib/arm-linux-gnueabihf")
    else ()
      set(_LIBDIRS "~/.local/lib:/usr/local/lib:/usr/lib")
    endif ()	

Also, the platform-specific library probably have a higher priority and should be tried first:
`~/.local/lib:/usr/local/lib:/usr/lib/arm-linux-gnueabihf:/usr/lib`

Hakansv, 2019-12-19:
Ok, I fix PR or do you want?

leamas, 2019-12-19:
You fix... I think I have had my share ;)

Hakansv, 2019-12-19:
Fixed by #1532

closed 2019-12-19 by Hakansv

leamas, 2019-12-19:
Just a workflow heads-up: I would say that the common pattern is to leave the bug (i. e., this bug) open until the PR fixing it is merged.  It makes things easier for people running into the same problem...

Hakansv, 2019-12-19:
Sounds like a structural and well organized work flow. Agreed!