OpenCPN/OpenCPN #1525: Waypoint properties window bad size

Hakansv, 2019-12-16:
I move this from "Large commit"
Initial trouble as of the pict below.
To make the sizers inside the main frame to fit I changed "newFit" to true so "fitted_size" will have effect. So far so good.
But still is the main frame to small for the tab: "Extended". One way to fix that is to setMinSize in function "UpdateProperties."
But I've bad feelings with this solution.
- First we evaluate fitted_size and then destroy it with SetMinSize. If fitted_size could fix also the main size it would be the best?
- How does my SetMinSize fit on smaller screens and not Win?
Anyone with more wxWidgets frame creation knowledge may take over? 
Håkan
```
@@ -1562,7 +1562,7 @@ bool MarkInfoDlg::UpdateProperties( bool positionOnly )
     #endif
     
     Fit();
-    //SetMinSize(wxSize(-1, 600));
+    SetMinSize(wxSize(320, 500));
     RecalculateSize();
     
     return true;

-------------------------------- src/chcanv.cpp --------------------------------
index 04f9c8299..c5862f65b 100644
@@ -8905,7 +8905,7 @@ void ChartCanvas::ShowMarkPropertiesDialog( RoutePoint* markPoint ) {
         wxPoint canvas_pos = GetPosition();
         wxSize fitted_size = g_pMarkInfoDialog->GetSize();;
 
-        bool newFit = false;
+        bool newFit = true;
         if(canvas_size.x < fitted_size.x){
             fitted_size.x = canvas_size.x - 40;
             if(canvas_size.y < fitted_size.y)
```
Initial size
[image]

After my patch:
[image] 
[image]

bdbcat, 2019-12-18:
Hakan...
Corrected by 6b3804c4dbd593607c2fe1d70ffb4d4a1894fde8.  OK to close?

Hakansv, 2019-12-18:
Dave..
No, not really see pict below. The description box is better and the button functional but compare the outlook to the two last pictures above.
And still is the nice "fitted_size" code not used by a if() condition. chcanv.cpp:8908
So my (ruff?) method above still works. First set "SetMinSize" and then check it's fiting on screen by the use of "Fitted_size". If nothing better comes up I can PR after a RPi, small screen, test?

[image]

Hakansv, 2019-12-18:
Dave..
I've made a simple solution. It's not the best result for this complex code. The dialog creation jumps around on different functions conditionally from where it's started. I'll make a PR. Take it or scrap it.

closed 2019-12-18 by Hakansv