OpenCPN/OpenCPN #1527: Runtime error: OpenCPN does not start

pcav, 2019-12-17:
A fresh install on a Debian Sid box apparently went fine, but the program refuses to start:
```
./opencpn 

12:01:09 CET: No styles found at: /usr/local/share/opencpn/uidata/
12:01:09 CET: No styles found at: /home/paolo/
12:01:09 CET: No styles found at: /home/paolo/.opencpn/
12:01:09 CET: The requested style was not found: 
12:01:09 CET: Using SVG Icons
12:01:09 CET: The requested style was not found: MUI_flat
```

pcav, 2019-12-17:
Sorry, my fault: I ran the executable directly from the build dir, without ``make install`` first.
However, the program segfaulst straight after giving **OK** to the first dialog (Welcome to OpenCPN).

leamas, 2019-12-17:
Now, to be able to  cope with this we need more information. As a starter: could you run the program in gdb, let it crash, make a stacktrace using the`bt` gdb command, and finally post it?

pcav, 2019-12-17:
Starting program: /usr/local/src/OpenCPN/build/opencpn 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffebce0700 (LWP 106983)]
[New Thread 0x7fffeb4df700 (LWP 106984)]
[Detaching after vfork from child process 106991]
[Detaching after vfork from child process 106993]
[Detaching after vfork from child process 106995]
[Detaching after vfork from child process 106997]
[Detaching after vfork from child process 106999]
[Detaching after vfork from child process 107005]
[Detaching after vfork from child process 107010]
[Detaching after vfork from child process 107016]
[Detaching after vfork from child process 107021]
[Detaching after vfork from child process 107023]
[New Thread 0x7fffea606700 (LWP 107025)]
[New Thread 0x7fffe8a33700 (LWP 107031)]

Thread 1 "opencpn" received signal SIGSEGV, Segmentation fault.
0x0000555555a371de in Piano::BuildGLTexture (this=0x55555643f000) at /usr/local/src/OpenCPN/src/piano.cpp:324
324	    for(unsigned int i = 0; i < (sizeof bitmaps) / (sizeof *bitmaps); i++) {

#0  0x0000555555a371de in Piano::BuildGLTexture() (this=0x55555643f000) at /usr/local/src/OpenCPN/src/piano.cpp:324
#1  0x0000555555a37515 in Piano::DrawGL(int) (this=this@entry=0x55555643f000, off=-17) at /usr/local/src/OpenCPN/src/piano.cpp:347
#2  0x0000555555b647c4 in glChartCanvas::DrawChartBar(ocpnDC&) (this=this@entry=0x55555650b280, dc=...) at /usr/local/src/OpenCPN/include/chcanv.h:269
#3  0x0000555555b79373 in glChartCanvas::Render() (this=0x55555650b280) at /usr/local/src/OpenCPN/src/glChartCanvas.cpp:4806
#4  0x0000555555b79e26 in glChartCanvas::OnPaint(wxPaintEvent&) (this=0x55555650b280, event=...) at /usr/local/src/OpenCPN/src/glChartCanvas.cpp:1650
#5  0x00007ffff71f6231 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) () at /lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#6  0x00007ffff71f6323 in wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*) () at /lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#7  0x00007ffff71f666c in wxEvtHandler::TryHereOnly(wxEvent&) () at /lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#8  0x00007ffff71f66f3 in wxEvtHandler::ProcessEventLocally(wxEvent&) () at /lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#9  0x00007ffff71f6791 in wxEvtHandler::ProcessEvent(wxEvent&) () at /lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#10 0x00007ffff71f6527 in wxEvtHandler::SafelyProcessEvent(wxEvent&) () at /lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#11 0x00007ffff757d43d in wxWindow::GTKSendPaintEvents(_cairo*) () at /lib/x86_64-linux-gnu/libwx_gtk3u_core-3.0.so.0
#12 0x00007ffff7d1062c in wxGLCanvas::OnInternalIdle() () at /lib/x86_64-linux-gnu/libwx_gtk3u_gl-3.0.so.0
#13 0x00007ffff772b6f3 in wxWindowBase::SendIdleEvents(wxIdleEvent&) () at /lib/x86_64-linux-gnu/libwx_gtk3u_core-3.0.so.0
#14 0x00007ffff772b728 in wxWindowBase::SendIdleEvents(wxIdleEvent&) () at /lib/x86_64-linux-gnu/libwx_gtk3u_core-3.0.so.0
#15 0x00007ffff772b728 in wxWindowBase::SendIdleEvents(wxIdleEvent&) () at /lib/x86_64-linux-gnu/libwx_gtk3u_core-3.0.so.0
#16 0x00007ffff75c6b9f in wxFrame::SendIdleEvents(wxIdleEvent&) () at /lib/x86_64-linux-gnu/libwx_gtk3u_core-3.0.so.0
#17 0x00007ffff760686d in wxAppBase::ProcessIdle() () at /lib/x86_64-linux-gnu/libwx_gtk3u_core-3.0.so.0
#18 0x00007ffff7537385 in wxApp::DoIdle() () at /lib/x86_64-linux-gnu/libwx_gtk3u_core-3.0.so.0
#19 0x00007ffff7537473 in  () at /lib/x86_64-linux-gnu/libwx_gtk3u_core-3.0.so.0
#20 0x00007ffff5758d7e in g_main_context_dispatch () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#21 0x00007ffff5759130 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#22 0x00007ffff5759403 in g_main_loop_run () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#23 0x00007ffff5a94705 in gtk_main () at /lib/x86_64-linux-gnu/libgtk-3.so.0
#24 0x00007ffff7554265 in wxGUIEventLoop::DoRun() () at /lib/x86_64-linux-gnu/libwx_gtk3u_core-3.0.so.0
#25 0x00007ffff70b599d in wxEventLoopBase::Run() () at /lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#26 0x00007ffff75bbd1e in wxDialog::ShowModal() () at /lib/x86_64-linux-gnu/libwx_gtk3u_core-3.0.so.0
#27 0x000055555583b23b in TimedMessageBox::TimedMessageBox(wxWindow*, wxString const&, wxString const&, long, int, wxPoint const&)
    (this=0x7fffffffcf90, parent=0x0, message=..., caption=..., style=5, timeout_sec=<optimized out>, pos=...) at /usr/local/src/OpenCPN/src/chart1.cpp:11422
#28 0x000055555583b2f0 in OCPNMessageBox(wxWindow*, wxString const&, wxString const&, int, int, int, int)
    (parent=parent@entry=0x0, message=..., caption=..., style=style@entry=5, timeout_sec=timeout_sec@entry=-1, x=x@entry=-1, y=-1) at /usr/include/wx-3.0/wx/gdicmn.h:546
#29 0x0000555555b11f54 in TCMgr::LoadDataSources(wxArrayString&) (this=0x555557c70780, sources=...) at /usr/include/wx-3.0/wx/translation.h:249
#30 0x0000555555830257 in MyFrame::LoadHarmonics() (this=<optimized out>) at /usr/local/src/OpenCPN/src/chart1.cpp:9339
#31 0x00005555558bd332 in ChartCanvas::ShowTides(bool) (this=0x5555565087a0, bShow=<optimized out>) at /usr/local/src/OpenCPN/src/chcanv.cpp:1212
#32 0x00005555558bddba in ChartCanvas::ApplyCanvasConfig(canvasConfig*) (this=0x5555565087a0, pcc=0x555556351050) at /usr/local/src/OpenCPN/src/chcanv.cpp:1101
#33 0x0000555555840b35 in MyFrame::CreateCanvasLayout(bool) (this=0x555556380c40, b_useStoredSize=<optimized out>) at /usr/include/wx-3.0/wx/dynarray.h:837
#34 0x00005555558506a6 in MyApp::OnInit() (this=<optimized out>) at /usr/local/src/OpenCPN/src/chart1.cpp:2301
#35 0x00007ffff7106a92 in wxEntry(int&, wchar_t**) () at /lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#36 0x000055555577a1c8 in main(int, char**) (argc=<optimized out>, argv=<optimized out>) at /usr/local/src/OpenCPN/src/chart1.cpp:1083

leamas, 2019-12-17:
OK, looks like some graphics stuff. I'm not in this business, and cannot help.

Have you done a git bisect before? That would probably be extremely useful here.

pcav, 2019-12-17:
Sorry, I don't have much time to debug this right now.
Thanks for your interest.

leamas, 2019-12-17:
hm, again (I should do something else!)

Line piano.cpp line 322-324 reads
 ```
wxBitmap *bitmaps[] = {m_pInVizIconBmp, m_pTmercIconBmp, m_pSkewIconBmp, m_pPolyIconBmp};

    for(unsigned int i = 0; i < (sizeof bitmaps) / (sizeof *bitmaps); i++) {
```
What's wrong with this? Corrupted stack?

leamas, 2019-12-18:
@bdbcat: This is far above my paygrade...

I can reproduce the same crash on Fedora 30.  Git bisect result: 
```
55e6cb2b4859dcab00ff64785e6da691c97e936 is the first bad commit
commit 755e6cb2b4859dcab00ff64785e6da691c97e936
Author: dsr <bdbcat@yahoo.com>
Date:   Mon Dec 2 20:41:35 2019 -0500

    Adapt core rendering engines to GLES2 on Android
```

Bisect log: 

```
git bisect start
# bad: [ac47faf7b79db24b5680840ad4f9db948e3bc2bb] Merge pull request #1528 from leamas/master
git bisect bad ac47faf7b79db24b5680840ad4f9db948e3bc2bb
# good: [48a72a0b05be7f405c8bc9d39b0c24ead955b38c] Draw route points if they are visible in GL mode same as non GL
git bisect good 48a72a0b05be7f405c8bc9d39b0c24ead955b38c
# bad: [e1bfe73355e98c13b01276c9fe09f8028f811433] msvc: Add missing algorithm header (msvc FTBS).
git bisect bad e1bfe73355e98c13b01276c9fe09f8028f811433
# good: [ce8d6f9e7bc47f39fd68245bcaa14bd7d28ad4b1] Merge pull request #1471 from leamas/mingw-fix
git bisect good ce8d6f9e7bc47f39fd68245bcaa14bd7d28ad4b1
# bad: [ac376a82af34105fc7e022d737caf6e17ea573f8] Update WMM plugin build rules for Android
git bisect bad ac376a82af34105fc7e022d737caf6e17ea573f8
# bad: [07dd82c9e73f823b00174d1c277adcd51faf3dba] Merge pull request #1418 from caesar/menu-shortcuts
git bisect bad 07dd82c9e73f823b00174d1c277adcd51faf3dba
# bad: [9db70700aa33ad19f08b8c2b8c1304e345b1b019] Update Platform specific module and methods.
git bisect bad 9db70700aa33ad19f08b8c2b8c1304e345b1b019
# good: [ad306e3ba46a35656c095d09ad13deeea1425919] Update build rules for Android
git bisect good ad306e3ba46a35656c095d09ad13deeea1425919
# bad: [c5563cc18ec878841d8df240a5db04ff7b020d63] Update Chart database access methods
git bisect bad c5563cc18ec878841d8df240a5db04ff7b020d63
# bad: [755e6cb2b4859dcab00ff64785e6da691c97e936] Adapt core rendering engines to GLES2 on Android
git bisect bad 755e6cb2b4859dcab00ff64785e6da691c97e936
# first bad commit: [755e6cb2b4859dcab00ff64785e6da691c97e936] Adapt core rendering engines to GLES2 on Android

```

leamas, 2019-12-18:
This is pretty bad, blocks all testing on current master...

pcav, 2019-12-18:
Please let me know, I'm available to rebuild once presumably fixed.
Thanks.

bdbcat, 2019-12-18:
I pushed a possible fix, just simplifying the needlessly obtuse code.  The failure must be related to compiler optimization?
Dave

pcav, 2019-12-18:
Rebuilt, still segfaults

pcav, 2019-12-18:
```
#0  0x0000555555a12dbe in Piano::BuildGLTexture() (this=0x555555fc1d20)
    at /usr/local/src/OpenCPN/src/piano.cpp:324
#1  0x0000555555a130f5 in Piano::DrawGL(int) (this=this@entry=0x555555fc1d20, off=-17)
    at /usr/local/src/OpenCPN/src/piano.cpp:347
#2  0x0000555555b439b4 in glChartCanvas::DrawChartBar(ocpnDC&) (this=this@entry=0x555556437cf0, dc=...)
    at /usr/local/src/OpenCPN/include/chcanv.h:269
#3  0x0000555555b58583 in glChartCanvas::Render() (this=0x555556437cf0)
    at /usr/local/src/OpenCPN/src/glChartCanvas.cpp:4819
#4  0x0000555555b59036 in glChartCanvas::OnPaint(wxPaintEvent&) (this=0x555556437cf0, event=...)
    at /usr/local/src/OpenCPN/src/glChartCanvas.cpp:1663
#5  0x00007ffff71b5231 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) () at /lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#6  0x00007ffff71b5323 in wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*) ()
    at /lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#7  0x00007ffff71b566c in wxEvtHandler::TryHereOnly(wxEvent&) ()
    at /lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#8  0x00007ffff71b56f3 in wxEvtHandler::ProcessEventLocally(wxEvent&) ()
    at /lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#9  0x00007ffff71b5791 in wxEvtHandler::ProcessEvent(wxEvent&) ()
    at /lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#10 0x00007ffff71b5527 in wxEvtHandler::SafelyProcessEvent(wxEvent&) ()
    at /lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#11 0x00007ffff753e43d in wxWindow::GTKSendPaintEvents(_cairo*) ()
    at /lib/x86_64-linux-gnu/libwx_gtk3u_core-3.0.so.0
#12 0x00007ffff7cd162c in wxGLCanvas::OnInternalIdle() ()
    at /lib/x86_64-linux-gnu/libwx_gtk3u_gl-3.0.so.0
#13 0x00007ffff76ec6f3 in wxWindowBase::SendIdleEvents(wxIdleEvent&) ()
    at /lib/x86_64-linux-gnu/libwx_gtk3u_core-3.0.so.0
#14 0x00007ffff76ec728 in wxWindowBase::SendIdleEvents(wxIdleEvent&) ()
    at /lib/x86_64-linux-gnu/libwx_gtk3u_core-3.0.so.0
#15 0x00007ffff76ec728 in wxWindowBase::SendIdleEvents(wxIdleEvent&) ()
    at /lib/x86_64-linux-gnu/libwx_gtk3u_core-3.0.so.0
#16 0x00007ffff7587b9f in wxFrame::SendIdleEvents(wxIdleEvent&) ()
    at /lib/x86_64-linux-gnu/libwx_gtk3u_core-3.0.so.0
#17 0x00007ffff75c786d in wxAppBase::ProcessIdle() () at /lib/x86_64-linux-gnu/libwx_gtk3u_core-3.0.so.0
#18 0x00007ffff74f8385 in wxApp::DoIdle() () at /lib/x86_64-linux-gnu/libwx_gtk3u_core-3.0.so.0
#19 0x00007ffff74f8473 in  () at /lib/x86_64-linux-gnu/libwx_gtk3u_core-3.0.so.0
#20 0x00007ffff517dd7e in g_main_context_dispatch () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#21 0x00007ffff517e130 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#22 0x00007ffff517e403 in g_main_loop_run () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#23 0x00007ffff5680705 in gtk_main () at /lib/x86_64-linux-gnu/libgtk-3.so.0
#24 0x00007ffff7515265 in wxGUIEventLoop::DoRun() () at /lib/x86_64-linux-gnu/libwx_gtk3u_core-3.0.so.0
#25 0x00007ffff707499d in wxEventLoopBase::Run() () at /lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#26 0x00007ffff757cd1e in wxDialog::ShowModal() () at /lib/x86_64-linux-gnu/libwx_gtk3u_core-3.0.so.0
#27 0x0000555555816d2b in TimedMessageBox::TimedMessageBox(wxWindow*, wxString const&, wxString const&, long, int, wxPoint const&)
    (this=0x7fffffffcf90, parent=0x0, message=..., caption=..., style=5, timeout_sec=<optimized out>, pos=...) at /usr/local/src/OpenCPN/src/chart1.cpp:11422

```

pcav, 2019-12-18:
```
#28 0x0000555555816de0 in OCPNMessageBox(wxWindow*, wxString const&, wxString const&, int, int, int, int) (parent=parent@entry=0x0, message=..., caption=..., style=style@entry=5, timeout_sec=timeout_sec@entry=-1, x=x@entry=-1, y=-1) at /usr/include/wx-3.0/wx/gdicmn.h:546
#29 0x0000555555af1144 in TCMgr::LoadDataSources(wxArrayString&) (this=0x555557b9d330, sources=...) at /usr/include/wx-3.0/wx/translation.h:249
#30 0x000055555580bd47 in MyFrame::LoadHarmonics() (this=<optimized out>) at /usr/local/src/OpenCPN/src/chart1.cpp:9339
#31 0x0000555555898e22 in ChartCanvas::ShowTides(bool) (this=0x555556435070, bShow=<optimized out>) at /usr/local/src/OpenCPN/src/chcanv.cpp:1212
#32 0x00005555558998aa in ChartCanvas::ApplyCanvasConfig(canvasConfig*) (this=0x555556435070, pcc=0x555556280530) at /usr/local/src/OpenCPN/src/chcanv.cpp:1101
#33 0x000055555581c625 in MyFrame::CreateCanvasLayout(bool) (this=0x5555563acd30, b_useStoredSize=<optimized out>) at /usr/include/wx-3.0/wx/dynarray.h:837
#34 0x000055555582c196 in MyApp::OnInit() (this=<optimized out>) at /usr/local/src/OpenCPN/src/chart1.cpp:2301
#35 0x00007ffff70c5a92 in wxEntry(int&, wchar_t**) () at /lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#36 0x000055555575a838 in main(int, char**) (argc=<optimized out>, argv=<optimized out>) at /usr/local/src/OpenCPN/src/chart1.cpp:1083

```

bdbcat, 2019-12-18:
One more commit, fixing this time?
Error is only triggered when there are no tide/current sources available, so the popup message interferes with the pianobar initialization.
Dave

pcav, 2019-12-18:
Definitely I have no luck (how do I miss a deb, even more!).
Tried compiling v5.0.0, but:
```
...
/usr/bin/ld: CMakeFiles/opencpn.dir/src/glChartCanvas.cpp.o: in function `wxGLCanvas::~wxGLCanvas()':
/usr/include/wx-3.0/wx/gtk/glcanvas.h:20: undefined reference to `wxGLCanvasX11::~wxGLCanvasX11()'
/usr/bin/ld: /usr/include/wx-3.0/wx/gtk/glcanvas.h:20: undefined reference to `vtable for wxGLCanvas'
/usr/bin/ld: /usr/include/wx-3.0/wx/gtk/glcanvas.h:20: undefined reference to `wxGLCanvasX11::~wxGLCanvasX11()'
/usr/bin/ld: CMakeFiles/opencpn.dir/src/glChartCanvas.cpp.o:(.data.rel.ro._ZTI12glTestCanvas[_ZTI12glTestCanvas]+0x10): undefined reference to `typeinfo for wxGLCanvas'
/usr/bin/ld: CMakeFiles/opencpn.dir/src/glChartCanvas.cpp.o:(.data.rel.ro._ZTI13glChartCanvas[_ZTI13glChartCanvas]+0x10): undefined reference to `typeinfo for wxGLCanvas'
/usr/bin/ld: CMakeFiles/opencpn.dir/src/glChartCanvas.cpp.o:(.data.rel.ro._ZTV12glTestCanvas[_ZTV12glTestCanvas]+0x10): undefined reference to `wxGLCanvas::GetClassInfo() const'
/usr/bin/ld: CMakeFiles/opencpn.dir/src/glChartCanvas.cpp.o:(.data.rel.ro._ZTV12glTestCanvas[_ZTV12glTestCanvas]+0x1f8): undefined reference to `wxGLCanvasX11::IsShownOnScreen() const'
/usr/bin/ld: CMakeFiles/opencpn.dir/src/glChartCanvas.cpp.o:(.data.rel.ro._ZTV12glTestCanvas[_ZTV12glTestCanvas]+0x338): undefined reference to `wxGLCanvas::SetBackgroundStyle(wxBackgroundStyle)'
/usr/bin/ld: CMakeFiles/opencpn.dir/src/glChartCanvas.cpp.o:(.data.rel.ro._ZTV12glTestCanvas[_ZTV12glTestCanvas]+0x460): undefined reference to `wxGLCanvas::OnInternalIdle()'
/usr/bin/ld: CMakeFiles/opencpn.dir/src/glChartCanvas.cpp.o:(.data.rel.ro._ZTV12glTestCanvas[_ZTV12glTestCanvas]+0x620): undefined reference to `wxGLCanvasX11::SwapBuffers()'
/usr/bin/ld: CMakeFiles/opencpn.dir/src/glChartCanvas.cpp.o:(.data.rel.ro._ZTV12glTestCanvas[_ZTV12glTestCanvas]+0x638): undefined reference to `wxGLCanvas::GetXWindow() const'
/usr/bin/ld: CMakeFiles/opencpn.dir/src/glChartCanvas.cpp.o:(.data.rel.ro._ZTV13glChartCanvas[_ZTV13glChartCanvas]+0x10): undefined reference to `wxGLCanvas::GetClassInfo() const'
/usr/bin/ld: CMakeFiles/opencpn.dir/src/glChartCanvas.cpp.o:(.data.rel.ro._ZTV13glChartCanvas[_ZTV13glChartCanvas]+0x1f8): undefined reference to `wxGLCanvasX11::IsShownOnScreen() const'
/usr/bin/ld: CMakeFiles/opencpn.dir/src/glChartCanvas.cpp.o:(.data.rel.ro._ZTV13glChartCanvas[_ZTV13glChartCanvas]+0x338): undefined reference to `wxGLCanvas::SetBackgroundStyle(wxBackgroundStyle)'
/usr/bin/ld: CMakeFiles/opencpn.dir/src/glChartCanvas.cpp.o:(.data.rel.ro._ZTV13glChartCanvas[_ZTV13glChartCanvas]+0x460): undefined reference to `wxGLCanvas::OnInternalIdle()'
/usr/bin/ld: CMakeFiles/opencpn.dir/src/glChartCanvas.cpp.o:(.data.rel.ro._ZTV13glChartCanvas[_ZTV13glChartCanvas]+0x620): undefined reference to `wxGLCanvasX11::SwapBuffers()'
/usr/bin/ld: CMakeFiles/opencpn.dir/src/glChartCanvas.cpp.o:(.data.rel.ro._ZTV13glChartCanvas[_ZTV13glChartCanvas]+0x638): undefined reference to `wxGLCanvas::GetXWindow() const'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/opencpn.dir/build.make:1732: opencpn] Error 1
make[2]: uscita dalla directory "/usr/local/src/OpenCPN/build_5"
make[1]: *** [CMakeFiles/Makefile2:470: CMakeFiles/opencpn.dir/all] Error 2
make[1]: uscita dalla directory "/usr/local/src/OpenCPN/build_5"
make: *** [Makefile:155: all] Error 2

```

bdbcat, 2019-12-18:
pcav...
Can you not run a generic Xenial or Trusty build of OCPN on sid?
Dave

pcav, 2019-12-18:
Mixed repos==trouble :)
I'm not talking about me, I can compile, even if with troubles, but I think a proper official package will help the spread of OCPN.
Anyway, now r8nning master, thanks!

leamas, 2019-12-18:
The official debian package is accepted, but still at 4.8.8. I'm about to update to 5.0.0, will look into it later.

leamas, 2019-12-18:
I can now compile and run  on Fedora 30 without the previous crash.  Instead, here is an assert message box at startup:

    ./src/unix/glx11.cpp(194): assert "xid" failed in SetCurrent(): window must be shown
  
Annoying, but not blocking.  Using a more elobarated cmake invocation:

      cmake \
        -DCMAKE_INSTALL_PREFIX=/usr 
        -DOCPN_USE_BUNDLED_LIBS=OFF 
        -DOCPN_USE_NEWSERIAL=OFF 
        -DCMAKE_CXX_FLAGS="-DNDEBUG -g -O2" \
        ..

and it looks OK (that is, starts without crash or assert messages.)

pcav, 2019-12-18:
I missed this, thanks a lot!

pcav, 2019-12-18:
Sorry to hijack the thread: AFAICT there is no make uninstall, am I wrong?

leamas, 2019-12-19:
No, you are not. There should be....

pcav, 2019-12-19:
Thanks. For the record, I opened ticket #1531 not to forget.
Thanks for confirming.

leamas, 2019-12-19:
Great. Does this mean that  you can now run this without crash i. e., that this bug could be closed?

pcav, 2019-12-19:
Yes, thanks!

closed 2019-12-19 by pcav