OpenCPN/OpenCPN #1530: Plugin installer UI. Plugin version versus required API

Hakansv, 2019-12-18:
Now I eventually turned into the installer after initial large update general test,
First a compliment for the [Project installer wiki](https://github.com/leamas/OpenCPN/wiki) It explains and settle a lot. Next is to dig into [Plugin GitHub](https://github.com/OpenCPN/plugins)

A first UI issue:
Here is a fear for user confusion. See shot. The API designation can easily be interpreted as the plugin version and question why should I downgrade from 1.16.0 to 1.0.17? 
I know the extended view for each plugin it's more clear but I do think we still need to print the version on the collapsed header. 
Something like: Squiddio version 1.0.17. Requires API 116 (1.16)

[image]

leamas, 2019-12-19:
Indeed. See also #1519.

leamas, 2019-12-22:
hm... this is not really about the api. To make things interesting, squiddio has a version level which matches the API level.

The core issue is a mismatch between the XML metadata (reported in the download page) and  what the plugin reports when loaded (in the active plugins page). Not quite sure how to handle it. 

Thoughts?

rgleason, 2019-12-22:
This situation is created by the new pi system, and comparison to the older pi. Maybe the difference could be identified somehow as distinct? If that makes any sense?

leamas, 2019-12-23:
I think the sort term solution is

  - Display the metadata version in download window
  - Only display the version reported when loading the plugin in the plugin window
  - Log a warning if they don't match, nothing more than that. Any user info like 
    *metadata version does not match loaded plugin version* would just add confusion.

rgleason, 2019-12-23:
Why not use
Cmake Assigned Version:
Metadata Version:

leamas, 2019-12-24:
I think this is too much information in an UI, it will just confuse users. Do you really think this info will be valuable in the UI?

I see users thinking *WTF is "Cmake assigned version"*? Just an example...

rgleason, 2019-12-24:
Well, the decision was to utilize a new unique automatically generated "Metadata version number" rather than the "Cmake assigned version number", which creates the confusion.

So now my question is WTF is "Metadata Version"  too? 

You could just pretend that there were no version numbers before PI Installer and just show the automatically generated metadata number.   Then perhaps the question about version numbers will never arise, but there is a problem I had not thought of.

What happens if a plugin author chooses not to use Cloudsmith to deploy, and since cloudsmith is the generator of the number, that plugin will not have a "metadata version".?

Why don't we just use the normal CMake assigned version number always?

taruti, 2019-12-25:
Personally I think that plugin API version is a technical detail that should not be important to end users. Actual plugin author provided versions are more interesting.

Typically I want to know:
"Which version do I have and what is the newest version?"

rgleason, 2019-12-25:
However, the API number is essential to understanding compatibility, 
1. Every time there is a wxWidgets version change, backwards compatibility is lost.
2. It is unknown when the wxWidgets version will change.

Thus the PI Version number gives not a clue about what OpenCPN versions it is compatible with.
The API Number is necessary somewhere, even if it is in the "About" of the Plugin.

leamas, 2019-12-26:
I still think that the proper fix to this bug is as described in  https://github.com/OpenCPN/OpenCPN/issues/1530#issuecomment-568588928. 

The question about how visible the plugin API should be is IMHO a sidetrack.

rgleason, 2019-12-26:
Yes the visibility of Plugin API version is a sidetrack to this topic, but since it was raised, it had to be explained. And since we have agreed that it should be shown somewhere, that matter is settled.

rgleason, 2019-12-26:
I think this is very perverse, to only show metadata version when the plugin is being loaded. Why?
Why isn't the user going to be able to see the cmake version the pi author coded?
Why is the metadata number that is automatically generated by cloudsmith the only concern when installing? Why is it not necessary to know for the user, later, if they are having trouble with the plugin? It becomes a second identifier.

rgleason, 2019-12-26:
I wish others would weigh in here.

leamas, 2019-12-26:
> Why isn't the user going to be able to see the cmake version the pi author coded?

My proposal is (again):

  - For plugins in the download list: Show only available version from metadat.
  - For  loaded plugins: only show the compiled-in version, don't bother about metadata.

This is just a rewording of https://github.com/OpenCPN/OpenCPN/issues/1530#issuecomment-568588928.

Hakansv, 2019-12-26:
OK [image]

rgleason, 2019-12-26:
OK sounds good. You'll have the compiled version to refer to.

leamas, 2019-12-26:
hmpf... "blushes"

This has been a long and partly confused discussion. But, we are going in circles... To a large extent my bad. Sorry.

Håkan's screenshots  basically shows that everything is as it should. Be aware of  that Squiddio's version level 1.16/1.17 matches  our current API level, a sheer coincidence.

Now, the new plugin manager windows shows 1.17 (from metadata). The old plugin window show 1.16, from the compiled plugin version. 

This is as it should, besides the fact that the XML metadata doesn't match the actual plugin version. I don't really see what could be done here (besides updating the metadata). Badly maintained metadata will happen, and the consequence is what we see.

An evil effect of all this is that the  download will be marked as an update even after the update is done (since the metadata version is higher than the plugin version). The only remedy is to fix the metadata. Will do.

For now, I personally mark this as WONTFIX. If someone comes up with a reasonable alternative what to do UX-wise. I'm certainly interested.

Hakansv, 2019-12-26:
Now I don't understand. The suggestion above is good but nothing is so far changed. At least not on Github master.
The plugin loader shows recent plugin version as it should and as it did before.
The Options plugin tab shows needed API as before. Now when we have the Plugin Manager to compare to it would show the plugin version. (And maybe needed API as well.) 
For example here installed oeSENC_pi should read v 3.3.14 and oeRNC_pi v 1.0.4.

[image]

leamas, 2019-12-26:
Can we just forget this discussion... it shouldn't take so long to make a simple patch.

Mostly my bad. Sorry for the mess.

Hakansv, 2019-12-27:
If agreed I could PR a patch changing from API to version major.minor like the shot.
Version.patch is omitted for plugins up to API 116 since it's not supported. 
Requested API is printed while selected as before. See oeSENC in the pict below.
OK?

[image]

leamas, 2019-12-27:
Already did that -- https://github.com/leamas/OpenCPN/commit/124e40cfec1eef25ce21adcabd5c06b607e16132. But thanks

Hakansv, 2019-12-27:
With all respect but that commit didn't changed to show version instead of API. It just omitted the required API in selected mode, see pict.
My proposal to achieve acc. to my picture above.
https://github.com/Hakansv/OpenCPN/commit/23481fc270f6f59548be0d9103a28eac21d4689d
If nothing else I'll PR this tonight?
[image]

leamas, 2019-12-27:
IT's fine, and makes things better.  

One nit-pick: the title. This is about not displaying the undefined patch level :)

rgleason, 2019-12-27:
One nit pick. Why not display it? Otherwise troubleshooting will be more difficult!
Why is this so difficult? Always butting heads?

leamas, 2019-12-27:
Not display what? Title? Plugin API level? Undefined patch version?

Hakansv, 2019-12-27:
Rick..
Just for info:
The plugin patch is not known for OCPN pluginmanager before API 117. So instead of print "x.x.0" that might not be true we print x.x only.
The plugin itself would know about its patch and can print that in preferences but there is no method for the plugin to transfer it to O before API 117.

rgleason, 2019-12-27:
Ok, I will be clear about this.
For every plugin, every preferably every OS version, the user should be able to see the full version number (major.minor.patch) and the Plugin API number. 
Having these available and shown automatically with every plugin, in some way, without requiring plugin dependent programming, should be a requirement, in my opinion.

rgleason, 2019-12-27:
If the patch number does not exist in earlier versions, of course it would not be shown.

rgleason, 2019-12-27:
Thanks Hakan and Alec.

closed 2019-12-29 by bdbcat

Hakansv, 2019-12-29:
Strange? The issue is closed but neither solved nor explained not relevant. 
The referred patch didn't exchanged plugin API-version by plugin-version.  Instead it omitted needed API version from "selected" mode.

leamas, 2019-12-29:
My bad, sorry. I was too quick  to add a *Closes: #1530* to my patch -- this closes the bug automatically when Dave merges it.

Just to make things clear: I think your patch on top of mine is the proper solution.  Are you used to rebasing patches?

PR #1593 (2019-12-29): pluginmanager: Don't display required plugin API level.
Closes: #1530
commit 124e40cf (2019-12-29): pluginmanager: Don't display required plugin API level.

Closes: #1530