OpenCPN/OpenCPN #1534: RPi 7" screen size failures.

Hakansv, 2019-12-19:
This is not urgent but maybe changeling for a Linux equilibrist.
What's noted here is about the "Official" RPi 7" screen. From resent forum posts it's clear more and more OCPN system have started to use that in some respect odd screen. I've used it for a year or so on a RPi3b+ Raspbian/Debian stretch.
This note is nothing new for the actual master code but more a collection of observations for a possible solution by our Linux experts.
The screen is a touch screen but two finger zoom and long click is not detected by OCPN, and others. But sometimes actually detected by for example Chromium browser.  There are obviously a install available as a work around for this so it's only mentioned for info.
One disturbing and rather common fact is though that many dialog windows are not detecting the screen size so buttons in the bottom part are often hidden and no side slider is present. Thus can a option change not be activated since the OK button is unreachable. The only way to close the window is to use the upper right cross. This appearance is not isolated to OCPN core but also plugins suffers from this.
One common example is the first shot below from the new plugin installer. Other shots show how the Dashboard is not fully viewed and no slider to move it down. The WP preferences window don't shove the buttons and I can't move the window upwards since the whole screen is already covered.
If it's wxWidgets that can't evaluate the screen limits or anything else I don't understand. I've copied relevant opencpn.log entries herein but one thing I found strange is this:
"MyFrame...size(800, 460)  position(-2, 18)". A negative x-position? Can that fool something? As seen from the log I also had to manually adjust the horizontal screen width.

If anyone can find a way to convince wxWidgets about the actual screen size it's fine and that can be spread to plugin developers as well. Solve the plugin Catalog installer window for example. I've the screen and can test. I can also apply and test different code if there are thoughts about possible solutions.

[image]

[image]

[image]

```
20:33:15 CET:  ------- Starting OpenCPN -------
20:33:15 CET: 5.0.0
20:33:17 CET: wxWidgets version: wxWidgets 3.0.2 Linux 32 bit wxGTK
20:33:17 CET: OS Version reports as:  4.19
20:33:17 CET: MemoryStatus:  mem_total: 926 mb,  mem_initial: 0 mb
....
20:33:17 CET: Detected display size (horizontal): 212 mm
20:33:17 CET: Display size (horizontal) config override: 153 mm
....
20:33:17 CET: Creating MyFrame...size(800, 460)  position(-2, 18)
20:33:18 CET: Creating glChartCanvas
...
20:33:18 CET: Creating glChartCanvas
20:33:18 CET: Metrics:  m_display_size_mm: 153     wxDisplaySize:  800:480   
20:33:18 CET: OpenGL-> Renderer String: Gallium 0.4 on llvmpipe (LLVM 3.9, 128 bits)
20:33:18 CET: OpenGL-> Version reported:  3.0 Mesa 13.0.6
20:33:18 CET: OpenGL-> Texture rectangle format: de1
20:33:18 CET: OpenGL-> Using Vertexbuffer Objects
20:33:18 CET: OpenGL-> Using Framebuffer Objects
20:33:18 CET: OpenGL-> Using FBO Stencil buffer
20:33:18 CET: OpenGL-> Using Stencil buffer clipping
20:33:18 CET: OpenGL-> Using Scissor Clipping
20:33:18 CET: OpenGL-> Not Using compression
20:33:18 CET: OpenGL-> Minimum cartographic line width:  1.0
20:33:18 CET: OpenGL-> Minimum symbol line width:  1.0
....
20:33:18 CET: ChartDB Cache policy:  Application target is 462 MBytes
20:33:18 CET: Loading chart db version: V018
....
20:33:28 CET: DPMM: 5.22876   ScaleFactorExp: 1
....
20:33:46 CET: oesenc_pi:  Calculated pix/mm = 5.22876
20:33:46 CET: oeSENC_PI:  OpenGL-> Minimum cartographic line width:  1.0
20:33:46 CET: oeSENC_PI RasterSymbols texture: 141
```

bdbcat, 2020-01-08:
@Hakansv 
MarkInfo dialog should be corrected by f5a6a7ea00d8438723ce3f0efe94d557ea1cd63f.
Dashboard needs a little longer.
Lets leave PluginManager until the UI stabilizes.

Any more problems noticed on rPI 7" display?

Hakansv, 2020-01-08:
@bdbcat 
Markinfo dialog is better. The buttons are shown all time. Good.
Bottom controls in the Extended tab are though not shown for Dialog fonts >6. Using font "8" it's like shot 2. Could a scroll bar be added to the Extended tab as on the Basic tab?
No more critical RPi 7" display for the moment.

[image]
[image]

bdbcat, 2020-01-09:
Extended properties layout fixed by 8fd587606c62d14d0da0ebbdd2a19c2ddd84a763

Hakansv, 2020-01-09:
This is complex. Equal function but two versions of code.
While opening WP properties for a solely WP there's no scroll bar and lower details are hidden. Pic 1.
When opening WP properties for a mark that belongs to a route there's a scroll bar, Pic 2. 

[image]

[image]

bdbcat, 2020-01-09:
@Hakansv 
The missing items on isolated WP are truly missing, as designed  The Arrival radius, planned speed, etc are only related to a route that the WP belongs to, if any.
So I think this is OK.

Hakansv, 2020-01-09:
@bdbcat 
Yes, you're as usual correct. 
Thanks

closed 2020-01-09 by Hakansv