OpenCPN/OpenCPN #1535: Linux plugin path finder

transmitterdan, 2019-12-19:
The function `void PluginPaths::initLinuxPaths()` does something strange.

When it calls `g_Platform->GetPluginDir()` it then tries to append "/opencpn" which at least on the Linux I have doesn't exist.  On my systems `g_Platform->GetPluginDir()` returns something ending in "/lib/opencpn" so appending another "/opencpn" to that creates a path that doesn't exist.

leamas, 2019-12-19:
Ack. This is a corner-case, but still a bug.  I'll look into it, should be an easy fix. 

Thanks for reporting!

leamas, 2019-12-19:
Is it so that you possibly have a testcase since you discovered this? If so, I would appreciate if you had some time to test attached patch. It *looks* ok, but...

[0001-PluginPaths-Don-t-add-extra-opencpn-to-path-1535.patch.gz](https://github.com/OpenCPN/OpenCPN/files/3985528/0001-PluginPaths-Don-t-add-extra-opencpn-to-path-1535.patch.gz)

transmitterdan, 2019-12-20:
That is pretty much the same solution I came up with.  However, the test case involves compiling and installing to a local folder.  Thus I think it is important to put the folder at the front of the list of folders rather than pushback to the end.  In my test case I want it to find the test install folder first before it looks in the "usual" places.  Does this make sense?

leamas, 2019-12-20:
hm... first: this is not just about m_libdirs, it's also about m_datadirs some lines below. We should treat them the same way.

Secondly: the platform_dir is basically hardcoded. If we put this first, it will shadow anything installed in dirs defined in the environment. That's a rather big drawback.

Can you run your testcase by fiddling with `OPENCPN_PLUGIN_DIRS` and `XDG_DATA_DIRS `in the environment instead? That would resolve this particular issue, and the `platform_dir` could be made the last resort (which I think is the Right Thing).

leamas, 2019-12-25:
Ping? Is this possibly a low hanging fruit to fix?

leamas, 2019-12-25:
I'll submit my fix. If/when Dave merges it he will close this bug. Please file a new one if you cannot run your testcase using   `OPENCPN_PLUGIN_DIRS` and `XDG_DATA_DIRS` .

transmitterdan, 2019-12-25:
The use case here is building a "test" version of OpenCPN using an Cmake install prefix directory other than /usr/local. In such a case we want the plugin loader to get the plugins from the prefix directory path and not the normal /usr/local or /usr/bin paths. This is a pretty standard case in Linux. Should we have to resort to environment variables when we already have a Cmake install path defined via the prefix option?

I realize you are asking developers to fiddle with environment variables which is a small group of users. I think that is so users can use environment variables to control search paths for plugins.  But I imagine almost no users will want to do that.

Users are trained now to do one of two things.

1) Run OpenCPN as it was installed in the default place with plugins found in the "default" place.
2) Run OpenCPN in portable mode after installing it pretty much anywhere they like.

They don't know how do much else in my experience. Nor should they.

I admit I don't fully understand the complexity of the new plugin loader system.  So maybe it has some great benefit and I just haven't found the use case yet.  I mostly work on the core and 4 basic plugins.  The debugging task is already a huge pain in Windows because there is no automatic installation of plugins when building debug versions.  In Linux it seems like debugging plugins is becoming a pain too.

leamas, 2019-12-25:
> Should we have to resort to environment variables when we already have a Cmake install path defined via the prefix option?

Certainly not. This is #1535, and there is a PR 1582 which is supposed to fix this.

> I realize you are asking developers to fiddle with environment variables 

For normal end users: definitely not. Let's agree on that.

Environment variables are for packagers, developers and  to make sure the application is a well-behaved linux citizen (important for e. g., flatpak). But not for  end users, and not for simple developing like above.

>  In Linux it seems like debugging plugins is becoming a pain too.

What I have  done:
 
   -  install buggy plugin using installer. 
   -  Rebuild a debug version of the .so file (cmake, make)
   -  Copy the built  so-file to it's place; in linux to ~/.local/lib/opencpn
   -  Run opencpn, hopefully reproducing  the same bug
   -  Start debugging. 

It works as long it's not a problem with external  data files etc.

rgleason, 2019-12-25:
This seems like a good approach, should it be put in some appropriate document about debugging?
Similar for windows?

leamas, 2019-12-25:
Let's wait utnil more people than me finds this workflow useful. But similar on Windows, yes.

transmitterdan, 2019-12-26:
When we build and install the core and 4 pre-installed plugins we need to have all the data files go along for the ride.  When we use Cmake's install folder prefix= feature this works.  But the prefix folder has to come first in the list of paths else O will use a plugin from some other path.  As a developer I want to use the prefix feature of Cmake to avoid polluting the "standard" locations for O and its standard plugins.  A normal invocation of O will not start the prefix version nor will any of the prefix plugins be found.  But when I manually (or from an IDE) launch the prefix= version I expect the prefix= plugins to be automatically found without any fiddling or extra copying.  I believe this is the "Linux way".

The Windows workflow for developers plain sucks IMO. The Windows section of CMakeLists.txt needs an install feature like Cmake provides for Linux. This could be accomplished via a post-build event for O and each plugin.  But so far no developer has had the time (or impetus) to do this.  I gave up trying to coerce Cmake into doing what was needed and made a Windows script to mimic the install process.  This simplified debugging but it isn't optimal.

Another point is now that Visual Studio fully supports remote debugging it is possible to use Visual Studio to debug O on pretty much any Linux like platform (except perhaps MacOS) regardless of the processor type.  It uses the target build system and remote access to gdb.  We should not ignore that Visual Studio is becoming a powerful IDE that fully embraces building and debugging large complex Linux programs.  I recently did an experiment where I built, installed and debugged the entire O system on a RPi from within the free version of Visual Studio.  It's slow but it works and I expect the speed to improve rapidly.  VS has powerful features beyond just break-point debugging.  I have never been that happy with the debuggers available on Windows or Linux but VS is getting close to optimal.

rgleason, 2019-12-26:
TDan, you are describing "remote debugging" with Visual Studio, which sounds very powerful, but I am sorry, what exactly is that?

closed 2019-12-26 by bdbcat

leamas, 2019-12-28:
>  But when I manually (or from an IDE) launch the prefix= version I expect the prefix= plugins to be automatically found without any fiddling or extra copying.

The installation directory, wherever it is, is always added to the search path. This was affected by #1535, but should now be fixed. However, it's added to the end.

If put it in the beginning, it will basically make it impossible to override the files in the install dir. I hesitate to do this, the environment variables are supposed to be the swiss army knife doing all sorts of things without limitations.

Have been thinking about using a shell wrapper for opencpn (already do in flatpak). It would, among other things,  make the necessary environment setup fiddling so much easier.

leamas, 2019-12-28:
@rgleason 
Actually, JS is late to the party ;) Supported by
 
  - [eclipse, since "long"](
https://help.eclipse.org/2019-12/index.jsp?topic=%2Forg.eclipse.jdt.doc.user%2Fconcepts%2Fcremdbug.htm)
  - [gdb](https://developers.redhat.com/blog/2015/04/28/remote-debugging-with-gdb/)
  - [VS](https://docs.microsoft.com/en-us/cpp/linux/deploy-run-and-debug-your-linux-project?view=vs-2019)

It's actually not that hard to use gdb directly, bypassing the IDE. You have the usual stuff available: breakpoints, inspect data, evaluate data expression etc. Contrary to IDE:s it is ligthweigth but, well, also light on features.

leamas, 2019-12-28:
@transmitterdan wrote:

> The Windows section of CMakeLists.txt needs an install feature like Cmake provides for Linux.

Indeed. And we need to support uninstall.

In the end, I think it's possible to create a _make install_ which basically does the same thing as the installer. For example on Linux, installs in _~/.local/{bin,lib,share}_. This would be a simplified workflow aimed for developers so they don't need to build, upload and download a plugin to test it.

transmitterdan, 2019-12-28:
> The installation directory, wherever it is, is always added to the search path. This was affected by #1535, but should now be fixed. However, it's added to the end.

Because it is added at the end, O finds old plugins before it finds the just built plugins.  This is not obvious and wastes time.  This issue only affects the 4 plugins packaged with O which are never installed by the plugin installer.  Other plugins will never be found in the prefix= install folder so it seems otherwise harmless to put that folder first in the search path.

leamas, 2019-12-28:
It's not that don't understand your problem and your line of reasoning. Also, putting install_dir  at front solves your problem, agreed.

But I still hesitate to make a change which breaks the basic promise to be able to have an arbitrary search path. It's hard to foresee all future usecases, but by keeping the dir at the end we keep the flexibility.

Just an example: what if a user installs a local plugin package in the same prefix? It will then override the path in ~/.local, and we definitely don't want that. 

It's basically a classic ease of use vs flexibility trade off. In this case, I lean to prioritizing the flexibility.

One possible solution is of course a cmake option which makes it possible to hardcode the search path, overriding the defaults. Would that satisfy your needs?

jongough, 2019-12-28:
https://github.com/OpenCPN/OpenCPN/issues/1535#issuecomment-568934009
Your suggestion:
# Install buggy plugin using installer.
# Rebuild a debug version of the .so file (cmake, make)
# Copy the built so-file to it's place; in linux to ~/.local/lib/opencpn
# Run opencpn, hopefully reproducing the same bug
# Start debugging.
Does not work in many instances. Working on the weather routing plugin when I tried this process the 'installed' code did not work, the 'built' code did. It actually requires the 'installed' code to be built and packaged with debugging information which will at least show you the frame stack and line numbers of each of the modules involved. This then helps track down the issue. But you still cannot single step through to see what is happening. rgleason has a version of the weather routing pi [https://github.com/rgleason/weather_routing_pi/tree/api-1.16](https://github.com/rgleason/weather_routing_pi/tree/api-1.16) that does the 'debug' build.

I raised this as an issue before these changes were merged. You will find some of them here:
https://github.com/OpenCPN/OpenCPN/pull/1457#issuecomment-562823495
https://github.com/OpenCPN/OpenCPN/pull/1457#issuecomment-562740561
etc.

The squiddio plugin that leamas modified still has crashes when using the install process. I am not sure where the fix for that is yet.

leamas, 2020-02-23:
@transmitterdan : What is this bug about these days?

leamas, 2020-02-23:
sorry, missed it is closed.   Please disregard.

PR #1582 (2019-12-26): PluginPaths: Don't add extra /opencpn to path (#1535).
Closes: #1535.
commit 8f9a11e3 (2019-12-26): PluginPaths: Don't add extra /opencpn to path (#1535).

Closes: #1535.