OpenCPN/OpenCPN #1536: linux: gtk2 -> gtk3 plugin ABI breakage

leamas, 2019-12-19:
As the last major distro Debian is now dropping the gtk2 support. This means that from now on we need to build opencpn on debian using gtk3. This works fine, but it breaks the binary ABI to old plugins.

I don't think it should be a major problem to port plugins to gtk3, the majority probably needs to be recompiled against gtk3 instead. However, they need to be recompiled.

I don't really know what all this means. Should we make sure that all plugins using the installer are gtk3? Should we announce that opencpn is moving to all plugin devs (is there even any channel for such messages)

I just see the problem, not the solutions...

bdbcat, 2019-12-20:
1.  There is never an instantaneous cut-over on these types of ABI breaking updates.  So we will need to come up with a way to support gtk2 and gtk3 in our distribution channels.
2.  It seems logical to require a gtk2 version of OCPN to only offer and install gtk2 plugins.  Same for gtk3.  In linux we can use an elf extraction library to determine the dependencies of a plugin dynamically.  Just more code, but this is an example of the situation the new plugin scheme is designed to fix for users, i.e. to actively prevent trying something that we know will not work and thus avoid a support event.

rgleason, 2019-12-20:
Would a list of plugins using gtk2 be a start?  If so I can make one.

leamas, 2019-12-20:
Hm... simple-minded packager approach (linux only!):

  - We declare that from next API 1.17 linux is gtk3.
  - We note that all existing 1.16 plugins are gtk2
  - We refuse to load any plugin <= 1.16.

This would simplify things for coders. But, important, doesn't it also create a simple view for users, acknowledging the simple, sad fact that we have a ABI breakage on linux?

leamas, 2019-12-20:
The more I think about it, this is more of a problem for the plugins. They will for some time need to support 1.16/gtk2 and 1.17/gtk3. Doable, but will probably require yet another ci build channel.

leamas, 2019-12-20:
> Would a list of plugins using gtk2 be a start? If so I can make one.

It would probably be easier to make a list of those *not* using gtk2. All plugins having some kind of UI are linked against gtk2, so that list will be really short.

bdbcat, 2019-12-20:
Alec...
We can't drop trusty.  Expecting casual users to upgrade their entire linux distro in order to run the most modern OCPN is not the right way.
We can agree to support trusty/xenial with gtk2 only, if that helps matters.
And for plugins, we will need the full matrix, understanding that some plugins will never be migrated to gtk3 as a practical matter.

bdbcat, 2019-12-20:
Also, assuming wxWidgets port to gtk3 is complete and viable, many plugins won't care about gtk2/3.

leamas, 2019-12-20:
> We can't drop trusty. Expecting casual users to upgrade their entire linux distro in order to run the most modern OCPN is not the right way.

Starting from another angle: Can we agree that next opencpn version, plugin API 1.17, will be gtk3 on linux?

leamas, 2019-12-20:
> Also, assuming wxWidgets port to gtk3 is complete and viable, 
 
We must assume so for the modern releases, it is supposed to be true in Debian 10+. Older releases is another thing....

> many plugins won't care about gtk2/3.

They must still decide to link against one of them, or provide multiple builds. The latter should not be anything more than cmake setup (and probably some patches) for the vast majority.

transmitterdan, 2019-12-20:
Pretty soon this hack:

```  # As of cmake 3.11.2, these libraries are missing in list despite that we
  # looked for them. This is a nasty fix which might fail miserably. Assumption:
  # All builds using GTK uses unicode and wxWidgets 3.0
  if (GTK3_FOUND)
    list(APPEND wxWidgets_LIBRARIES "-lwx_gtk3u_aui-3.0")
    if (OPENGL_FOUND)
      list(APPEND wxWidgets_LIBRARIES "-lwx_gtk3u_gl-3.0")
    endif ()
  endif ()
```

will begin to cause grief because wxWidgets is past 3.0 now.

leamas, 2019-12-20:
Indeed (I wrote that note...)

Still, isn't the best strategy here to just close our eyes and wait for the inevitable? In other words, let the tools have a chance to catch up so we just can drop the fix? Or?

leamas, 2019-12-20:
@bdbcat : Can we continue from https://github.com/OpenCPN/OpenCPN/issues/1536#issuecomment-568101501?

bdbcat, 2019-12-20:
Alec....
Would not that leave trusty and xenial without our modern plugin management?

leamas, 2019-12-20:
It depends on their gtk3 support. My gut feeling is that xenial is OK, trusty not. Checks needed, assuming this is how it is for now.

The other side is that we cannot build gtk2-based stuff on any major distro today.

It's not that I like this, believe me. But what we are facing is a main development line including gtk3 and the installer. If we sort this out, including some example plugins, we can then look into what we can backport to 1.16/gtk2. 

Perhaps it's also important to not let the small linux user group block the development for windows/MacOS.

And, BTW, we also have the MacOS ABI breakage looming: if the signing issues will be resolved, we still need to move to 64-bit.

I didn't say this was easy. "Confused"

leamas, 2019-12-20:
BTW: Good night... I'm on CET :)

bdbcat, 2019-12-21:
Alec...
I admit I do not know the internal details of our CI implementation.
But why can we not build gtk2 packages for trusty and xenial.  Or, why are we forced to gtk3.
I note in passing that the Ubuntu PPA bots can build to gtk2 today.

Here, I run everyday Kubuntu 16.04 (xenial).  Still in LTS support.
The latest CI build for xenial on cloudsmith repo installs and runs fine.  No sign of gtk3.
What critical detail am I missing?
Dave

leamas, 2019-12-21:
The pressure comes from Debian, which Ubuntu is built on. Current sid/Bullseye does not support gtk2. As a consequence, the upcoming official Debian 5.0.0 package is gtk3. This will hit Ubuntu 20.04. 

We don't have any builds for the latest Ubuntu LTS Bionic.

Now, we have two solutions for traditional Ubuntu desktops: flatpak and .deb packages. We havn't moved flatpak to gtk3, but probably should before it becomes widely used. Still, we can run run gtk2 for a long time inside flatpak due to it's bundling.

That was the background. In my simple-minded packaging mindset, an ABI is well-defined. So, my simple question is: at which point do we do the inevitable breakage? To have two different ABI called 1.17/gtk2 and 1.17/gtk3 is just creating a mess.

What if we say that 1.17 actually is gtk2 and postpone gtk3 to 1.18? That would leave us a window to start supporting gtk3 in the plugins.

Technically, we can build all sorts of builds in the CI build chains. The problem is just what we can maintain, resources are limited.

leamas, 2019-12-21:
A more concrete proposal:  Next opencpn version has

  -  Plugin API 1.18 on MacOS , Windows and Android.
   - For linux compiled against gtk2 it is 1.17
   - For  linux/gtk3 is 1.18. which is identical to 1.17 besides that it
     does not load plugins built for <= 1.17.

bdbcat, 2019-12-23:
@leamas 
One more time, because I seem to be dense....
I see that current master is building fine for trusty in travis ci.  I look at the build log, and I see gtk2.
However, somehow, it does not make it into the cloudsmith repo.
So:
1.  Tell me again why we cannot support trusty, since we seem to build OK for it now....
2.  But why is it not getting into cloudsmith repo?

Or, is this a problem building plugins, and not a problem with core?
Help me understand...

leamas, 2019-12-23:
> However, somehow, it does not make it into the cloudsmith repo.

This is a plain bug, not related to this discussion. I 'll try to lake a look, but Christmas is looming...

> Tell me again why we cannot support trusty, since we seem to build OK for it now.

It's not that we cannot build trusty, it's the overall picture.

First, we do not make any builds for bionic.To not have artifacts for the current LTS is IMHO a bigger problem than a possibly dropped trusty.

From a CI perspective, we currently build in four lines. As can be seen,  maintaining this is not trivial. We need to add bionic. That makes five.

We also probably need to split at least bionic and perhaps also xenial in gtk2/gtk3 builds, see #1536. That makes six or seven builds. The question is if we can maintain such a number of build chains (as long as I'm alone on this I'd say no). One thing which could be done is to drop trusty.

When it comes to gtk3, it depends on how we resolve #1536. My current proposal is that we support gtk2 during the next release (5.1?) and drops it at next after that (6.0?). When we drop gtk2, we cannot build trusty.

The official 5.0.0 package which will hit debian bullseye/sid (11) and Ubuntu focal (20) is gtk3-only.  This is the final sunset for the trusty support.

So, while we *could* maintain the trusty builds, it comes to a cost.

The comparison done to windows updates (for example XP -> win 10) is not quite accurate. To update trusty to xenial is a far less complicated. Overall, updating the OS is part of the linux landscape. When it comes to old boxes with constrained resources, the solution is rather to use a less resource-hungry window manager such as fvwm than to stick to trusty.

leamas, 2019-12-23:
Oops, sorry, referring to this very bug. Thought I was writing in #1547

rgleason, 2019-12-23:
Dave 
Possibly the encrypyed key needs to be updated. Create a github  personal token for use with opencpn and travis, then encrypt it with travis ci and put that encrption directly after "secure:" in .travis.yml.  Then you will probably need to get your "CLOUDSMITH_API_KEY" and enter the name and key value into the environment variables for your program opencpn in circleci.

There are better instructions in the Dev Manual.

rgleason, 2019-12-23:
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:pi_installler_build_deploy

leamas, 2019-12-23:
no, it is not that. I am on it.

taruti, 2019-12-24:
Would plugins providing two build platforms "linux-gtk2" and "linux-gtk3" work? And the plugin installer just listing plugins for the compatible platform?

For some time all the infrastructure has to support both old (GTK2) and new (GTK3). But this is mostly an ABI issue, not actual API differences for most plugins?

rgleason, 2019-12-24:
I don't suppose the gtk2 and gtk3 switch could be a dynamic choice?

leamas, 2019-12-25:
> I don't suppose the plugin gtk2 and gtk3 switch could be a dynamic choice?

The gtk2/gtk3 linkage is a traditional linkage, not dl-loaded. So no, I don't understand how this could be dynamic.

> Would plugins providing two build platforms "linux-gtk2" and "linux-gtk3" work? And the plugin installer just listing plugins for the compatible platform?

This is more or less what I propose: 1.17 and 1.18 represents the same API besides that 1.17 is gtk2 and 1.18 gtk3.  On linux, 1.18 refuses to load or list plugins <= 1.17.

If the next-next opencpn version after 1.17/1.18 also should support both gtk2 and gtk3 we are effectively branching the API.  The combinations of different releases and gtk/2/gtk3 will become a complicated mess. Rather than working with sorting out this mess we should help plugins maintainers to move to gtk3, for example by making PRs.

After all, nothing built on debian/sid or ubuntu/focal can provide gtk2 support. So we need to move anyway.

leamas, 2019-12-25:
@bdbcat:

Trusty is not supported by Canonical either. Basically, this forces any user who wants to update ubuntu software to update. Same goes for security fixes. See https://wiki.ubuntu.com/Releases.

bdbcat, 2019-12-27:
@leamas 
1.  Trusy stays in.
2.  Adding Bionic seems pretty easy, cloning the steps for xenial.  I'll try that one, and see if we can share the load.

bdbcat, 2019-12-27:
Regarding the gtk2/3 connundrum:
It seems that OCPN core does not directly depend on libgtkx at all.  The dependency is implicit, through wx.  Try it:
$ldd opencpn | grep gtk
Since this is so, then supporting gtk3 comes automatically if the underlying distro contains wx built for gtk3.  This historically is a problem, forcing us to bundle wx on Mac and Windows, for instance.  We shall see what happens with Ubuntu in April.  We can only hope...
Worst case (and it is bad):  A distro does not support gtk2, and does not have a wxWidgets_gtk3 package available.  Our only alternative for this distro may be docker, or something like that...

On the other hand, we need to find out if any plugin REQUIRES gtk (other than implicit wx binding) in order to function.   I doubt we will find much, since whatever function they provide by direct access to gtk will be unavailable on Mac and Win, and so unlikely to be critical.

leamas, 2019-12-27:
Right. Debian sid has dropped the wxgtk2 support which sets a sunset for anything depending on that to Debian bullseye (11) / Ubuntu focal (20). This is the definitive sunset for anything depending on wxgtk2. 

Bundling wxgtk2 is possible, flatpak does that.  However, the official debian packages are built using wxgtk from 5.0.0 (yet not uploaded). Building against wxgtk 2 is not possible here since that package is dropped.

leamas, 2019-12-27:
> We shall see what happens with Ubuntu in April. We can only hope...

Actually, it's just to see what Debian is doing. Ubuntu follows Debian which is dropping wxgtk2 but have wxgtk3 packages which looks well maintained.

Some more info on the Debian situation: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=946018

> Since this is so, then supporting gtk3 comes automatically if the underlying distro contains wx built for gtk3. 

Not on linux, it FTBS (see above). cmake supports building against gtk3, but uses different compilation flags and library names. 

Given that we should support trusty (forever? this cycle?):  New proposed linux build artifacts:

    trusty/gtk2
    xenial/gtk2
    bionic/gtk2
    bionic/gtk3

Since together with the other builds this is > 4 builds we should probably add circleci builders so we can run the builds in parallel (the free plan on trusty is limited to four parallel builds, so is circleci).

leamas, 2019-12-27:
We have still no conclusion how long we should support wxgtk2-based builds. Announcing a sunset would probably make things easier for plugin devs. After all, maintaining  multiple builds for the smallest platform is a pain for any plugin developer. 

I suggest  that next API release 1.17 (gtk2)/1.18 (gtk3) is the transition window. That means at least next release + following dot releases as long as the API isn't updated.

bdbcat, 2019-12-27:
"Not on linux, it FTBS (see above)"
Sorry, what is FTBS?

leamas, 2019-12-27:
Fails To Build from Source  (sorry, packager jargon, filtered through a native Swede...)

transmitterdan, 2019-12-27:
I would note that some systems support simultaneously both wx with GTK2 and GTK3. But some wx<->GTK3 systems do not support webview yet. We might need some kind of Venn diagram of what is in and out of each distro.

leamas, 2019-12-27:
> I would note that some systems support simultaneously both wx with GTK2 and GTK3. 

Yes. But most (all?) are dropping gtk2 these days. Debian and Fedora comes to mind.

bdbcat, 2019-12-30:
@leamas 
As you can see, after some fussing, OCPN build on linux/debian no longer has any direct GTK2/3 dependency.  It has an indirect dependency through wxlibgtkx.
What this should mean is that we should be able to build and run on any debian-like distro that has a wxWidgets-dev package, regardless of the gtk version present or required by wx.  So we don't need to tie gtk version to our API level at all, I don't think.
There may still be troubles with wxWebview, TBD.
I wonder if we still need the gtk3-force option?

taruti, 2020-01-02:
@bdbcat this sounds like a wonderful thing and an elegant solution to the issue.

leamas, 2020-01-02:
Indeed some nice work! 

That said, isn't the basic problem the same: the resulting ocpn binary will be linked either to gtk2 or gtk3, and the plugins used must then also be linked to the either gtk2 or gtk3 (one cannot use plugins linked to gtk2 in a gtk3 opencpn binary etc.)

Looking at the xenial build at 36daac0 I get:
```
 ldd /usr/bin/opencpn | grep gtk
	libwx_gtk2u_gl-3.0.so.0 => /usr/lib/x86_64-linux-gnu/libwx_gtk2u_gl-3.0.so.0 (0x00007ff3fbd29000)
	libwx_gtk2u_html-3.0.so.0 => /usr/lib/x86_64-linux-gnu/libwx_gtk2u_html-3.0.so.0 (0x00007ff3fb5f1000)
	libwx_gtk2u_adv-3.0.so.0 => /usr/lib/x86_64-linux-gnu/libwx_gtk2u_adv-3.0.so.0 (0x00007ff3fb20b000)
	libwx_gtk2u_aui-3.0.so.0 => /usr/lib/x86_64-linux-gnu/libwx_gtk2u_aui-3.0.so.0 (0x00007ff3faf72000)
	libwx_gtk2u_core-3.0.so.0 => /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0 (0x00007ff3fa6f9000)
	libgtk-x11-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0 (0x00007ff3f4b35000)

```

I. e.,. it is linked to gtk2 (or, really, wxgtk2) libs. What  am I missing?

rgleason, 2020-01-02:
Thanks, Alec for asking about what plugins should do, I am finding that Sean's weatherfax and weather_routing use gtk3 and I expect most of the other ones in his stable do to.

leamas, 2020-01-02:
The same is true for bionic at 36daac0:

```
$ ldd /usr/bin/opencpn | grep gtk
	libwx_gtk2u_gl-3.0.so.0 => /usr/lib/x86_64-linux-gnu/libwx_gtk2u_gl-3.0.so.0 (0x00007f366c99e000)
	libwx_gtk2u_html-3.0.so.0 => /usr/lib/x86_64-linux-gnu/libwx_gtk2u_html-3.0.so.0 (0x00007f366c277000)
	libwx_gtk2u_adv-3.0.so.0 => /usr/lib/x86_64-linux-gnu/libwx_gtk2u_adv-3.0.so.0 (0x00007f366be9f000)
	libwx_gtk2u_aui-3.0.so.0 => /usr/lib/x86_64-linux-gnu/libwx_gtk2u_aui-3.0.so.0 (0x00007f366bc0c000)
	libwx_gtk2u_core-3.0.so.0 => /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0 (0x00007f366b3da000)
	libgtk-x11-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0 (0x00007f3665210000)

```

which means  that there is yet no official builds supporting plugins linked to gtk3

bdbcat, 2020-01-02:
@leamas 
So now we can discuss the impact on plugins.
Sure, the plugins must use the same version of wxlibs as the core.  But again, it is unlikely that any plugin needs direct linkage to GTK.  This must be verified.  Think of it this way: "A plugin that absolutely requires GTKx will not work at all on Win or Mac.  Not a very interesting plugin, at least to 90+% of our user base."

So, for each distro, we will need to determine what version of wx is supported in its standard distribution channel, and provide the core and plugin tarballs to suit.  I think we will find that there are some current distros that support either wxGTK2 or wxGTK3 (bionic?).  In those cases I suggest that we provide OCPN core as GTK2 (as we do now via Ubuntu PPA), and plugins to suit.
Or, to say another way:  On any linux distro, OCPN supports GTK2 if available, GTK3 otherwise.

AFICT, there is no functional advantage to OCPN running GTK3 vs. GTK2.

bdbcat, 2020-01-02:
btw, I think the X11 dependency can also be driven out.  I will investigate.

rgleason, 2020-01-02:
Would Sean have some thoughts about this?

leamas, 2020-01-02:
> we will need to determine what version of wx is supported in its standard distribution channel

For linux, this is the same picture. Up to around "now" most distros support both gtk2 and gtk3 and thus both wx-gtk2 and wx-gtk3. 

And from "noẃ", most distros including Debian are in the process of dropping the wx-gtk2 libraries. 

As I understand it the things I have stated earlier still applies, in particular https://github.com/OpenCPN/OpenCPN/issues/1536#issuecomment-569292074.

bdbcat, 2020-01-02:
Yes, with the proviso that when we say GTK2 or GTK3, we mean the equivalent wxWidgets build.  So we need to build wxGTK3 versions of some or all plugins, at least to start.  Not unexpected.
First chore is to scrub GTKx dependencies from the plugins.  Yes, Sean will have some opinions, I expect.
But again, this has nothing to do with our plugin API declarations.  I cannot see any reasonable scenario where a plugin using wxGTK3 would be substantially different or "better" than one using wxGTK2.  Or indeed, the Win and Mac versions. It really amounts to just one more dimension in our support matrix.

Take some practical examples:  I claim that there is no reason to support wxGTK3 on Xenial, even if it is possible for a particular user to install GTK3 and thus want that configuration.  I say the same thing for bionic.  There is no practical user benefit at all.  And on latest debian, there is no reason to jump through hoops to support wxGTK2, even if it is possible.

leamas, 2020-01-02:
>  I claim that there is no reason to support wxGTK3 on xenial

I agree, totally. My point is just that even if we have source compatibility (now enhanced, great work!), there is still an ABI breakage we need to handle.

If we switch focus to the plugins: the simplistic way would be to say that opencpn will switch to wx-gtk3-based builds from Ubuntu focal  (ETA april-20).  Up to this point we only provide wx-gtk2 builds.

Personally, I think this is too simplistic. plugin writers needs at least builds to test against during the bionic cycle. So, providing both wx-gtk2 and wx-gtk3 builds in bionic makes sense.

When it comes to focal, Debian has already set this up for us: only wx-gtk3 builds are possible.

bdbcat, 2020-01-02:
Agreed.
I would like to conjure some way to build for both wxGTK2 and wxGTK3 in the ci workflow without requiring OCPN CMakeLists.txt to make specific callouts to wx version, if possible.  Just trying to keep things simple in the code base. But this may not be possible.
Overall goal is to avoid requiring casual users to know about any of this.

leamas, 2020-01-02:
> I would like to conjure some way to build for both wxGTK2 and wxGTK3 in the ci workflow without requiring OCPN CMakeLists.txt to make specific callouts to wx version

It is definitely possible. The current cmake logic is sound: it looks for wx-gtk2 and uses it if found, otherwise it tries to use wx-gtk3. However, IIRC, there is (was?) a bug here which made it necessary to add the force-gtk3 flag even if only gtk3 was available. It should be easy to fix it's still a bug.

That said, we should keep the OCPN_FORCE_GTK3 option since it is the only way for users with both wx-gtk2 and wx-gtk3 installed to make a gtk3 build.

rgleason, 2020-01-02:
Alec wrote:
"But again, it is unlikely that any plugin needs direct linkage to GTK. This must be verified." Yes, it should be confirmed by someone who really knows the plugins.

Weather_routing_pi and Weatherfax_pi both use wxGtk3.0 libraries. I believe that if you remove these libraries, it will fail because I recently had to change the exact name of the library otherwise it failed, 

Is this "direct linkage"?

rgleason, 2020-01-02:
Looking at this again, are plugins going to have to have cmake rewritten to set gtk2 as default and then use gtk3 if found?   As is done here ?
https://github.com/OpenCPN/OpenCPN/commit/8f5778dd72f081239d474ed964816fe1dc92d2d5

bdbcat, 2020-01-02:
@rgleason 
Yes, in general, the plugin cmake files (for linux) should follow the lead of OCCPN core, and use wxGTK2 as default if available, and wxGTK3 otherwise.

rgleason, 2020-01-03:
Thanks.  So the cmake for each of the new Plugin Manager plugins needs to be modified, plus testplugin and weather_routing.

leamas, 2020-01-03:
@bdbcat: I see that you dropped the unarr-dev build dep. Was this on purpose (to use the bundled code), or just a fix to get it running?

bdbcat, 2020-01-03:
I did nothing except drop the dependency from control file, and it just worked.  Not sure where or what platform has that dependency.  Hints?

leamas, 2020-01-03:
Dropping it means using the bundled code and not the system variant. Since unarr is  available on Debian it does IMHO sense to use it.

Thoughts?

leamas, 2020-01-03:
BTW: unarr is available from Debian/Buster+, Ubuntu/Bionic+

bdbcat, 2020-01-04:
@leamas 
I cannot find any use of libunarr except optionally in chartdownloader plugin.  Am I missing something else?

leamas, 2020-01-04:
No, I have the same picture. OTOH, I havn't seen the dependency as a problem since it's available both as bundled code and on system level.

As I read the chartdldr code it's cmake will use the bundled code if the system library isn't available. In my eyes, using the system lib is better in this case since system libs tends to be better maintained, this is basically what this is all about.

[1 more reply]

closed 2020-03-02 by leamas