OpenCPN/OpenCPN #1537: Size of right-click menu.

tai-fun, 2019-12-19:
The size of the right click menu has increase a lot. May be OK for Android but not generally.

file:///home/thomas/Pictures/Screenshot_20191219_184729.png
file:///home/thomas/Pictures/Screenshot_20191219_184633.png

tai-fun, 2019-12-19:
[image]
[image]

tai-fun, 2019-12-19:
(Dragging worked but not pasting!)

bdbcat, 2019-12-20:
Thomas...
What platform is this?  What device?

tai-fun, 2019-12-20:
Dave
Ubuntu 19.10(!)
Laptop Panasonic CF-31
Screen width 271 mm

Thomas

bdbcat, 2019-12-21:
Thomas...
Corrected by e49c8985cd2e11b175dddf379a642db2dc83f504

Thanks
Dave

closed 2019-12-21 by tai-fun