OpenCPN/OpenCPN #1538: Install Squiddio from cloudsmith via XML, OCPN crashes when plugin enabled

jongough, 2019-12-19:
Clean Ubutu 16.04 installed OCPN from cloudsmith (`https://github.com/OpenCPN/OpenCPN/issues/1521`), installed Squiddio using new plugin installer, selecting the version shown. Install completed OK. Try to enable Squiddio, OCPN crashes with `Illegal instruction (core dumped)` shown on the command line used to start OCPN. The attached file is the log from the instance of OCPN which crashed.

[OCPN Crash while enabling Squiddio log.txt](https://github.com/OpenCPN/OpenCPN/files/3985698/OCPN.Crash.while.enabling.Squiddio.log.txt)

The Plugin Manager shows `squiddio 1.0.17`, when installed it shows `squiddio 1.0.17` installed, however, in the plugin box it shows `squiddo 1.16.0`.
[image]
[image]

leamas, 2019-12-20:
First: could you please test against the mainline as of https://cloudsmith.io/~david-register/repos/opencpn-unstable/packages/ ?

Is this possibly a duplicate of #1530?

leamas, 2019-12-20:
As for the crash: question is if this is an opencpn or squiddio issue. However, let's sort out the versioning stuff first.

jongough, 2019-12-20:
https://github.com/OpenCPN/OpenCPN/issues/1538#issuecomment-567801685
That is the test I did. I used the latest xenial build from that repository.

I reported the crash as [https://github.com/OpenCPN/OpenCPN/issues/1538](https://github.com/OpenCPN/OpenCPN/issues/1538)

jongough, 2019-12-20:
I don't know how to find out where the crash occurred as there is no debugging mechanism when running this process. If I build OCPN locally and run the debugger I am not using the code on cloudsmith. If I build the plugin locally the same issue occurs. If I do any of this stuff locally you will want me to do it with the code in cloudsmith.

So, I have reported the issue, you will need to investigate it. This issue is about the crash not the versioning.

leamas, 2019-12-20:
> So, I have reported the issue, you will need to investigate it. 

You are certainly not my boss, and cannot decide what I should do or not.  Nor anyone else in this project, besides you.

jongough, 2019-12-20:
I have explained that I cannot look into it as I cannot do it to your satisfaction, nor do I know how to debug this type of issue. I presume that you have the knowledge, skill, understanding and capability to do so as these changes are yours. That leave only one person that can look into it. If you decide not to that is your call. I am not dictating that you have to.

CarCode, 2019-12-20:
> 
> You are certainly not my boss, and cannot decide what I should do or not. Nor anyone else in this project, besides you.

Your wording is not acceptable.
For macOS your attempt is strange. Why you want to use another folder in ~/Library ?
There is already established the folder ~/Library/Preferences/opencpn/ and that is OK.
After all checking the beta 5.0.522 does not work with macOS.

leamas, 2019-12-20:
> Your wording is not acceptable

OK. So, what now? Going to enforce what I should do? Or, leave my person aside since this is not about me, it's about the project?

To be clear: of course I appreciate testing and other input on technical issues. But the words "you" , "your code" or "you should" creates an irritated resistance from my side. I  will certainly not refer to "your testing" in this way.

leamas, 2019-12-20:
If you can live with my ideas how we work together: the statement _5.0.522 does not work with macOS_ needs to be more specific to be a usable bug report. That said, I do have a bad feeling about some parts of the MacOS code. Will do some more testing  on my own.

leamas, 2019-12-20:
As for the new library: the installer differs between configuration and libraries. Basically, this mirrors how plugins are installed today -- also now libraries and configuration lives in different directories. See for example https://github.com/leamas/opencpn/wiki/Tarballs which hopefully sorts this out.

leamas, 2019-12-20:
Hm... is the crash the same as https://github.com/mauroc/squiddio_pi/issues/108 ?

Hakansv, 2019-12-20:
Just for info: On Win10 the squiddio_pi installed from plugin loader is working as it should. Uninstall as well.

jongough, 2019-12-20:
https://github.com/OpenCPN/OpenCPN/issues/1538#issuecomment-567932020
It appears similar in what happens, but I do not know if it is the same as I cannot debug it.

The same person that worked on the plugin installer also worked on squiddio. I would have thought they would have looked at this issue as it would seem quite important.

leamas, 2019-12-21:
@CarCode  wrote:

> After all checking the beta 5.0.522 does not work with macOS.

I wrote:

> That said, I do have a bad feeling about some parts of the MacOS code. Will do some more testing on my own.

Testing OpenCPN_5.0.522+1.e49c898.dmg  from https://cloudsmith.io/~david-register/repos/opencpn-unstable/packages/ on an almost  pristine mojave machine. Basic functionality install, enable, uninstall seems OK for oesenc, radar and squiddio. 

Please file a more specific bug report if problems persist on MacOS.

leamas, 2019-12-29:
I was able to reproduce the crash. Fixed by rebuilding squiddio on an updated xenial host. Probably some library problem.

Updated ocpn-plugins.xml available on https://github.com/leamas/plugins/

jongough, 2019-12-29:
I cannot get the ocpn-plugins.xml to work. The Catalog Manager shows 'Fail' when using this
[image]

jongough, 2019-12-29:
I copied the contents of the config file and pasted it into the one on the test machine and then squiddio installed. So the crash has been fixed and the new version of squiddio works.

leamas, 2019-12-29:
That link is to the html file. Always check links using wget or curl before you use them. You get the correct link to copy after selecting the Raw display on github.

jongough, 2019-12-30:
Nope, not going to go into full debugging mode just to follow a link to see if it goes where it is supposed to.

If you provide a link that appears to show it is pointing to the location of the xml file, then that is where it should point. Not every one is a developer.

rgleason, 2019-12-30:
So Alec gave us this link to his repository https://github.com/leamas/plugins/blob/master/ocpn-plugins.xml
but that is not a link to the actual file that we can use.
I click on the file ocpn_plugins.xml  and this is the link:
https://github.com/leamas/plugins/blob/master/ocpn-plugins.xml

BUT That does not work.  You have to then hit "RAW" to get the right link to use in the Advanced URL for installing the new version of the plugin.  That URL is

https://raw.githubusercontent.com/leamas/plugins/master/ocpn-plugins.xml

How the devil do we explain this to users?

jongough, 2019-12-30:
Good question, well phrased. Not sure Alec cares.

closed 2019-12-30 by bdbcat