OpenCPN/OpenCPN #1539: Plugin Installer plugins not removed when OCPN is uninstalled on Ubuntu 16.04

jongough, 2019-12-19:
Clean Ubuntu 16.04, install current OCPN from cloudsmith (`https://github.com/OpenCPN/OpenCPN/issues/1521`), install Squiddio from new plugin installer `+` icon. Close OCPN without running squiddio (because it currentl crashes OCPN).

Now issue `sudo apt purge opencpn`, opencpn and it's default plugins are uninstalled. The installed version of squiddio is left installed in the user area. If the root program, OCPN, is uninstalled then dependant programs, Squiddio, should also be uninstalled.

leamas, 2019-12-20:
This is IMHO no bug. The downloaded stuff is not managed by dpkg, so it should not be removed. Basically, installing a tool, downloading stuff with this tool and finally removing the tool should not mean that the downloaded stuff should be removed

If it is a bug, it should be reported against the downstream packaging and evaluated against Debian packaging policies. It is just not an upstream issue (we don't even have an uninstall target...)

jongough, 2019-12-20:
The plugin is installed by your process and placed into a location that is dependant on your process. So if OPCN is removed then part of the removal process should be to run a script to remove all plugins that have been installed by your process. The process does know where to look for plugins and where to look for plugin data so it should remove this. No other process will have this information.

It is only a bug within the plugin if the plugin gets notification that the uninstall process is running and does nothing about it. Does the uninstall process try to inform the plugins (installed by the new method) that this is happening? If not then it is a problem with this new plugin installation process as there is not a co-requisite uninstall process.

The 'legacy'  installation process for Ubuntu keeps track of dependant installations. When you issue an `apt purge opencpn`  all the dependant packages that have been installed will be uninstalled. This capability needs to continue with this new process.

leamas, 2019-12-20:
This is far from clear. I'll check  the debian policy on this, but short of time right now.

leamas, 2019-12-20:
The basic plugin uninstall tool is opencpn itself, which understands the packaging info created when installing the plugins. Using opencpn, user can install/uninstall plugins in user directories

However, opencpn is still a traditional package installed in /usr on Debian. There is no reasonable way to make an uninstall script in the packaging which messes with user files. It just doesn't make sense. And the downstream packaging script is the only code which could be executed when the package is uninstalled (opencpn is certainly not).

You could see this as a design shortcoming. However, I think it's fair to say that those involved has agreed to installing in user areas. That plugins not are removed when opencpn is removed using the package management tools is a straight-forward consequence of this design.

jongough, 2019-12-20:
If OCPN installs anything on behalf of the user and doesn't use the system packing process then it is responsible for all the files. I am sure this is why some packages use sandboxes so that when the sandbox is deleted all the associated files are deleted as well. I 'think' this is the way flatpak works

If OCPN uses the platform installation process then that installation process has the capability to clean up any installation. So if you manage to uninstall OCPN using the system process and leave behind a package which depends on OCPN it can be uninstalled later using the platform installation process. That is the task of the platform installation process.

If OCPN does not use the platform installation process but uses its own internal installation process then the platform installation process knows nothing about what has happened and cannot clean up. In this case it is up to OCPN to ensure all installations are cleaned up when it is uninstalled as it is the only place where installation information is stored. 

At the moment the happy case is covered. If OCPN installs a package and it is still capable of running it can uninstall the package. In the unhappy case where OCPN is not capable of running, or it cannot be run for whatever reason, or the user 'forgets' to run it for uninstallation purposes, then the uninstallation has to happen when OCPN is uninstalled. The user should not have to hunt around the system trying to find files that have been left behind. Most users would not know how to do this.

If this is a design shortcoming then it should be addressed by the designer now that it has been identified. At best it should be a 'simple' fix, at worst it may entail a complete rethink of the process. I am not the designer and was not involved in any of the design process so I am not well placed to help fix this issue.

leamas, 2019-12-20:
I think the current design is proper, including the fact that the user plugins are not installed. If someone comes up with patches they should IMHO be evaluated.

You are of free to think that this is reason to revert the merge of the installer. Simply put, I disagree.

> I am sure this is why some packages use sandboxes so that when the sandbox is deleted all the associated files are deleted as well.

Citation needed...

jongough, 2019-12-20:
https://github.com/OpenCPN/OpenCPN/issues/1539#issuecomment-568086565
The 'designers way' or 'no way'. A very blinkered approach. If this was taken for a plugin the users have a choice, but taking this view for the mainline program does not give the user a choice. 

I have a different point of view from the designer and like to take the end users perspective. The objective of this change is laudable, but the implementation is lacking and needs more thought. The happy cases have been considered, the unhappy cases have been discarded. At least 70% of ICT projects effort has to deal with unhappy cases and edge cases to make them usable and 'transparent' to the user base.

As for citations, the designer should do their own homework, particularly if they are going to quote standards as excuses for not thinking. But try thinking about Docker and why it is used. It is a self contained environment that stops leakage into the host environment, is light weight, and deleting containers removes 'all' changes associated with that environment. The flatpak OCPN environment does something similar, but the use of tarballs and not tracking when installing into hidden user areas does not.

leamas, 2019-12-20:
I don't think we can come to a conclusion here. Let's agree to disagree

jongough, 2019-12-20:
I think the designer has put this in the 'too hard basket', not a good look when they claim these changes are to make it easier for the user.

Yes we disagree, but the designer needs to accept all input and not be blinkered. There are alternate valid perspectives and these need to be given serious consideration.

jongough, 2019-12-21:
Has the designer look at or considered the 'prerm' script option of the deb files to see what can be done with this to remove 'user installed' extensions?

jongough, 2019-12-21:
It would require storing information in the opencpn.ini file for each plugin that is installed through the new method, i.e. the executable location and the shared and private data locations. This could then be used by the prerm script to remove these items.

This would still fail if there is corruption on the disk that means the opencpn.ini file could not be found, but it is a step forward.

rgleason, 2019-12-21:
Whether it is stored in the ini file or in a remove.xml file it probably does not matter, but it would be useful if the new PI Installer would be able to use that stored information to remove pi artifacts, and then remove the remove.xml or change the ini file upon completion of the job. 

I particularly think this is needed because we have decided not to provide information about where the new plugins are installed!

leamas, 2019-12-21:
It is already storing info about the installed files. See https://github.com/leamas/opencpn/wiki/Program-design#installations

leamas, 2019-12-21:
@jongough: 

The basics your reasoning are based on  few assumptions:

  - That I am a "designer" whatever that means and thus have specific obligations.
  - That it's normal that an application which downloads stuff removes ít when it is
    removed by the packaging system
  - That is is allowed and even desirable that a postrm script touches anything under /home
  - That the reason for using Docker, flatpak etc is that they make it possible to clean up
     when uninstalled
  

All of these assumptions are wrong. I know you don't accept my word for it, and suggest you look for information at other places which will confirm my statement. Some proposals:

  - Dave (that is, bdbcat) knows that I'm just a developer which have created some PR
    which have been merged by him. No more, and no less.
  - As for how packaged applications are handled on removal I'd suggest that you seek
    advice on the debian-devel or fedora-devel mailing list. Both have a lot of expertise
    available. They will confirm that the very idea to touch anything under /home in a postrm 
    script is not allowed.
  - As for the ideas why flatpak exists I can recommend the flatpak-devel mailing list. You
    might get an answer from Alexander Larsson, the main developer. He will explain that
    flatpak is a way to create GUI applications which works on multiple Linux distros, and
    the isolation it the tool to achieve that.

As Greta says: don't listen to me, listen to the scientists. Or, in this case, those who 
understand this stuff.

rgleason, 2019-12-21:
Dear Alec, 
Thank you for creating PI Installer. We are not trying to torment you, (...to support your non-sequitur -I am a big Greta fan).  Most of us, being users first, have expectations about software based upon experience. Installing and uninstalling plugins should be pretty straightforward (except when it isn't and there are 5 different OS, etc, then it isn't easy.)

> It is already storing info about the installed files. See https://github.com/leamas/opencpn/wiki/Program-design#installations

I did not know about this documentation.  Thank you. 
https://github.com/leamas/opencpn/wiki/Installation-paths

jongough, 2019-12-21:
https://github.com/OpenCPN/OpenCPN/issues/1539#issuecomment-568188079
When I referred to 'you' you got upset, so to be neutral I referred to the 'designer', if you would prefer I can refer to the 'developer'. The term does not affect me.

The person who build this change (is that better), needs to think about more than 'can I make it install' and more about the whole landscape. If you install something you should uninstall the same. I will let you do the research about leaving crap in the system after an uninstall. It is generally not appreciated on any platform. Remember, the people that use OCPN are mainly users not designers, developers, programmers, etc. and therefore do not understand most things. They also prefer to use an application and not read up about all the intricacies or chase up on the internet to find out how to uninstall, clean up. I certainly do not want to know about all the stuff I use, I just want it to work and clean up after itself, and I build this stuff!

So, either you accept you have had something to do with this change or you point us to someone who did. We need to fix it up and make it as usable as possible. I am just doing testing and finding issues which I am reporting in the hope that someone will fix them. I could fix some of this, but having tried to help you out in the past I know you will not accept this help. So I just point out the issues. That is what this 'Issues' list is for.

Please stop making excuses and pointing us all around the net to prove to you that something needs doing. It would take far less time if you, or the person that 'developed' this change, did the fixing rather than making excuses.

You refer to yourself as a 'Developer', well that entails medium to low level design, coding, testing, support, fixing, and dealing with users. If you are a programmer, then take out the medium design part. A 'Developer' should understand that items will come out in testing that have not been catered for but to which solutions are needed. As a Developer you apparently don't understand this. 

I also refer to 'Docker' (https://github.com/OpenCPN/OpenCPN/issues/1539#issuecomment-568092815) as an example of cleaning up and a self contained environment. I mentioned 'flatpak' as an example of cleaning up after itself and not the reason for flatpak.

And to quote you:

> As Greta says: don't listen to me, listen to the scientists. Or, in this case, those who
understand this stuff.

Be very careful in your assumptions about people who are involved in the OCPN project. I assumed you understood what you were doing as the basis for engaging with you. You are now demonstrating that I should lower my expectations somewhat.

And finally, I did already point out to you that docker does this.
> EDIT: It would also strengthen your case if you could find an example of a packaged linux application which works the way you seem to recommend.

leamas, 2019-12-22:
That was a long rant about me. I see no reason to comment on that.

leamas, 2019-12-22:
As for this bug: my position is that is should be resolved as NOT-A-BUG. 

In particular, that the .deb packaging done by us  does the same as other .deb packages when it comes to stuff downloaded or generated in  $HOME is not a bug -- it's how native packaging works.

If someone still think this is a bug, it's WONTFIX when it comes to my personal priorities.

leamas, 2019-12-22:
>> EDIT: It would also strengthen your case if you could find an example of a packaged linux application which works the way you seem to recommend.

>And finally, I did already point out to you that docker does this.

docker is certainly not a packaged linux application, it's a container system. *Packaged* in this context refers to native .deb, .rpm , etc. packages

If someone would like to step up to make a version of opencpn running in a docker container (that is, an opencpn docker image), this is an interesting alternative to using flatpak. However, from what I have understood docker isn't really suited to handle GUI applications, and that's one of the reasons  flatpak (and snap) has been developed.

My plate is more than full and I'm personally not interested in this. My bet is that  flatpak is more suited to handle opencpn than docker. And, for resource-constrained systems like the Rpi we still need a native .deb package.

leamas, 2019-12-22:
@rgleason 
> Thank you for creating PI Installer. We are not trying to torment you [...]

I don't really see the "we" here. I see you, jongough and, not to be forgotten, the current issue.

When it comes to you:  I actually skimmed through the two bugs where we developed the installer. Without you overall help, enthusiasm, testing and social skills we would not have been able to achieve what we have done. A big thanks!

rgleason, 2019-12-22:
Thank you Alec, I am a very lucky guy to be able to help such talented programmers who have contributed so much to OpenCPN. -What a team! Thank you Alec and Jon for all your work and passion. It is a process and PI Installer is a work in progress.

jongough, 2019-12-24:
https://github.com/OpenCPN/OpenCPN/issues/1539#issuecomment-568248536
It is still a change in behaviour. Instead of using the system installation process which tracks the install and allows for uninstall this change has gone a different direction and does the installation into 'hidden' parts of the file system. As such it is usurping/replacing the standard installation process. So this process needs to clean up like the standard installation process.

This is an issue and is a bug irrespective of the fact that the person who built/created the change does not want to address it. Someone will have to address the issues to ensure a smooth user experience. Your attitude has now required me to lower my expectations even further. Please stop quoting 'standards' and making excuses, open your eyes to the bigger picture and start addressing the issues being found.

leamas, 2019-12-24:
This is you opinion, not mine. Let's agree to disagree.

jongough, 2019-12-24:
Interesting perspective for a change that is supposed to 'improve' the user experience. I am a user and I need this fixing as it now degrades the experience for me. Perhaps you should restrict your changes to ones that do not impact on the usability of OCPN.

I agree that we disagree, I am just disappointed you are taking this approach. You are not filling me with confidence with the support of this process in future.

bdbcat, 2020-03-18:
Jon...
Yes, on linux, if OCPN is uninstalled, the plugins will be leftover in the .local directory.
For this release, this anomaly is judged non-blocking.  Perhaps for a later release we can investigate incorporating an uninstall hook that will scrub any and all local data created by OCPN.
But not now.

leamas, 2021-04-28:
This is basically the same issue as #1556, but on Linux. This issue has been discussed extensively. The basic conclusion is that while Jon's position is that this is a bug, my is that things work as they should.

This an example on a bug [beaten to death due to honest differences of engineering opinion, but without resolution](https://github.com/OpenCPN/OpenCPN/discussions/2224#discussion-3334592).  According to Dave (same link) such bugs should be closed, with status quo as the resolution (i. e., NOT-A-BUG). 

I concur.

rgleason, 2021-04-28:
Alec, they do not, and after Users learn more, you will be back to fix this.

leamas, 2021-04-28:
OK, let's start the discussion again...

I will not fix this, won't happen.  At least not in the sense implied by the heading, adding some logic to the package which removes all stuff installed in $HOME when the opencpn package is uninstalled.

The reason is simple: touching anything under $HOME is strictly forbidden for all packaging scripts in all known packaging guidelines, including Debian/Ubuntu. For that reason, it's not possible to remove anything installed under $HOME when package is uninstalled. This includes all sorts of interactive questions when uninstalling (which  BTW are frown upon for other reasons).

Simplest solution might be to document how to get rid of remaining stuff, should it be necessary. Another path would be to add  a script to the package which could be run  explicitly by the user, obviously before uninstalling the package.  

But, again: removing things under $HOME when uninstalling the opencpn package is not allowed.

leamas, 2021-04-28:
> Alec, they do not, 

Which "they"? what is it they don't?

rgleason, 2021-04-28:
Yes, documentation would be helpful.  I guess a separate plugin that cleans up is not possible but a script explicitly run by the user sounds ok to me.

leamas, 2021-04-28:
Still not a reason to leave this open. Making such a script is a simple RFE issue. The fact that opencpn leaves stuff untouched in $HOME when uninstalled is not a bug, it's a feature -- it's what a package should to according to applicable guidelines (and it's what other packages do).

Again: for reasons described [above](https://github.com/OpenCPN/OpenCPN/issues/1539#issuecomment-828185556) this issue should IMHO be closed.

nohal, 2021-04-28:
If any program, ever, removes data I have created with it when I am uninstalling it, I will certainly not be happy and will consider it a fatal flaw in that program and it's packaging. And, honestly, I have probably never had the "pleasure" of using such a program.

But I can understand that @jongough does not consider the plugin installation artifacts falling into the "data created by the user" category. If the plugins were installed using a traditional packaging mechanism like DEB or RPM, they would simply be removed on OpenCPN's uninstallation as well because of their dependence on core `opencpn` (While of course leaving their data created by the user in place).
What is different in this case? The only things I can see is the installation location and the semantics of the "install plugin X" command issued by the user.
So to conclude, as far as I'm concerned, removing the plugin installation artifacts (and only those) during OpenCPN's uninstallation would be legit.

rgleason, 2021-04-28:
I agree with Pavel, and thought that was what was being discussed.

leamas, 2021-04-28:
We cannot do this in the official package, it will not (and should not) pass a review. In fact, I wouldn't even propose it in this case for these reasons. So if we do it, it must be in the PPA packages only, creating different semantics in the two packages.

If we still consider to go this path, there is the question of "user". Package uninstall is run as root,  and the only sensible interpretation of "user" in this context means removing from all users on the system.  Which  often will be fine, but perhaps not always.

Other options includes changing installation paths to something like /var/opencpn/$USER. However, this will  conflict with the freedesktop specs, which assumes that user data lives under ~/.local.  It will most likely end up in using symlinks which becomes broken on uninstall.

Also: we don't remove ~/.opencpn either when uninstalling the package. This actually holds references to the installed plugins, so this needs to be taken care of as well in this case.

In short: there are multiple gotchas (probably more than listed here) if trying to remove the installed plugins implicitly when uninstalling opencpn. So many that I personally find it not worth the price (actually, I don't even know if it's desirable).  Documentation how to do after uninstalling + perhaps a script to run to prepare the uninstall are sensible alternatives, but that's it as I see it.

Also, here is an underlying idea I don't buy: that any change in behaviour compared to the legacy installation is a bug. This is a new system, with new pros and cons and partly new behaviour.

rgleason, 2021-04-28:
I think this is getting too complicated with all the protocols. Let's just close it and deal with it later. Sorry.

However I don't like the idea that garbage programs are left on my machine and I have to find them by accident and remove them when I thought they were removed.

jongough, 2021-04-28:
Nope.
OCPN has introduced a new package management process to replace the original one. The original package management process created two types of data, installation provided, i.e. icons, and user provided, i.e. config, and user data. The original package management would normally remove the installation provided data but leave the user provided data. In the case of linux (Ubuntu) the user had two options with the dpkg program. The following is directly from the man page:
```
        -r, --remove package...|-a|--pending
              Remove  an  installed package.  This removes everything except conffiles and other data cleaned up by the postrm script, which may avoid having to reconfigure the package if it is reinstalled later (conffiles are configuration files that are listed in
              the DEBIAN/conffiles control file).  If there is no DEBIAN/conffiles control file nor DEBIAN/postrm script, this command is equivalent to calling --purge.  If -a or --pending is given instead of a package name, then all packages unpacked,  but  marked
              to be removed in file /var/lib/dpkg/status, are removed.

              Removing of a package consists of the following steps:

              1. Run prerm script

              2. Remove the installed files

              3. Run postrm script

       -P, --purge package...|-a|--pending
              Purge  an  installed  or  already  removed package. This removes everything, including conffiles, and anything else cleaned up from postrm.  If -a or --pending is given instead of a package name, then all packages unpacked or removed, but marked to be
              purged in file /var/lib/dpkg/status, are purged.

              Note: some configuration files might be unknown to dpkg because they are created and handled separately through the configuration scripts. In that case, dpkg won't remove them by itself, but the package's postrm script (which is called by  dpkg),  has
              to take care of their removal during purge. Of course, this only applies to files in system directories, not configuration files written to individual users' home directories.

              Purging of a package consists of the following steps:

              1. Remove the package, if not already removed. See --remove for detailed information about how this is done.

              2. Run postrm script.

```
So, the introduction of the new package management process should at least maintain compatibility with the old one.

Further, the original package management split the data into two different locations, one being 'system managed' and the other being 'user managed', for want of better terms. The 'system managed' were is system areas that normally required root access to modify, the 'user manage' allowed the user to modify the data however they wanted.

In the new package management there is no separation of data, it is all in  a 'user managed' area, and supplied data is mixed in with config files, which should not happen. For ODraw a large number of icons are provided for the user to use and they have the ability to add extra icons if they wish, these two should not be mixed up, but they are.

So, back to the issue, when OCPN or a plugin is uninstalled the 'system managed' data should be removed. But there must also be a process to allow the user to get rid of the 'user managed' data as well. I frequently use the latter option when 'trying out' new software and I don't want to leave cruft lying around the system. At the moment you have to have OCPN installed to get an indication of where files might be stored (on Ubuntu they are in $HOME/.opencpn, a normally hidden user directory). Standard users would not know how to find these files and wouldn't know they even existed and should be deleted.

leamas, 2021-04-29:
> the introduction of the new package management process should at least maintain compatibility with the old one.

The thought that the new installer should be compatible in this respect has never occurred to me. The new installer is a new piece of software  with it's own pros, cons and behaviour. In particular, it is specifically limited to not evolve into a more or less complete package management system.  

>  But there must also be a process to allow the user to get rid of the 'user managed' data as well.

An end user could just uninstall the plugins from within OpenCPN. This removes the same data as legacy plugins installs under /usr. 

> In the new package management there is no separation of data, it is all in a 'user managed' area.

Actually, it is. The location ~/.local mirrors /usr. Legacy /usr/share/opencpn goes into `~/.local/share/opencpn` and so on. Although plugins theoretically could write their own data under `~/.local`. I see no reason why they should. Most plugins which needs to write data would use something under the config directory (`~/.opencpn` on plain linux) I guess, it's the simple path.

In the end, I still think  the current state is the best compromise between conflicting objectives. It's not that I don't see the problem you describe, it's just that the suggested fixes has too many drawbacks.

rgleason, 2021-04-29:
So the user should first Uninstaller all plugins and then Uninstaller opencpn?

Could a plugin or script help?

leamas, 2021-04-29:
If the user cares about the left-over cruft in ~/.local:

  - Uninstall all plugins from within OpenCPN
  - Uninstall the opencpn .deb package using regular tools, command line or GUI.

We cannot really script opencpn. It might be possible to create a script which clean up .local and .opencpn after uninstalling the package.  This is the third time I point this out ;)

jongough, 2021-04-29:
Just done an uninstall of OCPN and it leaves all the data in ~/.local/opencpn/plugins that it installed through Plugin Manager. I have tried doing a `dpkg -P opencpn opencpn-doc` which according to the man page should remove all of the installed elements and should use a 'postrm' script to remove dpkg unknown objects (objects that have been created but dpkg does not know of them).

There is a large amount of data left lying around which should be cleaned up by the package installer uninstall process. 

The uninstall process that was used for plugins before would have removed all there data in `/usr` or `/usr/local`. OD provides quite a few icons for the user, i.e. not the required ones for OCPN, and these should be removed as they were before.

leamas, 2021-04-30:
Marking this issue as an "enhancement" i. e., it's an RFE is an act of diplomacy. I bow my head.

Seen as an RFE I agree: it is a problem that things are left in  $HOME/.local/ after uninstalling the .deb package. As always, RFEs are by definition in a "patches welcome" state.

However, there are boundary conditions on a possible patch: When it comes to the official package it is not allowed to touch anything under $HOME in a packaging script. Expanding on this, it's probably not a good idea to do it in the PPA packages either as explained [above](https://github.com/OpenCPN/OpenCPN/issues/1539#issuecomment-828374772)

> which according to the man page should remove all of the installed elements: 

There is a difference between being part of a package and created by a package. The text does not apply.

> The uninstall process that was used for plugins before would have  [...]

Yes, the legacy process has many upsides not present in  the current installer. We have traded them for other advantages.  There is indeed no free lunch.

TL; DR  The left stuff in $HOME is a problem, and some solution would be nice. However, the solution is not to add packaging scripts which touches files under $HOME.

jongough, 2021-04-30:
Patches are not welcome, that has been amply demonstrated to me. I don't do patches to any of Plugin Manager any more as if they are accepted they are modified, marginalised, then removed in fairly short order. I am however identifying user issues in the current code and changes from normal accepted installation/removal procedures.

As for putting data in $HOME and then not removing it because it is in $HOME is a null argument. If there is no intention on removing and cleaning up after an install for items generated in $HOME, then don't use $HOME. Simple. If $HOME is to be used then there should be "systems" files and "user" files, the two types should be kept apart. System files only should be configuration files and items that are needed to allow the application to run and which can be recreated easily by reinstallation. User files should only be objects that the user has created using the application and which would be difficult, problematic, or impossible to recreate and should be maintained across most install/update/remove cycles.

This is not an enhancement request, but a request to fix issues identified in alpha and beta testing before producing production versions.

leamas, 2021-05-01:
> f there is no intention on removing and cleaning up after an install for items generated in $HOME, then don't use $HOME. Simple.

To use $HOME i. e., to install the plugins using regular user permissions is a basic design decision in the new installer. It gives us a number of advantages  but also drawbacks. This issue is one of those drawbacks. That said, changing this decision is certainly not simple, it's a major re-design.

> If $HOME is to be used then there should be "systems" files and "user" files, the two types should be kept apart. 

Agreed, and they are. "system" files lives under ~/.local, "user" files wherever the plugins writes them,  usually under ~/.opencpn. If a plugin modifies anyting under ~/.opencpn it's  a bug.

> This is not an enhancement request [...]

It is currently tagged as an Enhancement.  Until this is changed, this is how it is.

leamas, 2021-05-01:
BTW: Just a data point, since this seems to be about how package uninstall works (should work) on Debian. 

In my case, removing the firefox package leaves about 500MB of data in ~/.mozilla. This includes database files and plugins which are totally useless without the firefox package. But it's left in place for the same reasons I have described earlier.

This behaviour is certainly not regarded as a bug in Debian, the Firefox package has been reviewed extensively since it's such an important one.

jongough, 2021-05-01:
If this has been marked and an Enhancement that marking is wrong. This is an issue with the implementation leaving items after uninstall. 

Just because someone drives on the wrong side of the road does not make it correct nor does it stop the inevitable accident.

Why does dpkg have a specific capability to remove all of an installation? Because this is what should be done. Just because some software decides not to play well does not mean that all software should. This is still a bug. Plugin Manager is still and install manager. Install managers should remove all of what they install.

Life is too short to argue with you as it is pointless.

leamas, 2021-05-01:
Seems that this boils down to how to read and understand the Debian packaging guidelines, According to you these says data like ~/.local/\*/opencpn or ~/.mozilla should be removed when the package is removed. According to me data like this should  not be removed.

I will stand corrected should you raise this issue on the debian-devel or debian-mentors list and get a message that it's ok to remove this data on package removal (I can assure you won't, though).

As the maintainer of the Debian package (as well as some others) I have gained some experience in these issues. Leaving to others to judge who to trust.

rgleason, 2021-05-01:
While working on logbookkonni I had a chance to observe what happens when you uninstall and uninstall a plugin until there are no more versions cached.  The plugin is totally removed from  windows  %localappdata%\opencpn\plugins\logbookkonni_pi  

So plugins are allowed to behave in this way (no pejorative slant put on this) but OpenCPN operates in an policy zone which does not allow this?   Isn't this inconsistent?.

leamas, 2021-05-02:
> so plugins are allowed to behave in this way (no pejorative slant put on this) but OpenCPN operates in an policy zone which does not allow this? Isn't this inconsistent?.

Here we go again. 

There is no difference between how plugins are handled by OpenCPN and how Debian packages are handled by the system.

For plugins, the same files as was installed are removed when uninstalling the plugin. However, if the plugin create some file(s), typically in ~/.opencpn (Linux), these files are *not* removed. As a consequence, if the plugin later is re-installed it will still have the state saved and sort of being able to start on the same spot.

For Debian packages, the files which are part of the OpenCPN package and was installed at installation are removed on uninstall. As for plugins, any files *created* by the package are not removed. This includes for example ~/.local/\*/opencpn, ~/.opencpn  and similar locations (it also includes ~/.mozilla for firefox, the reference example). 

In the same way this also means that if a Debian package later is re-installed, all settings in ~/.opencpn and installed plugins  in ~/.local are still in place. From the overall Debian perspective this is a feature and not bug. Actually, it's a requirement.

We could perhaps do other things  in  Windows or MacOS. But this bug is already too long, we should not add those elements to this discussion. The subject is about Ubuntu.

leamas, 2021-05-05:
@jongough: I raised your question/statement about removing files on [debian mentors](https://lists.debian.org/debian-mentors/2021/05/msg00006.html).  The replies from the packaging people there are pretty clear, but if you want to check about the left-over stuff in ~ /.local  debian-mentors certainly is the place. The list is open, and the climate is friendly to newcomers. 

Just raise your concerns that the opencpn package seemingly doesn't remove everything in the package (your interpretation), and there will be answers from others than me.

jongough, 2021-05-09:
Went to debian mentors and it would appear that the way to install plugins would be with 'deb' files such that they can be uninstalled. There is no way for OCPN deb to uninstall the plugins (the thinking is muddled and confuses user data with installation data, but most of them do not get this can don't care) if they are not uninstalled by the user before OCPN is uninstalled. 

So there are two issues now. One installing plugins that have executable code and other binary objects into the user $HOME area means that they cannot be uninstalled reliably. Two, OCPN is not using the correct installation process for plugins that have executables and binary objects. The mailing list thought that these types of object should be in /usr or /usr/local.

The easy solution is to use an uninstall script that 'understands' the plugin manager process and can cause that to do the plugin uninstalls before OCPN is fully uninstalled. In this case it would be OCPN plugin manager that does the uninstall not the 'deb' file. If this solution is not acceptable then the plugins should be installed using the platform installation manager to allow that to clean up.

Remember it is not just the 'installed' plugins but the cache of downloaded plugins that also persists. With the current process it is quite possible to have multiple versions for different systems, i.e. Ubuntu 18.04, Ubuntu 20.04, etc. in the cache, so the cache is going to continue to grow.

With the increased usage of phones and tablets which are resource constrained to start with this process is going to exacerbate the issue.

leamas, 2021-05-09:
It's great that you sent your question to mentors. The thread is available [here](https://lists.debian.org/debian-mentors/2021/05/msg00014.html)

However, the conclusions you make are not shared by the mentors community. In particular,  they do not share your idea that is wrong by an application to install binaries and libs  which are left in .local when the opencpn package is uninstalled.  To me, this is rather obvious by reading ther replies. If in doubt, just ask them on the thread.

As I see it, the replies repeats, again and again:
  - That a packaged application like OpenCPN must not remove anything in $HOME when uninstalled (we seem to have some common ground here nowadays).
  - That leaving stuff, even binaries and such, in ~ /.local after package being uninstalled is a non-issue.
  - That there are multiple packages doing exactly what we are doing i. e. installing programs, binaries and data in $HOME.  Examples includes browsers and IDE:s. The problems you describe are present in all these packages, but it's not considered a problem. It's just how application-specific plugins are handled.

I suggest that we discuss this issue from the these three bullets. If you find that any of them contradicts the answers you have got, please  link this issue to mentors and check there.

leamas, 2021-05-10:
> It was like talking to a brick wall. 

Yes, none of them seems to agree with you as can be seen in the   [thread](https://lists.debian.org/debian-mentors/2021/05/msg00014.html). This BTW includes Tobias Frost who has been kind enough to sponsor the OpenCPN Debian package.

> I got told $HOME is sacrosanct and is a dumping ground

This is your wording. They just tried to explain that a package cannot touch  $HOME, and in many cases also tried to explain to you why this rule is necessary. Some of these reasons was new also to me, so I learned something.

> Finally, if debian mentors is supposed to be an area for mentors then they need to be more open minded

Perhaps, this is your interpretation. Another is that you need to be more open-minded when discussing with specialists which has been working together with these specific issues together for many years.

leamas, 2021-05-10:
Trying to summarize: Uninstalling the OpenCPN package leaves data  in ~/.local.  The core of this issue is that you consider this behaviour to be a bug. However:

  - According to Debian packaging rules this is **not**  a bug but rather the recommended practise (unless you think you have a better understanding of these rules than the collective wisdom of debian-mentors).
  - According to the OpenCPN community the corresponding behaviour on Windows is not a bug,  see #2255 an #1556.
  - Packages like Firefox, google-chrome and Eclipse exhibits the same behaviour which according to you is a bug. These are not ordinary packages, they are backed up by very talented open-source communities.

> Just because everyone else 'seems' to be doing it is no reason to do it. There are multiple instances in history where you can look and see how wrong this reasoning is.

Indeed, and I think I understand your reasoning and arguments. It's just that the counter-arguments are so much stronger. Some of them is the detailed stuff described by mentors. Others are the simple fact that we have traded some drawbacks such as the one discussed here for other advantages of the current plugin handling.

The new fact from the last round in this long bug seems to be that you acknowledge you are sort of a lone voice in the wilderness when pushing forward that this is a bug. Simply put, you are alone. You have no support from any open-source community, and  you have no examples on traditional .deb packages working the way you think they should when handling plugins installed without root privileges.

I once again think this bug is  [beaten to death due to honest differences of engineering opinion, but without resolution](https://github.com/OpenCPN/OpenCPN/discussions/2224#discussion-3334592) and thus should be closed.  Closing it would not mean that we don't see the problem (I can see it , for sure), just that we decide leave things as they are since there are no reasonable way to fix it.

jongough, 2021-05-11:
'The earth is flat! Everyone knows that', not sure this would hold up now, but in the past you were a heretic if you said otherwise.

I understand what is being said and why, but I did not get an answer to my question. I got answers to other questions I did not ask. I wanted to know the correct process to clean up when a package is fully uninstalled, including non-user generated data. i.e. installation data, that would be put into the only writeable area available, $HOME. Basically the current process is that 'deb' does not touch $HOME. I was told that if a plugin installs large amounts of binary data that is not user generated or modified then this installation should be in a 'deb' so that it would be installed in a system area and could be uninstalled.

However, in the 'deb' packages there is a 'postinst' script that copies files from the install area to the user area. There is the capability to put in a 'prerm' that could do the exact opposite, i.e. remove the files that were installed. This would not break anything as the next install would do the copy again overriding any changes that had been made. So this is 'installation' data. Plugin manager has all the files needed to clean up during an uninstall of OCPN if requested. It would be OCPN doing the reverse of what it does not on first startup, and would be a script provided in the installation process.

Your comment above says that the $HOME being untouchable is a 'recommendation' not 'forbidden'. Your comment about windows suggests that it is a bug, but no one can be bothered to fix it. Also, just because a bunch of people decide being antisocial is OK does not make the behaviour OK.

I thought we had moved on from the old days of every byte and cpu cycle was very expensive and the domain of large corporations, to personal computers that were expensive, then to personal electronic devices for the masses and, finally, incorporated user friendliness as a core concept, i.e. it is easy to use and doesn't need changing because an app has been installed then half uninstalled (Microsoft is really good at leaving stuff lying around and making good PC's useless). The devices still cost money and wasting resources is showing contempt for the user by saying their device is no good and they need to upgrade because we cannot be bothered to do the job correctly, i.e. Google, Apple, etc.

So, I still think the plugin manager process needs updating to either use a 'prerm' script or allow the installation of 'deb'/'exe'/'pkg' plugins that can clean up after themselves.

nohal, 2024-02-15:
That if the user wants to uninstall plugins, he has to do so before uninstalling OpenCPN is described in the manual. If anybody feels the manual should word it differently or more prominently, it is a community maintained resource available at https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn, there is nothing to change on the logic that would not be conflicting with the number one requirement of not touching data the user introduced to the system (And yes, even the plugin binaries is data from the core application point of view).

closed 2024-02-15 by nohal