OpenCPN/OpenCPN #1540: Plugin Installer does not display correctly if old plugin installed when OCPN running

jongough, 2019-12-19:
Clean Ubuntu 16.04, install cloudsmith () opencpn, start OCPN and open options/plugins
[image]
close window. Install plugin from ppa repository then open options/plugins the display is now not correct
[image]

leamas, 2019-12-20:
Is this a duplicate of #1530?

jongough, 2019-12-20:
No this is not a duplicate it is a new one, notice the two '+' (plus) signs on the bottom image. That is the issue being reported.

leamas, 2019-12-20:
Ah, OK. 

I have a some UI fixes under way. Let's drop this until they make their way to the mainline. This is clearly a bug that should be fixed, but not a blocker.

tai-fun, 2019-12-25:
This is real strange.
Every time you change to a new template, Options->General->Templates, a "+" is added to the top "Plugins" tab in "Options". After 6 changes no plugins are visible, only multiple "+". Then....if I change to a template that includes changing language ....all the "+", except the one at the bottom, are gone.
Repeatable.

leamas, 2019-12-25:
That's indeed strange. Which build are you using?

leamas, 2019-12-26:
I cannot reproduce this on 5.0.522+1.a9628ef, by Jon's or tai-fun's method. Could you please verify that the bug is still there on this os later builds?

jongough, 2019-12-26:
I purged both OpenCPN and ocpn_draw_pi to ensure it started clean. I have just installed OpenCPN 5.0.522+a9628ef from `https://cloudsmith.io/~david-register/repos/opencpn-unstable/packages/`, I started OpenCPN and looked at the installed Plugins:
[image]
Then installed opencpn-plugin-draw from the current official ppa repository. This also autoinstalled wx3.0-i18n as a co-requisite, went into Plugins:
[image]
This issue occurs only when OpenCPN is running and the install of the plugin occurs. If I stop OpenCPN and restart the display issue is not seen.

tai-fun, 2019-12-26:
The problem I described above is still there using fresh git code compiled locally.

leamas, 2019-12-26:
> This issue occurs only when OpenCPN is running and the install of the plugin occurs. If I stop OpenCPN and restart the display issue is not seen.

Installing a packaged plugin while opencpn is IMHO running is nothing we can support with a reasonable effort. This basically boils down to the plugin installer, which should require at least a restart of opencpn after installing the plugin package.

leamas, 2019-12-26:
> The problem I described above is still there using fresh git code compiled locally.

Hm... it's hard to fix a bug which cannot be reproduced. The possible differences could be either your local build or your local configuration.

Could you possibly try:

  - One of the official snapshots at https://cloudsmith.io/~david-register/repos/opencpn?
  - With a  minimum configuration without any templates, and then just an empty one (that's
    what I have tried). 

That would give some common ground.

tai-fun, 2019-12-26:
The box where this happens is on Ubuntu 19.10. A guess would be that it's
the changes in wx libraries behind this problem. Using Ubuntu 18.04 shows
no problems.
I withdraw my case , for now. Let's see what happens, later on, with Ubuntu
20.04.

jongough, 2019-12-26:
https://github.com/OpenCPN/OpenCPN/issues/1540#issuecomment-569059121
The plugin installer process sets user expectations. If they can install from the Plugin Manager and then use the plugin without restarting OCPN then that is what they expect of all installations. The fact that the newly system installed plugin shows up is good and it is usable, the issue is the display is incorrect and needs to be fixed. There should not be spurious artefacts displayed. This problem suggests an underlaying issue in the way that the information for display is built. It would appear that, in my example, the screen is built twice, once without any plugins and again with plugins.

bdbcat, 2020-03-18:
Please verify on current code base.

bdbcat, 2021-04-29:
Closing, ancient, inactive

closed 2021-04-29 by bdbcat