OpenCPN/OpenCPN #1541: PluginMgr vs Debs

free-x, 2019-12-20:
Hi,

I'm using Ubuntu 20.04beta. I have compiled current git master and packaged it as Debian package. Furthermore oesenc is installed as package.
``` 
$dpkg --get-selections | grep opencpn
opencpn						install
opencpn-gshhs-crude				install
opencpn-gshhs-full				install
opencpn-gshhs-high				install
opencpn-gshhs-intermediate			install
opencpn-gshhs-low				install
opencpn-plugin-oesenc				install
opencpn-tcdata					install
```
Plugins are located in /usr/lib/opencpn as always
```
$ ls /usr/lib/opencpn/lib
libchartdldr_pi.so  libgrib_pi.so       libwmm_pi.so
libdashboard_pi.so  liboesenc_pi.so
```
I notice that the plugins are not visible and there is no way to activate
[image]

Regards
free-x

free-x, 2019-12-20:
I found in log following message
```
09:08:37 CET: PlugInManager: loading plugins from /home/oleg/.local/lib/opencpn;/usr/local/lib64/opencpn;/usr/lib64/opencpn
```
Why /usr/lib/opencpn is no longer accepted?

leamas, 2019-12-20:
hm... IIRC the cmake code looks for /usr/lib64 and if it finds it uses it in favor of /usr/lib. That is, it assumes system is Fedora  or OpenSuse if it finds /usr/lib64.

Now it seems that Ubuntu has introduced a /usr/lib64 directory. This looks strange to me, since Ubuntu being a Debian derivative supports multi-arch with platform directories under /usr/lib. Is this directory there on a pristine installation?

You can walk-around this by specifying XDG_DATA_DIRS in the environment.

leamas, 2019-12-20:
> Now it seems that Ubuntu has introduced a /usr/lib64 directory

After some googling the conclusion is that the /usr/lib64 directory is gone since long on ubuntu, replaced by /usr/lib/x86_64-linux-gnu. So this is basically about you having some old paths still on your system.

OTOH, you may not be alone. Perhaps a cmake option disabling the automatic use of /usr/lib64 if it exists is motivated, dunno.  

Using XDG_DATA_DIR/OCPN_PLUGIN_PATH environment variables should still be able to handle your existing situation in any case.

free-x, 2019-12-22:
@leamas 
I think it's related to this code
https://github.com/OpenCPN/OpenCPN/blob/9d5859469ef1ec2d47597187a380868327d6ea64/CMakeLists.txt#L256
shouldn't that be extended?
set(_LIBDIRS "~/.local/lib:/usr/local/lib64:/usr/lib64:/usr/lib")

leamas, 2019-12-22:
I don't think so. When it comes to systems which actually uses /lib64 etc like fedora, suse and arch lib64 is for 64-bit libraries and /lib for 32-bit. And there is no opencpn 32-build for linux supported. So trying to load anything from /lib (and related /usr/lib etc.) is plain wrong.

What you have is basically a misconfigured system -- on a modern ubuntu there should not be any /lib64 and friends. Question is how to handle this. One way is that you do what should do anyway: move whatever stuff you have in /usr/lib64 to /usr/lib/gnu-linux-x86_64 and remove the directory. This would make your ubuntu installation sane, and current cmake code would work just fine.

Another way might be to add a cmake option overriding the autodetection, basically allowing user to define OCPN_LINUX_LOAD_PATH manually. While simple enough, there are already too many options and if possible we should avoid this.

Thoughts?

free-x, 2019-12-22:
I have changed and now it looks pretty good
```
15:30:50 CET: PlugInManager: loading plugins from /home/oleg/.local/lib/opencpn;/usr/local/lib64/opencpn;/usr/lib64/opencpn;/usr/local/lib/opencpn;/usr/lib/opencpn
15:30:50 CET: Using LD_LIBRARY_PATH: /home/oleg/.local/lib/opencpn:/usr/local/lib64/opencpn:/usr/lib64/opencpn:/usr/local/lib/opencpn:/usr/lib/opencpn:/usr/lib/opencpn
15:30:50 CET: Using PATH: /home/oleg/.local/lib/opencp/bin:/usr/local/lib64/opencp/bin:/usr/lib64/opencp/bin:/usr/local/lib/opencp/bin:/usr/lib/opencp/bin:/home/oleg/.local/lib/opencp/bin:/
usr/local/lib64/opencp/bin:/usr/lib64/opencp/bin:/usr/local/lib/opencp/bin:/usr/lib/opencp/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap
/bin
15:30:50 CET: Loading plugins from dir: /home/oleg/.local/lib/opencpn
15:30:50 CET: PlugInManager searching for PlugIns in location /home/oleg/.local/lib/opencpn
15:30:50 CET:    Directory /home/oleg/.local/lib/opencpn does not exist.
15:30:50 CET: Loading plugins from dir: /usr/local/lib64/opencpn
15:30:50 CET: PlugInManager searching for PlugIns in location /usr/local/lib64/opencpn
15:30:50 CET:    Directory /usr/local/lib64/opencpn does not exist.
15:30:50 CET: Loading plugins from dir: /usr/lib64/opencpn
15:30:50 CET: PlugInManager searching for PlugIns in location /usr/lib64/opencpn
15:30:50 CET:    Directory /usr/lib64/opencpn does not exist.
15:30:50 CET: Loading plugins from dir: /usr/local/lib/opencpn
15:30:50 CET: PlugInManager searching for PlugIns in location /usr/local/lib/opencpn
15:30:50 CET:    Directory /usr/local/lib/opencpn does not exist.
15:30:50 CET: Loading plugins from dir: /usr/lib/opencpn
15:30:50 CET: PlugInManager searching for PlugIns in location /usr/lib/opencpn
15:30:50 CET: Found 5 candidates
15:30:50 CET: Checking plugin candidate: /usr/lib/opencpn/libgrib_pi.so
15:30:50 CET: Checking plugin candidate: /usr/lib/opencpn/libdashboard_pi.so
15:30:50 CET: PLugin is compatible: true
15:30:50 CET: PlugInManager: Loading PlugIn: /usr/lib/opencpn/libdashboard_pi.so
15:30:50 CET: PlugInManager:  /usr/lib/opencpn/libdashboard_pi.so
              API Version detected: 106
              PlugIn Version detected: 102
15:30:50 CET: Checking plugin candidate: /usr/lib/opencpn/libwmm_pi.so
15:30:50 CET: Checking plugin candidate: /usr/lib/opencpn/liboesenc_pi.so
15:30:50 CET: Checking plugin candidate: /usr/lib/opencpn/libchartdldr_pi.so
15:30:50 CET: oesenc_pi:  Calculated pix/mm = 4.66212
15:30:50 CET: oeSENC_PI RasterSymbols texture: 205
15:30:50 CET: ChangeLocale: Language load for:  en_US
15:30:50 CET: ChangeLocale: Locale Init OK for:  en_US
15:30:50 CET: Loading catalog for:  opencpn-grib_pi
15:30:50 CET: Loading catalog for:  opencpn-wmm_pi
15:30:50 CET: Loading catalog for:  opencpn-oesenc_pi
15:30:50 CET: Loading catalog for:  opencpn-chartdldr_pi
15:30:50 CET: Loading catalog for opencpn core.
```

leamas, 2019-12-22:
what have you changed  and how?

rgleason, 2019-12-22:
I wanted to know the same things!
Free-x helping with this information will help others!

free-x, 2019-12-22:
here is my trivial patch
```
--- OpenCPN/CMakeLists.txt	2019-12-22 16:26:48.097025873 +0100
+++ OpenCPN-patched/CMakeLists.txt	2019-12-22 10:31:29.925304558 +0100
@@ -254,7 +254,7 @@
 
 if (NOT WIN32 AND NOT APPLE AND NOT QT_ANDROID)
     if (EXISTS "/usr/lib64")
-        set(_LIBDIRS "~/.local/lib:/usr/local/lib64:/usr/lib64")
+        set(_LIBDIRS "~/.local/lib:/usr/local/lib64:/usr/lib64:/usr/local/lib:/usr/lib")
     elseif (EXISTS "/usr/lib/arm-linux-gnueabihf") 
         set(_LIBDIRS "~/.local/lib:/usr/local/lib:/usr/lib/arm-linux-gnueabihf:/usr/lib")
     else ()
```
Regards
free-x

leamas, 2019-12-22:
well, if it works for you that is great, It is however MHO not acceptable, see beginning of https://github.com/OpenCPN/OpenCPN/issues/1541#issuecomment-568247777

taruti, 2019-12-24:
Loading manually installed plugins is also broken on master with Arch linux.

Log tells me
```
16:45:20 GMT: PlugInManager: loading plugins from /home/t/.local/lib/opencpn;/usr/local/lib64/opencpn;/usr/lib64/opencpn;/usr/local/lib/opencpn/opencpn
```

Plugins are installed like:
```
/usr/local/lib/opencpn/libocpn_draw_pi.so
/usr/local/share/opencpn/plugins/ocpn_draw_pi/...
```

XDG_DATA_DIRS is `XDG_DATA_DIRS=/home/t/.local/share/flatpak/exports/share/:/var/lib/flatpak/exports/share/:/usr/local/share/:/usr/share/`.

leamas, 2019-12-25:
hm... on Fedora I'd expect locally installed plugins to end up in /usr/local/lib64. The contract is really that there are no 64-bit plugins in /usr/lib:/usr/local/lib. 

But not on arch. Any idea why? 

`
16:45:20 GMT: PlugInManager: loading plugins from /home/t/.local/lib/opencpn;/usr/local/lib64/opencpn;/usr/lib64/opencpn;/usr/local/lib/opencpn/opencpn`

This seems to be #1535, right?

taruti, 2019-12-25:
Looking at ocpn_draw_pi the lib vs lib64 selection is done by https://github.com/jongough/ocpn_draw_pi/blame/master/cmake/PluginInstall.cmake 

To summarize on amd64 it installs to lib64 on Fedora+Suse and lib on Debian+Arch+Gentoo. 

This seems also the case for some other plugins.

Perhaps it is #1535. I'll look more into it in few days!

leamas, 2019-12-26:
That code tries to work it out by looking at exact distro. This leads to maintenance problems, for example the outdated lib64 branch for Debian, dropped since about five years (?).

It might boil down to a cmake variable overriding the automatic setup. That would provide a loophole for any distro, although not that easy to use at all times...

But that is not a reason to not have better defaults.

taruti, 2019-12-27:
PR #1595 fixes things for me.

leamas, 2020-10-18:
@free-x : This should now be fixed since long, opencpn definitely looks in /usr/lib/opencpn/plugins  for plugins. Time to close?

free-x, 2020-10-19:
@leamas you are right. closed

closed 2020-10-19 by free-x