OpenCPN/OpenCPN #1542: Win10: O crash/hang while waiting for MUI bars.

Hakansv, 2019-12-20:
Dave..
Win10. O 5.0.522. 
I think the described scenario and long waiting for the MUI canvas menus was introduced after "large commit" but before plugin installer commit.

Given the start sequence below an impatient user can easily get O to hang/crash. If I right click one or more times and/or click the tools menu during step 4 below O will hang. Sometimes it can survive but often it's crashed down. See last shot. (The half showed pop up is Windows saying the program has crashed and is about to shut down)

Start up sequence:
1. OCPN on screen. Base chart and a indication of the canvas border but it's far out to the right. Shot 1
2. Canvas border position corrected. GPS position established. Tools menu bar established and chart piano is about to be constructed but left canvas only "half way" Shot 2
3. If used is the oeSENC chart list established but don't react on any mouse click or enter key.
4. Waiting from 5 to 20 seconds.
5. Eventually are the MUI canvas menus established.  Shot 3
6. oeSENC chart list is responsible and can be shut down. But if I've succeed to open options window before this it must be closed in the background before oeSENC chart list can be closed.

See last logs before a crash at the end. Also the log part while waiting for the MUI bars on a "untouched" scenario.

Håkan

[image]

[image]

[image]

[image]

```
8:25:23 AM: Getting DefaultSystemLocale...
8:25:23 AM: Found LanguageInfo for: Swedish
8:25:23 AM: BuildFBO 3
8:25:23 AM: OpenGL Framebuffer OK, size = 413
8:25:23 AM: BuildFBO 3
8:25:23 AM: OpenGL Framebuffer OK, size = 413
8:25:23 AM: BuildFBO 3
8:25:23 AM: OpenGL Framebuffer OK, size = 870
8:25:23 AM: BuildFBO 3
8:25:23 AM: OpenGL Framebuffer OK, size = 870
8:25:24 AM: BuildFBO 3
8:25:24 AM: OpenGL Framebuffer OK, size = 413
8:25:24 AM: BuildFBO 3
8:25:24 AM: OpenGL Framebuffer OK, size = 870
8:25:24 AM: PlugInManager: Calling LateInit PlugIn: C:\Program Files (x86)\OpenCPN\plugins\ocpn_draw_pi.dll
8:25:24 AM: Changing NMEA Datasource for IIMTW to UDP:0.0.0.0:10110 (Priority: 3)
8:25:24 AM: Changing NMEA Datasource for IIVLW to UDP:0.0.0.0:10110 (Priority: 3)
8:25:24 AM: Changing NMEA Datasource for GPRMC to UDP:0.0.0.0:10110 (Priority: 3)
8:25:24 AM: Changing NMEA Datasource for GPRMB to UDP:0.0.0.0:10110 (Priority: 3)
8:25:24 AM: Changing NMEA Datasource for GPGGA to UDP:0.0.0.0:10110 (Priority: 3)
>>> Crash while right click and/or a menu bar click.

Next start up and I've done nothing until MUI bars appeared:
8:32:29 AM: Changing NMEA Datasource for PGRME to UDP:0.0.0.0:10110 (Priority: 3)
>>> Canvas redraw.
8:32:54 AM: options: got device count: 1
>>> MUI bars appeared 58-29=29 sec
8:32:58 AM: Changing NMEA Datasource for GPGLL to UDP:0.0.0.0:10110 (Priority: 3)
8:32:58 AM: Changing NMEA Datasource for PGRMZ to UDP:0.0.0.0:10110 (Priority: 3)
```

leamas, 2019-12-20:
Håkan,
> I think the described scenario and long waiting for the MUI canvas menus was introduced after "large commit" but before plugin installer commit.

Assuming that you build from source: have you ever done a git bisect? In situations like this it's a great help for those trying to fix the problem.

Hakansv, 2019-12-20:
I have sometimes used the bisect but since the "large commit" is just that and my description may lead someone to understand in what code area we should look closer. Maybe remember what have been changed and nail some tests to narrow the search?

bdbcat, 2019-12-20:
Hakan....
1.  Delay only appears with oeSENC active?
2.  Probably the delay come from the process of loading and converting all of the hundreds of user mark icons.  They should be cached after the first install run, but maybe the cache persistence in broken? See routeman.cpp:1417  I will investigate.

Hakansv, 2019-12-20:
Dave..
1. No, although shorter when less is loaded it's there even without oeSENC, oeRNC and Fugawi. The hang/crash described is though only present when oeRNC is active. 
2. I've a persistent file normally loaded. Although 5000 rows in the gpx it's not more than a couple of second difference with or without. 

Now, of course?, I couldn't really reiterate the 20 second delay. It's more 8 to 10 but all in connection to "options: got device count: 1" or from what happens before: "SoundFactory" i.e. CreateControls and CreatePanel_UI. Of course!? that's what we're talking about.
With or without oeSENC it's not a considerable difference.

```
Without oeSENC:
7:55:56 PM: OpenGL Framebuffer OK, size = 870
7:55:56 PM: PlugInManager: Calling LateInit PlugIn: C:\Program Files (x86)\OpenCPN\plugins\ocpn_draw_pi.dll
7:55:56 PM:    ***GPS Watchdog timeout at Lat:57.9851   Lon: 11.7393
7:56:02 PM: options: got device count: 1
7:56:03 PM: BuildFBO 3

With oeSENC
9:02:21 PM: OpenGL Framebuffer OK, size = 870
9:02:21 PM: PlugInManager: Calling LateInit PlugIn: C:\Program Files (x86)\OpenCPN\plugins\ocpn_draw_pi.dll
9:02:27 PM: options: got device count: 1
9:02:31 PM: Creating new chart
9:02:31 PM: Initializing Chart C:\Users\admin\Documents\Charts\Fugawi_SE\TD-ST-HG\932_1.BAP
9:02:31 PM: Available FALSE, retrying... nLoop: 1
9:02:32 PM: ofc_pi: Checking server utility at {C:\Program Files (x86)\OpenCPN\plugins\ofc_pi\ofcserverd.exe}
9:02:32 PM: ofc_pi: starting serverd utility: "C:\Program Files (x86)\OpenCPN\plugins\ofc_pi\ofcserverd.exe" -p OCPN_OFC8520
9:02:33 PM: ofc_pi: serverd Check OK...LoopCount: 10
9:02:33 PM: hk0
```

bdbcat, 2019-12-21:
Hakan...
I cannot reproduce.
Please minimize your system to simple cm93, with no plugins active, and test.  If startup OK there, add complexity until you notice a repeatable pattern.
Dave

Hakansv, 2019-12-21:
Dave..
Yes I've mostly been using a charts group with CM93 and a few KAP only. Of course it's some seconds shorter when oeSENC chart list is not loaded.
To test more carefully I also deactivated all chart plugins. senc, rnc and ofc. Faster, yes, but still a noticeable delay until MUI canvas menus are shown. 8-10 seconds. 
Some strange repeatable effects noticed:
First start up shot using CM93 and no chart plugins activated: Note: The single extended piano bar.
Second start up shot still only CM93 charts are used but oeSENC_pi is activated. Now more bars. The right bar is showing up whilst MUI canvas menus shows up.

For the first shot a zoom is not recalling missing bars but another redraw like after showing menu by the Alt key does.

This may be misleading and not relevant but anyhow: I can't debug without commenting out the two SendSizeEvent(); inChart1.cpp : 5545. The debug just hang without notice. And, yes I've tried release without SendSizeEvent() but no start up time difference. 

Log for first shot says nothing but plenty of:
8:59:09 PM: OpenGL Framebuffer OK, size = 486
8:59:09 PM: BuildFBO 3
8:59:09 PM: OpenGL Framebuffer OK, size = 783
8:59:09 PM: BuildFBO 3
8:59:09 PM: OpenGL Framebuffer OK, size = 783
8:59:09 PM: BuildFBO 3
8:59:09 PM: OpenGL Framebuffer OK, size = 486
8:59:09 PM: BuildFBO 3

[image]

[image]

bdbcat, 2019-12-21:
Hakan...
I mean really minimal.  No VDR, no connections, single screen only, no charts, even.  If MUIBar is still slow then, we have eliminated lots of potential candidates.

Hakansv, 2019-12-22:
Dave..
OK, will go from "scratch" but not today. I'll be back....
Thanks
Håkan

Hakansv, 2019-12-25:
Dave ..
To come back to this now when my system is "clean"
I couldn't reproduce the crash or hang any more. Maybe some icon/image trouble also here?
But there are still a delay until MUI bars are shown. When all "normal" charts and plugins are there we have some 10 seconds irresponsible delay. Some statistics below if you like.

OK, 10 seconds is not much but if you are just waiting and nothing is happened more than the boat is moving if GPS is available you may think O is ready for action, but he's not. 
That in combination with a not responsible oeSENC message may cause "irritation". Maybe a "busy cursor"
 
What could be worth adjusting is when the oeSENC Message box is viewed. Now it's shown before canvas MUI bar but it's not active and not responsive for any OK-click. It takes some time until it's activated and clickable. The user is just waiting and will of course close the box but he can click ten or more times wo reaction. See shot where the message is shown but no MUI bars.
The ultimate may be to not show the message box until the MUI bar is present. The log message "options: got device count: 1" would be a flag for that if available in a handy way. Or the ruff way is to delay it 5 to 10 seconds.  I know it can be totally hidden by a config setting but most users don't.
So if this message wont disappear in the oesenc/oerns combined plugin it would fit on your "little" list?

Thanks
Håkan

[image]

Some statistics:
Time from messages:
```
BuildFBO 3
14:25:47: OpenGL Framebuffer OK, size = 1261
Until this message that use to be printed the same time as the canvas MUI bar is viewed.
14:25:51: options: got device count: 1

 4   sec    Single canvas. No charts; No connections;  GSHHG: High. Activated plugins: WMM
 4   sec    Muilti canvas. No charts; No connections;  GSHHG: High. Activated plugins: WMM
 5   sec    Muilti canvas. No charts; No connections;  GSHHG: High. Activated plugins: WMM; Grib;
 5   sec    Muilti canvas. No charts; No connections;  GSHHG: High. Activated plugins: WMM; Grib; Dashboard;
 5   sec    Muilti canvas. No charts; No connections;  GSHHG: High. Activated plugins: WMM; Grib; Dashboard; ofc_pi
 5   sec    Muilti canvas. No charts; No connections;  GSHHG: High. Activated plugins: WMM; Grib; Dashboard; ofc_pi; oeSENC_pi;
10  sec    Muilti canvas. No charts; No connections;  GSHHG: High. Activated plugins: WMM; Grib; Dashboard; ofc_pi; oeSENC_pi; oeRNC_pi v1.0.4
10  sec    Muilti canvas. Fugawi charts; No connections;  GSHHG: High. Activated plugins: WMM; Grib; Dashboard; ofc_pi; oeSENC_pi; oeRNC_pi v1.0.4
10  sec    Muilti canvas. Fugawi charts; GPS, AIS UDP;  GSHHG: High. Activated plugins: WMM; Grib; Dashboard; ofc_pi; oeSENC_pi; oeRNC_pi v1.0.4
10  sec    Muilti canvas. Fugawi charts; GPS, AIS UDP;  GSHHG: High. Activated plugins: WMM; Grib; Dashboard; ofc_pi; oeSENC_pi; oeRNC_pi v1.0.4
10  sec    Muilti canvas. Fugawi, oeSENC charts; GPS, AIS UDP;  GSHHG: High. Activated plugins: WMM; Grib; Dashboard; ofc_pi; oeSENC_pi; oeRNC_pi v1.0.4
10  sec    Muilti canvas. Fugawi, oeSENC charts; GPS, AIS UDP;  GSHHG: High. Activated plugins: WMM; Grib; Dashboard; ofc_pi; oeSENC_pi; oeRNC_pi v1.0.4
10  sec    Muilti canvas. Fugawi, oeSENC; oeRNC charts; GPS, AIS UDP;  GSHHG: High. Activated plugins: WMM; Grib; Dashboard; ofc_pi; oeSENC_pi; oeRNC_pi v1.0.4
```

bdbcat, 2019-12-26:
@Hakansv 
Will you try a test with OpenGL OFF, please?

Hakansv, 2019-12-26:
@bdbcat 
With OpenGL off the start up is slightly different. Some seconds faster until MUI bar is present but there is something going on all the time. No delay is perceived.
Also are more or less every new commit making things better and better. Also in OpenGL mode. There are some cleaning coming in that may have effect also here. From yesterday where I measured 10 second it's now repeatable 5 seconds instead for the same condition. As it is now I shouldn't have raised this issue. And most of all it's not any more possible to force the crash/hang while waiting for the MUI bar(s).
Håkan

Hakansv, 2019-12-26:
Using Fugawi and oeSENC charts that are active on start up is the worst case in this matter. But it's not that serious we have to work on ofc_pi. It's enough we have to keep it alive for some more years. 
Thanks for listening.

Hakansv, 2020-01-02:
@bdbcat 
This issue is still obvious so reopened. 
I made some "Runtime debug" by log messages and it boiled down to options.cpp and chart groups.
This function is using 20-60 seconds on my system:
options.cpp:6050
CreatePanel_ChartGroups(m_pageCharts, border_size, group_item_spacing);

To avoid confusions:
I'm on master branch and rebase -i origin says: "nope"

Test method:
Debug info like this:
```
// ChartGroups must be created after ChartsLoad and must be at least third
  wxLogMessage("My Debug----CreateControl_ChartGoups:6050");
  CreatePanel_ChartGroups(m_pageCharts, border_size, group_item_spacing);
  wxLogMessage("My Debug----CreateControl_After_ChartGoups:6052");
  RecalculateSize();

Runtime debug info in options.cpp 
(Row numbers not always correct since I've add my messages in between)
24 hours time stamps.
Two O-starts:
16:03:18: My Debug----CreateControl_ChartGoups:6050
16:04:21: My Debug----RecalculateSize:6052
-----
16:26:51: PlugInManager: Calling LateInit PlugIn: C:\Program Files (x86)\OpenCPN\plugins\ocpn_draw_pi.dll
16:26:51: My Debug----CreateControl_ChartGoups:6050
16:27:12: My Debug----CreateControl_After_ChartGoups:6052
16:27:13: options: got device count: 1
```
From my config:
 ```
 [ChartDirectories]
ChartDir1=C:\\Users\\admin\\Documents\\Charts\\GSHHG^17327676454623251974
ChartDir2=C:\\Users\\admin\\Documents\\Charts\\oeSENC_SE^2135409829428753232
ChartDir3=C:\\Users\\admin\\Documents\\Charts\\oeSENC\\hphascreator-DE-2017-3^14613738141455235365
ChartDir4=C:\\Users\\admin\\Documents\\Charts\\oeSENC_DK^562894166239138365
ChartDir5=C:\\Users\\admin\\Documents\\Charts\\oeRNC_E-Caribb^18289704624954371388
ChartDir6=C:\\Users\\admin\\Documents\\Charts\\oeRNC_Greece^14965931650803710220
ChartDir7=C:\\Users\\admin\\Documents\\Charts\\BSB_SE_2018^5506054963713845111
ChartDir8=C:\\Users\\admin\\Documents\\Charts\\Fugawi_NV_west^14460342988945942174
ChartDir9=C:\\Users\\admin\\Documents\\Charts\\CM93_2015^
[AUI]
AUIPerspective=layout2|name=ChartCanvas;caption=;state=768;dir=5;layer=0;row=0;pos=0;prop=100000;bestw=620;besth=652;minw=258;minh=580;maxw=-1;maxh=-1;floatx=-1;floaty=-1;floatw=-1;floath=-1|name=ChartCanvas2;caption=;state=508;dir=2;layer=0;row=0;pos=0;prop=100000;bestw=5;besth=5;minw=258;minh=580;maxw=-1;maxh=-1;floatx=-1;floaty=-1;floatw=-1;floath=-1|name=DASH_43eeffff-4156-4fe8-b46d-3f5c7bb3ffff;caption=Momo;state=2098125;dir=4;layer=0;row=0;pos=0;prop=100000;bestw=150;besth=572;minw=150;minh=572;maxw=-1;maxh=-1;floatx=423;floaty=27;floatw=166;floath=611|name=DASH_449dffff-640c-438b-a609-9a9b5845ffff;caption=Instrumentpanel;state=2098127;dir=4;layer=0;row=0;pos=0;prop=100000;bestw=150;besth=382;minw=150;minh=382;maxw=-1;maxh=-1;floatx=100;floaty=100;floatw=166;floath=421|name=radar_pi-0;caption=Emulator;state=2099139;dir=2;layer=1;row=0;pos=0;prop=100000;bestw=206;besth=326;minw=256;minh=256;maxw=-1;maxh=-1;floatx=352;floaty=89;floatw=272;floath=326|dock_size(5,0,0)=263|dock_size(2,0,0)=771|
[Groups]
GroupCount=4
[Groups/Group1]
GroupName=Sverige
GroupItemCount=4
[Groups/Group1/Item0]
IncludeItem=C:\\Users\\admin\\Documents\\Charts\\oeSENC_SE
[Groups/Group1/Item1]
IncludeItem=C:\\Users\\admin\\Documents\\Charts\\oeSENC_DK
[Groups/Group1/Item2]
IncludeItem=C:\\Users\\admin\\Documents\\Charts\\oeSENC\\hphascreator-DE-2017-3
[Groups/Group1/Item3]
IncludeItem=C:\\Users\\admin\\Documents\\Charts\\BSB_SE_2018
[Groups/Group2]
GroupName=NV_väst
GroupItemCount=1
[Groups/Group2/Item0]
IncludeItem=C:\\Users\\admin\\Documents\\Charts\\Fugawi_NV_west
[Groups/Group3]
GroupName=test
GroupItemCount=1
[Groups/Group3/Item0]
IncludeItem=C:\\Users\\admin\\Documents\\Charts\\BSB_SE_2018
[Groups/Group4]
GroupName=CM93
GroupItemCount=1
[Groups/Group4/Item0]
IncludeItem=C:\\Users\\admin\\Documents\\Charts\\CM93_2015
```
Picture while waiting:
[image]

bdbcat, 2020-01-04:
@Hakansv 
I cannot reproduce.
Please try to run in VS debugger, without plugins, and break the debugger while stalled.  Should be enough time to switch back to VS GUI, then Debug->Break All.  Then take a look at the stack trace.
Also, please test to see if the delay is correlated with the number of Groups defined.
Finally, put a breakpoint at the beginning of void ChartGroupsUI::PopulateTrees(void), around line 8828.  This is an expensive subroutine, but should not be called until you actually enter options dialog the first time. Maybe getting called earlier than necessary?

Hakansv, 2020-01-04:
@bdbcat 
- ChartGroupsUI::PopulateTrees isn't running while init. Only when options dialog is to be shown as it should.
- CreatePanel_ChartGroups() runs in 1 second while debug. (Also including my normal plugins)
- Running the same package on two other Win10 didn't reproduce the same delay. Less then a second between my debug messages.
- You can't reproduce.
- So something on my dev-Win10 in runtime Release/RelWithDebInfo run.
Now we give this up - no common interest. 
Thanks
Håkan

closed 2020-01-04 by Hakansv

5KnotShitBox, 2022-04-05:
I'm having the same issue. OpenCPN is slow to react to the menu button (same about 10 sec), even on a clean install without charts loaded. 
Used to be very snappy on same hardware in previous versions.
Is there any solution to this?

transmitterdan, 2022-04-06:
> 

What is your OS and version of OpenCPN?

What means "menu button"? Do you mean the toolbar "Options" button?  If so, is it slow every time or just the first time you press it?

5KnotShitBox, 2022-04-06:
Windows 10 Pro 21H2 on a i7-7600 / 32 GB ram / 1 TB SSD 

OpenCPN 5.6.0 is always slow, both at opening and at pressing the 'gears' icon for the options menu.
OpenCPN 5.2.1+6b314e6 is opens fast and reacts instantly to the options menu.

So, for now I'm sticking to the 5.2.1 version until this issue is resolved.