OpenCPN/OpenCPN #1543: Plugin Installer - BSB4 on  non-windows system

tai-fun, 2019-12-20:
Nothing in the Plugin Installer stops me from installing the BSB4 plugin on Linux. The installation actually worked without a hiccup on a Kubuntu 18.04.
This plugin is a bit special as it's Windows only, and should not even show up n the Plugin Manager on a non Windows system.

tai-fun, 2019-12-20:
The NV-charts and Google Earth plugins are Windows only as well.

leamas, 2019-12-20:
> This plugin is a bit special as it's Windows only

Is it? IIRC, I have actually been using it on Linux. And the sources have the usual buildosx/ directory for building macos. 

"Mystified"  What have I missed?

That said, it's not hard to make a plugin not show up, it's just a question of removing the metadata from ocpn-plugins.xml.

tai-fun, 2019-12-20:
It's definitely windows only.
https://github.com/bdbcat/bsb4_pi
https://opencpn.org/OpenCPN/plugins/bsb4.html
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:plugins:charts:bsb4_charts

leamas, 2019-12-20:
Ack.

Now, we need to add another plugin before removing this. It still fulfills some strange role as a testcase... but before any release it must be removed, agreed.

bcn2, 2019-12-20:
I think BSB4 is basicly dead.
The Canadian CHS has  moved to BSB3 and at o-charts we are looking to present the Canadian raster charts as oeRNCs if this is doable with a reasonable effort.
Are there other chart sources left with BSB4?

leamas, 2019-12-20:
In any case: Just understand this as a badly chosen testcase for the installer...

closed 2020-03-13 by tai-fun