OpenCPN/OpenCPN #1544: Left docking problem on dual canvas.

tai-fun, 2019-12-20:
If I drag the Dashbord on the left canvas towards the divider  a blue-ish shaded vertical area i visible.
Letting the mouse go, makes the dashboard dock on the right hand side in the left canvas, filling out the blue-ish area.
Trying a similar thing in the right canvas is problematic. Trying to left-dock the dashboard  in the right canvas the blue-ish area split the right canvas horizontally in the middle, and the actual docking follows the same pattern.

[image]
[image]
[image]

bdbcat, 2020-03-30:
Thomas...
Fixed in github.  Now the vertical dashboard is only dockable at the  far right of either canvas.
There is a small glitch that I cannot fix without major refactoring.  When the mouse is exactly over the divider bar between canvases, it tries to drop the dashboard at the top of the right canvas.  It is a small enough target that  we can ignore?

tai-fun, 2020-03-30:
Dave
Now works as you say, but a bit tricky to right dock in the left canvas.
Needs good documentation to get users used to it.
No problem to ignore this, I only tested as it was a new feature.

closed 2020-03-30 by tai-fun