OpenCPN/OpenCPN #1546: Debugging on linux mint does not show list of plugins

jongough, 2019-12-21:
For debugging purposes there is an issue in download_mgr.cpp, in OcpnScrolledWindow, line 612, it does a check of plugin.target aginst PKG_TARGET. plugin.target(in my case) is "ubuntu" but PKG_TARGET is "linuxmint", "linuxmint" is ubuntu. But due to this case the scrolled window is not populated. I am trying to debug a build stored on cloudsmith of testplugin_pi, but this is stopping the process.

Getting past this by commenting out the test, the plugin gets loaded. This test is too sensitive.

leamas, 2019-12-21:
hm... yes, I have been thinking about this. The problem is obvious, the solution(s) not. Possible approaches:

  - Since the plugin works on ubuntu version xx the metadata could be patched to 
    match this.
  - We add a checkbox like Show All Plugins to the list, disabling the filtering completely
  - We add a list of targets to the XML metadata. This is probably most work.
  - We teach opencpn which targets that are compatible (maintenance problems...)

Thinking about it, perhaps the first option + the last is the best. Basically this is about Ubuntu derivatives. I presume that linuxmint keeps the compatibility with the ubuntu version it's based on, right? In this case the correct metadata should reflect Ubuntu, and we should teach opencpn that although the os is called linuxmint version xx it is actually ubuntu vers yy.

jongough, 2019-12-28:
On Linux mint the file /etc/os-release  'may' help, it has the following content
```
NAME="Linux Mint"
VERSION="19.2 (Tina)"
ID=linuxmint
ID_LIKE=ubuntu
PRETTY_NAME="Linux Mint 19.2"
VERSION_ID="19.2"
HOME_URL="https://www.linuxmint.com/"
SUPPORT_URL="https://forums.ubuntu.com/"
BUG_REPORT_URL="http://linuxmint-troubleshooting-guide.readthedocs.io/en/latest/"
PRIVACY_POLICY_URL="https://www.linuxmint.com/"
VERSION_CODENAME=tina
UBUNTU_CODENAME=bionic
```
I am currently working around this issue by changing the xml file to have linuxmint instead of ubuntu.

bdbcat, 2020-03-18:
Your patch implemented.
Close?

jongough, 2021-04-20:
Yep, but may need more work to give greater coverage