OpenCPN/OpenCPN #1547: CI artifacts review needed

leamas, 2019-12-21:
Currently, we don't build:

  - The latest Ubuntu LTS Bionics.
  - Headers and libraries plugins build and link against.

As for libraries, current review and CI implementation has clarified  that all windows builds (msvc and mingw) need to link to a import library. All platforms requires the ocpn_plugin.h header, and it must be synchronised with the libs. Ergo: we should ship header + import libraries.

We have a limitation of three (travis) or four (circleci) concurrent jobs. Building more than four platforms will take substantially longer time, and probably need a lot of resources to maintain.

Proposal: we drop trusty in favor of bionics

leamas, 2019-12-21:
Dropping trusty means that users will have to update ubuntu to use the latest opencpn. I don't think this breaks the expectations of the linux user group. Even though trusty is supported this basically means security and stability fixes, not more. The expectation is that users should be able to run their existing software, not that they should be able to update to the latest version.

If we support two LTS (bionic + xenial) this is probably better than most applications today. Keeping the very old trusty comes at a price since it makes further development harder.

rgleason, 2019-12-21:
Just a question: Could you use Travis to build and deploy Trusty to Cloudsmith, using circleci for the others?

leamas, 2019-12-22:
Certainly. But the risk is that we end up in an overly complicated setup that we can't maintain with limited resources. In short, that we create a mess.

But the main reason to drop trusty is something else: to keep the source compatibility at such an old level costs in complexity, features we cannot use etc. For example, without testing, I'm still convinced that the necessary gtk3 support isn't available in trusty. There are other issues, software evolves over time, linux even more so.

rgleason, 2019-12-22:
Yes, there is a cost to maintaining older OS, and good arguments to move on, but we don't want to turn the community on its head either. I wonder what % users use trusty.

Hakansv, 2019-12-22:
Opinion:
We proudly submit OCPN as the program for low power and affordable PCs and Androids. It would be possible to use "outdated" machines in the boat and get a low cost well functional nav-system. You don't need a "gaming" device. So for that we are still supporting WinXP, SP3, and up. I know a few still running XP on board. The same would be applicable for the Linux and MacOS world I suppose. Sometimes I use to say to those claiming there Win machine is slow to shift to Linux like Ubuntu 14, 16 or whatever suitable for an old "tractor". (And a nav-PC don't need to be on the Internet and get updates.)
This trusty discussion seems close to destroying this OCPN philosophy or is my limited Linux knowledge misleading me here?
I know if we move to next VS generation, VS2019 and up, we've to drop XP. But I don't know when and for what reason it's time for that.
But, whatever we do we ought to have a common philosophy agreement about it. - Or Dave to take the lead.
Håkan

leamas, 2019-12-25:
> The same would be applicable for the Linux and MacOS world I suppose

Not really. Overall, newer versions doesn't need that much better hardware(it at all). The exception is the window manager, gnome3 is certainly a resource hog. But there are many light-weight window managers out there (fvwm comes to mind); focusing on this gives much more bang for the bucks compared to staying in for example trusty (why not jessie?)

leamas, 2020-02-23:
The discussion about what Ubuntu versions to support is settled (add bionic, keep trusty). 

Remaining bullet is the need to ship import library + header file with next release. However, this is about next release, lets handle it then. Closing

closed 2020-02-23 by leamas