OpenCPN/OpenCPN #1549: Raspberry Pi crashes

transmitterdan, 2019-12-21:
I have been testing master branch on Raspberry Pi. I get consistent crashes from deep inside OpenGL.  Here is some output from dbg for one crash:
```
[New Thread 0x6b56b240 (LWP 1330)]
[New Thread 0x68b36240 (LWP 1331)]
 SENC: nCPU: 2    m_max_jobs :1
[New Thread 0x65fd9240 (LWP 1361)]
Draw call returned Cannot allocate memory.  Expect corruption.
[Thread 0x65fd9240 (LWP 1361) exited]
opencpn: Fatal IO error 0 (Success) on X server :0.0.
[Thread 0x6b56b240 (LWP 1330) exited]
[Thread 0x6f9a6000 (LWP 1293) exited]
[Inferior 1 (process 1293) exited with code 01]
(gdb)
```

Sometimes it crashes X11 and sometimes it just hangs X11 with no way to get out except through reboot.  I have tried the fake KMS and real KMS but both have the same issue.  This never happened in v5.0.x. I suspect this is some kind of thread issue where two threads are trying to use the OpenGL system but it doesn't support multi-threading directly.  I am not smart enough to say for sure that is the issue.  But this is a real problem as far as I can tell.

If someone gives me ideas on what to look for I can try and find it.  I have a good debugging environment set up.

bdbcat, 2019-12-21:
What user operation generally provokes the crash?
Also, what Pi and OS?
Dave

transmitterdan, 2019-12-21:
Operation is random.  Sometimes zooming.  Sometimes closing the options window will cause a crash.

ENC and mbTiles have both crashed so I don't think it is related to type of chart.

I have never had it happen when just letting it run with a VDR loop but haven't been able to do that much.  Will try that today for a few hours.

```
pi@raspberrypi:~ $ pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
VERSION_CODENAME=stretch
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.19.75-v7+ #1 SMP Tue Sep 24 20:03:24 CDT 2019 armv7l GNU/Linux
pi@raspberrypi:~ $ cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : a22082
Serial          : 00000000b8a57e2d
Model           : Raspberry Pi 3 Model B Rev 1.2
pi@raspberrypi:~ $
```

bdbcat, 2019-12-21:
Try NOAA ENC alone. If still crashes, might be related to gL VBO.  This is very memory intensive.
If it still fails, do this to disable temporarily:
glChartCanvas.cpp, around line 1353
add:
    g_b_EnableVBO = false;

Hakansv, 2019-12-21:
Dan..
Just for info:
I've the same RpI 3B+ stretch. It builds and run fine. Is there something we can compare? 
I'm using BSB and oeSENC but so far no "pure" ENC nor MBTiles.
It may I've seen more CPU used for 5.0.522 compared to 5.0. It has increased from 30-50% and up to 75% while zooming. On the other hand I think more CPU are actually used when needed. Before it may be less CPU was used but e.g. zoom took longer time.
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
pi@raspberrypi:~ $ cat /proc/cpuinfo
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a020d3
Serial		: 000000007278c90f
```
[image]

transmitterdan, 2019-12-21:
@bdbcat 

No change with g_b_
EnableVBO = false;

Steps: 
1) Start O with only mbTiles chart and single chart surface
2) Open Options and change User Interface Scale Factor to 5
3) Click Apply
4) Change UISF to 0
5) Click Apply

Got this message in gdb:
Draw call returned Invalid argument.  Expect corruption.

Window manager (lightdm) is no longer responsive. O is still running but cannot display anything.

transmitterdan, 2019-12-21:
Dave,

gdb says the stack is corrupted so can't do a bt.

```
(gdb) bt
#0  0x75a3054c in ?? () from /usr/lib/arm-linux-gnueabihf/libgtk-x11-2.0.so.0
#1  0x75f246d8 in g_type_from_name ()
   from /usr/lib/arm-linux-gnueabihf/libgobject-2.0.so.0
#2  0x75a3079c in ?? () from /usr/lib/arm-linux-gnueabihf/libgtk-x11-2.0.so.0
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb)
```

transmitterdan, 2019-12-21:
@Hakansv 

One difference I can think of is the screen size.  Mine is large.  Here is the relevant section of opencpn.log.

```
14:21:14 CST: Creating glChartCanvas
14:21:14 CST: Metrics:  m_display_size_mm: 1600     g_Platform->getDisplaySize():  1920:1080
14:21:14 CST: wxDisplaySize(): 1920 1080
14:21:14 CST: Loading Tide/Current data source: /usr/local/share/opencpn/tcdata/HARMONIC.IDX
14:21:15 CST: OpenGL-> Renderer String: Gallium 0.4 on VC4 V3D 2.1
14:21:15 CST: OpenGL-> Version reported:  2.1 Mesa 13.0.6
14:21:15 CST: OpenGL-> Texture rectangle format: de1
14:21:15 CST: OpenGL-> Vertexbuffer Objects unavailable
14:21:15 CST: OpenGL Framebuffer OK, size = 1916
14:21:15 CST: OpenGL-> Using Framebuffer Objects
14:21:15 CST: OpenGL-> Using FBO Stencil buffer
14:21:15 CST: OpenGL-> Using Stencil buffer clipping
14:21:15 CST: OpenGL-> Using Scissor Clipping
14:21:15 CST: OpenGL-> Not Using compression
14:21:15 CST: OpenGL-> Minimum cartographic line width:  1.0
14:21:15 CST: OpenGL-> Minimum symbol line width:  1.0
```
It may not be relevant but my ship icon is tiny on the screen and the menu bar is also really small.  I cannot effectively increase them by using the UI scaling slider all the way up to 5. They increase only by a tiny amount.

transmitterdan, 2019-12-21:
@bdbcat 

This might be relevant.  I made the change you suggested.  This is the end of the log file where O exited with result 1 (not 0).

```
16:29:14 CST: BuildFBO 3
16:29:14 CST: OpenGL Framebuffer OK, size = 1746
16:29:14 CST: All chart in cache locked, size: 2
16:29:15 CST: Metrics:  m_display_size_mm: 1600     g_Platform->getDisplaySize():  1920:1080
16:29:15 CST: wxDisplaySize(): 1920 1080
16:29:15 CST: All chart in cache locked, size: 2
16:29:15 CST: All chart in cache locked, size: 2
16:29:15 CST: BuildFBO 3
16:29:15 CST: OpenGL Framebuffer OK, size = 1746
16:29:15 CST: All chart in cache locked, size: 2
16:29:15 CST: BuildFBO 3
16:29:15 CST: OpenGL Framebuffer OK, size = 1746
16:29:15 CST: All chart in cache locked, size: 2
16:29:15 CST: BuildFBO 3
16:29:15 CST: OpenGL Framebuffer OK, size = 1746
16:29:15 CST: All chart in cache locked, size: 2
16:29:15 CST: BuildFBO 3
16:29:15 CST:     OpenGL-> glRenderbufferStorage error:  00000505
16:29:15 CST:     OpenGL-> buildFBOSize->Framebuffer Incomplete:  00008CD6
16:29:15 CST:     OpenGL-> glRenderbufferStorage error:  00000505
16:29:15 CST:     OpenGL-> buildFBOSize->Framebuffer Incomplete:  00008CD6
16:29:15 CST: BuildFBO C
16:29:15 CST: OpenGL-> FBO Framebuffer unavailable
16:29:15 CST: All chart in cache locked, size: 2
16:29:15 CST: All chart in cache locked, size: 2
'''

O did not crash, it just exited with result 1.

bdbcat, 2019-12-21:
@transmitterdan 
On the display size:  Please check options->Advanced->Physical screen width.  Your logfile reports:
Metrics:  m_display_size_mm: 1600.
I doubt your display is nearly 5 ft wide.  Override the value manually.  Or maybe it is...
Dave

Hakansv, 2019-12-21:
@transmitterdan 
"One difference I can think of is the screen size. Mine is large"
Yes could be. Mine only 7". And compare to Dave's note, I'd to manually adjust the size as well.

```
23:26:08 CET: Detected display size (horizontal): 212 mm
23:26:08 CET: Display size (horizontal) config override: 153 mm
23:26:09 CET: Metrics:  m_display_size_mm: 153     g_Platform->getDisplaySize():  800:480   
23:26:09 CET: wxDisplaySize(): 800 480
23:26:10 CET: OpenGL-> Renderer String: Gallium 0.4 on llvmpipe (LLVM 3.9, 128 bits)
23:26:10 CET: OpenGL-> Version reported:  3.0 Mesa 13.0.6
23:26:10 CET: OpenGL-> Texture rectangle format: de1
23:26:10 CET: OpenGL-> Using Vertexbuffer Objects
23:26:10 CET: OpenGL Framebuffer OK, size = 782
23:26:10 CET: OpenGL-> Using Framebuffer Objects
23:26:10 CET: OpenGL-> Using FBO Stencil buffer
23:26:10 CET: OpenGL-> Using Stencil buffer clipping
23:26:10 CET: OpenGL-> Using Scissor Clipping
23:26:10 CET: OpenGL-> Not Using compression
```

transmitterdan, 2019-12-22:
@bdbcat 

The size at 1600mm is correct.  The RPi gets that from the HDMI interface as reported by the display.

bdbcat, 2019-12-22:
@transmitterdan 
A definite corner case, using rPI to drive a 62" width display.
For my information, please use a caliper or steel rule to measure the size of ownship icon, and report. Should be scalable between 7 and 15 mm, independent of overall screen physical size.
As a guess, I would say that the memory faults are due to the size of the OpenGL frame buffers.  I do not have a super hi resolution HDMI display for testing.

transmitterdan, 2019-12-22:
@bdbcat 

I will try to measure the own ship symbol.

The display is 1920X1080 which is a lot of pixels but by no means unusual.  1080 line displays with 16X9 aspect are pretty common.  Isn't that what matters more than the physical size?  Actually, I didn't think size mattered.  Oh wait, that was what I thought in high school...

bdbcat, 2019-12-22:
@transmitterdan 
I think 1920x1080 may be a stretch for rPI.  One frame buffer will be about 8 mB, and several frame buffers are required.  This will definitely stress the glDriver on a limited resource environment with shared memory graphics buffers.
I'll try to simulate something...
And true, the physical size does not (usually) matter.

transmitterdan, 2019-12-22:
@bdbcat 
Understood. Just reminding that it worked ok on v5.0 and all variants.

We have a number of people observing that new releases don't work in situations where old releases did work. Not trying to poke the bear but just pointing out that we will need to be up front if we make a new release that requires better/bigger hardware.

bdbcat, 2019-12-22:
We want to investigate all of these situations.  There really is nothing in the latest commits that should require advanced hardware.  So I think your environment/crash needs investigation and closure.
Particularly, I would like to to verify that the log entries related to OpenGL on your system are identical for the v5.0 and lastest github.  This is where we should start.

transmitterdan, 2019-12-22:
@bdbcat 

Built the master branch against standard Stretch wxWidgets and gtk2.  Was moving around the options window and this happened:

```
Thread 1 "opencpn" received signal SIGSEGV, Segmentation fault.
s52plib::RenderToGLAC (this=this@entry=0x1032938, rzRules=rzRules@entry=0x8fe5fe8, rules=rules@entry=0x25115e8, vp=vp@entry=0x7effc000)
    at /home/pi/opencpn/src/s52plib.cpp:8624
8624                    box = p_tp->tri_box;
(gdb) bt
#0  s52plib::RenderToGLAC (this=this@entry=0x1032938, rzRules=rzRules@entry=0x8fe5fe8, rules=rules@entry=0x25115e8, vp=vp@entry=0x7effc000)
    at /home/pi/opencpn/src/s52plib.cpp:8624
#1  0x0054f3fc in s52plib::RenderAreaToGL (this=0x1032938, glcc=..., rzRules=0x8fe5fe8, vp=vp@entry=0x7effc000) at /home/pi/opencpn/src/s52plib.cpp:9670
#2  0x00552500 in s57chart::DoRenderOnGL (this=this@entry=0x4853b38, glc=..., VPoint=...) at /home/pi/opencpn/src/s57chart.cpp:1697
#3  0x00564514 in s57chart::DoRenderRegionViewOnGL (this=0x4853b38, glc=..., VPoint=..., RectRegion=..., Region=..., b_overlay=<optimized out>,
    b_overlay@entry=false) at /home/pi/opencpn/src/s57chart.cpp:1656
#4  0x005646e8 in s57chart::RenderRegionViewOnGLNoText (this=<optimized out>, glc=..., VPoint=..., RectRegion=..., Region=...)
    at /home/pi/opencpn/src/s57chart.cpp:1514
#5  0x0043dd38 in glChartCanvas::RenderQuiltViewGL (this=0x7effcd14, this@entry=0x0, vp=..., rect_region=...) at /home/pi/opencpn/src/glChartCanvas.cpp:3456
#6  0x004437d8 in glChartCanvas::RenderCharts (this=0x0, this@entry=0x97c5c0, dc=..., rect_region=...) at /home/pi/opencpn/src/glChartCanvas.cpp:3700
#7  0x00444e54 in glChartCanvas::Render (this=this@entry=0x97c5c0) at /home/pi/opencpn/src/glChartCanvas.cpp:4361
#8  0x00445378 in glChartCanvas::OnPaint (this=0x97c5c0, event=...) at /home/pi/opencpn/src/glChartCanvas.cpp:1659
#9  0x762274b0 in wxAppConsoleBase::CallEventHandler(wxEvtHandler*, wxEventFunctor&, wxEvent&) const ()
   from /usr/lib/arm-linux-gnueabihf/libwx_baseu-3.0.so.0
#10 0x7639e97c in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) ()
   from /usr/lib/arm-linux-gnueabihf/libwx_baseu-3.0.so.0
#11 0x7639ead4 in wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*) () from /usr/lib/arm-linux-gnueabihf/libwx_baseu-3.0.so.0
#12 0x7639ef90 in wxEvtHandler::TryHereOnly(wxEvent&) () from /usr/lib/arm-linux-gnueabihf/libwx_baseu-3.0.so.0
#13 0x7639f02c in wxEvtHandler::ProcessEventLocally(wxEvent&) () from /usr/lib/arm-linux-gnueabihf/libwx_baseu-3.0.so.0
#14 0x7639f09c in wxEvtHandler::ProcessEvent(wxEvent&) () from /usr/lib/arm-linux-gnueabihf/libwx_baseu-3.0.so.0
#15 0x7639ed5c in wxEvtHandler::SafelyProcessEvent(wxEvent&) () from /usr/lib/arm-linux-gnueabihf/libwx_baseu-3.0.so.0
#16 0x76c32120 in wxGLCanvas::OnInternalIdle() () from /usr/lib/arm-linux-gnueabihf/libwx_gtk2u_gl-3.0.so.0
#17 0x767acf8c in wxWindowBase::SendIdleEvents(wxIdleEvent&) () from /usr/lib/arm-linux-gnueabihf/libwx_gtk2u_core-3.0.so.0
#18 0x767acfd4 in wxWindowBase::SendIdleEvents(wxIdleEvent&) () from /usr/lib/arm-linux-gnueabihf/libwx_gtk2u_core-3.0.so.0
#19 0x767acfd4 in wxWindowBase::SendIdleEvents(wxIdleEvent&) () from /usr/lib/arm-linux-gnueabihf/libwx_gtk2u_core-3.0.so.0
#20 0x7662a564 in wxFrame::SendIdleEvents(wxIdleEvent&) () from /usr/lib/arm-linux-gnueabihf/libwx_gtk2u_core-3.0.so.0
#21 0x7666f9e4 in wxAppBase::ProcessIdle() () from /usr/lib/arm-linux-gnueabihf/libwx_gtk2u_core-3.0.so.0
#22 0x76583124 in wxApp::DoIdle() () from /usr/lib/arm-linux-gnueabihf/libwx_gtk2u_core-3.0.so.0
#23 0x76583200 in ?? () from /usr/lib/arm-linux-gnueabihf/libwx_gtk2u_core-3.0.so.0
#24 0x75f94fe8 in g_main_context_dispatch () from /lib/arm-linux-gnueabihf/libglib-2.0.so.0
#25 0x75f953b8 in ?? () from /lib/arm-linux-gnueabihf/libglib-2.0.so.0
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb)
```

transmitterdan, 2019-12-22:
And the log file:
```
12:01:28 CST: 2019-12-22
12:01:28 CST:  ------- Starting OpenCPN -------
12:01:28 CST: 5.0.522
12:01:30 CST: wxWidgets version: wxWidgets 3.0.2 Linux 32 bit wxGTK
12:01:30 CST: OS Version reports as:  4.19
12:01:30 CST: MemoryStatus:  mem_total: 925 mb,  mem_initial: 139 mb
12:01:30 CST: SData_Locn is /usr/local/share/opencpn/
12:01:30 CST: PrivateDataDir is /home/pi/.opencpn
12:01:30 CST: Using existing Config_File: /home/pi/.opencpn/opencpn.conf
12:01:30 CST: winPluginDir, read from ini file: 
12:01:30 CST: Setting Ownship Lat/Lon 37.8152, -122.47
12:01:30 CST: Switch InlandEcdis mode Off
12:01:30 CST: Styles loading from /usr/local/share/opencpn/uidata/styles.xml
12:01:30 CST: No styles found at: /home/pi/
12:01:30 CST: No styles found at: /home/pi/.opencpn/
12:01:31 CST: Using SVG Icons
12:01:31 CST: Detected display size (horizontal): 1600 mm
12:01:31 CST: Getting DefaultSystemLocale...
12:01:31 CST: System default Language:  en_US
12:01:31 CST: Config file language:  en_US
12:01:31 CST: Adjusted App language:  en_US
12:01:31 CST: ChangeLocale: Language load for:  en_US
12:01:31 CST: ChangeLocale: Locale Init OK for:  en_US
12:01:31 CST: Loading catalog for opencpn core.
12:01:31 CST: Getting DefaultSystemLocale...
12:01:31 CST: Opencpn language set to:  en_US
12:01:31 CST: Loading Configs catalog: /home/pi/.opencpn/Configs/configs.xml
12:01:31 CST: Creating MyFrame...size(1344, 730)  position(-2, 34)
12:01:31 CST:  **** Connect stuff
12:01:35 CST: Creating glChartCanvas
12:01:35 CST: Metrics:  m_display_size_mm: 1600     g_Platform->getDisplaySize():  1920:1080   
12:01:35 CST: wxDisplaySize(): 1920 1080
12:01:35 CST: Loading Tide/Current data source: /usr/local/share/opencpn/tcdata/HARMONIC.IDX
12:01:36 CST: OpenGL-> Renderer String: Gallium 0.4 on VC4 V3D 2.1
12:01:36 CST: OpenGL-> Version reported:  2.1 Mesa 13.0.6
12:01:36 CST: OpenGL-> Texture rectangle format: de1
12:01:36 CST: OpenGL-> Using Vertexbuffer Objects
12:01:36 CST: OpenGL Framebuffer OK, size = 1340
12:01:36 CST: OpenGL-> Using Framebuffer Objects
12:01:36 CST: OpenGL-> Using FBO Stencil buffer
12:01:36 CST: OpenGL-> Using Stencil buffer clipping
12:01:36 CST: OpenGL-> Using Scissor Clipping
12:01:36 CST: OpenGL-> Not Using compression
12:01:36 CST: OpenGL-> Minimum cartographic line width:  1.0
12:01:36 CST: OpenGL-> Minimum symbol line width:  1.0
12:01:36 CST: Loading World Chart Q=0 in 0 ms.
12:01:36 CST: GSHHS World chart files have wrong version. Found -1, expected 210-220.
12:01:36 CST: Loading World Chart Q=2 in 1 ms.
12:01:36 CST: ChartDB Cache policy:  Application target is 200 MBytes
12:01:36 CST: Loading chart db version: V018
12:01:36 CST: Chartdb: Chart directory list follows
12:01:36 CST:   Chart directory #0: /home/pi/Charts/ENC/US_CA
12:01:36 CST:   Chart directory #1: /home/pi/Charts/RNC/US_CA
12:01:36 CST:   Chart directory #2: /home/pi/Charts/ENC/US_INLAND
12:01:36 CST:   Chart directory #3: /home/pi/Charts/GSHHG
12:01:36 CST:   Chart directory #4: /home/pi/Charts/MBTiles/NOAA
12:01:36 CST: GPS Watchdog Timeout is: 6 sec.
12:01:36 CST: OpenChartUsingCache, NOT in cache:   cache size: 0
12:01:36 CST:    OpenChartUsingCache:  type 5  /home/pi/Charts/ENC/US_CA/US5CA13M/US5CA13M.000
12:01:36 CST: Creating new chart
12:01:37 CST: ChartSymbols loaded from /usr/local/share/opencpn/s57data/chartsymbols.xml
12:01:37 CST: Using s57data in /usr/local/share/opencpn/s57data
12:01:37 CST: Initializing Chart /home/pi/Charts/ENC/US_CA/US5CA13M/US5CA13M.000
12:01:37 CST: S57chart::Checking SENC file: /home/pi/.opencpn/SENC/714872EAEDDE_US5CA13M.S57
12:01:39 CST: OpenCPN Initialized in 11605 ms.
12:01:39 CST: BuildFBO 3
12:01:39 CST: OpenGL Framebuffer OK, size = 1340
12:01:39 CST: DPMM: 1.2   ScaleFactorExp: 1
[... 315 lines cut ...]
12:05:18 CST: All chart in cache locked, size: 13
12:05:28 CST: All chart in cache locked, size: 13
12:05:52 CST: All chart in cache locked, size: 13
12:06:12 CST: All chart in cache locked, size: 13
12:06:31 CST: All chart in cache locked, size: 13
12:06:36 CST: BuildFBO 3
12:06:36 CST: OpenGL Framebuffer OK, size = 1748
12:06:36 CST: All chart in cache locked, size: 13
12:06:36 CST: BuildFBO 3
12:06:36 CST: OpenGL Framebuffer OK, size = 1748
12:06:36 CST: All chart in cache locked, size: 13
12:06:49 CST: All chart in cache locked, size: 13
12:06:57 CST: BuildFBO 3
12:06:57 CST: OpenGL Framebuffer OK, size = 1748
12:06:57 CST: All chart in cache locked, size: 13
12:07:08 CST: All chart in cache locked, size: 13
12:07:21 CST: All chart in cache locked, size: 13
12:07:34 CST: All chart in cache locked, size: 13
12:07:42 CST: Error: File /usr/local/share/opencpn/plugins/dashboard_pi/data/plus.svg doesn't exist.
12:07:42 CST: Error: File /usr/local/share/opencpn/plugins/dashboard_pi/data/minus.svg doesn't exist.
12:07:42 CST: Error: File /usr/local/share/opencpn/plugins/dashboard_pi/data/dial.svg doesn't exist.
12:07:42 CST: Error: File /usr/local/share/opencpn/plugins/dashboard_pi/data/instrument.svg doesn't exist.
12:07:46 CST: All chart in cache locked, size: 13
12:07:59 CST: All chart in cache locked, size: 13
```

transmitterdan, 2019-12-27:
@bdbcat 

I think there is something fundamental wrong between current git master and v5.0.x.  I am confident this did not happen in v5.0.x.

I cannot change the screen size on the RPi.  The VC4 GL driver overlay seems to have no option except to read the HDMI display size.  O doesn't have an option to override what it gets from OpenGL so I cannot reduce the screen size in O either.

What I can do is increase the memory available to the VC4 driver.  So I gave it 512MB of buffer space.  O overran that in very short order.  It happens when closing the options dialog after making changes to the UI scale factor and chart object scale factor.  Something in the change over to the scaling causes O to gobble up all the available frame buffer space and then allocations start to fail. Apparently O cannot handle frame buffer allocation failure.  See session dmesg output below.

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.89-v7+ (pi@raspberrypi) (gcc version 6.3.0 20170516 (Raspbian 6.3.0-18+rpi1+deb9u1)) #3 SMP Thu Dec 26 04:27:31 CST 2019
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 512 MiB at 0x0ec00000
[    0.000000] On node 0 totalpages: 242688
[    0.000000]   Normal zone: 2133 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 242688 pages, LIFO batch:63
[    0.000000] random: get_random_bytes called from start_kernel+0xb0/0x4bc with crng_init=0
[    0.000000] percpu: Embedded 16 pages/cpu s36684 r8192 d20660 u65536
[    0.000000] pcpu-alloc: s36684 r8192 d20660 u65536 alloc=16*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240555
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 cma=256M video=HDMI-A-1:1920x1080@60 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 cma=512M console=ttyS0,115200 console=tty1 root=PARTUUID=885b30a0-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 421976K/970752K available (8192K kernel code, 653K rwdata, 2220K rodata, 1024K init, 821K bss, 24488K reserved, 524288K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xbb800000 - 0xff800000   (1088 MB)
                   lowmem  : 0x80000000 - 0xbb400000   ( 948 MB)
                   modules : 0x7f000000 - 0x80000000   (  16 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (9184 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   (1024 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   ( 654 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 822 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 26736 entries in 79 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000006] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000022] Switching to timer-based delay loop, resolution 52ns
[    0.000281] Console: colour dummy device 80x30
[    0.000888] console [tty1] enabled
[    0.000942] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000984] pid_max: default: 32768 minimum: 301
[    0.001320] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001355] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002307] CPU: Testing write buffer coherency: ok
[    0.002786] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.003451] Setting up static identity map for 0x100000 - 0x10003c
[    0.003619] rcu: Hierarchical SRCU implementation.
[    0.004432] smp: Bringing up secondary CPUs ...
[    0.005289] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.006214] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.007073] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.007193] smp: Brought up 1 node, 4 CPUs
[    0.007272] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.007296] CPU: All CPU(s) started in HYP mode.
[    0.007316] CPU: Virtualization extensions available.
[    0.008291] devtmpfs: initialized
[    0.021226] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.021492] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.021540] futex hash table entries: 1024 (order: 4, 65536 bytes)
[... 253 lines cut ...]
[  475.494391] [drm]                            V3D: 448552kb BOs (479)
[... cut ...]
[  475.554136] [drm]                           dumb:   8148kb BOs (4)
[  475.554140] [drm]                         binner:  16384kb BOs (1)
[  475.555021] [drm:vc4_bo_create [vc4]] *ERROR* Failed to allocate from CMA:
[  475.555033] [drm]                            V3D: 446468kb BOs (444)
[  475.555040] [drm]                     V3D shader:    428kb BOs (104)
[  475.555043] [drm]                           dumb:   8148kb BOs (4)
[  475.555047] [drm]                         binner:  16384kb BOs (1)
```

transmitterdan, 2019-12-27:
@bdbcat 

I failed to mention that building with 
```
    m_b_DisableFBO = true;
```
seems to solve the issue for me.

Hakansv, 2019-12-27:
To compare to my RPi3 I've not wast display failures but the user interface scale factor and Chart object scale factor sliders are having some updating problem. The sliders can be move and the value above the slider is changing according to any moves but the value below is not changed. The effect of the changed factor is still there.
This picture after "Apply" (And this is not about the "truncated" view. It's the 7" RPi screen. It's like that)
[image]

transmitterdan, 2019-12-27:
@Hakansv 
I don't understand.  When you move a slider the value above is associated with that slider and should change.  The value immediately below is for the slider below so it should not change.

Hakansv, 2019-12-27:
@transmitterdan 
Ooops.. To much Window in my genes? It's the opposite on Win where the value for the slider is below. Sorry. >> Think before talk!?
[image]

transmitterdan, 2019-12-28:
@bdbcat 

I suspect there is a GL memory leak in the buidFBO() + buildFBOSize() scheme in glChartCanvas.cpp.  I am far from an OpenGL expert but on my RPi I am seeing >500 fbo's allocated.  I can't imagine why O needs that many fbo's ever. No matter how much memory I give the GPU, O uses up all of it. Surely there is a way to control how many fbo's it uses.

bdbcat, 2019-12-28:
@transmitterdan 
On my list.  Hope to get to it today.

bdbcat, 2019-12-28:
@transmitterdan 
Try this latest commit:  923eecdc0ada154501c8824144eeda6fe28dba6c
The macros defining FBO deletion were not correct, so your conclusion that FBO memory was leaking was spot on.  Thanks for the detailed analysis.

transmitterdan, 2019-12-30:
30+ hours without any problem on this issue.

closed 2019-12-30 by transmitterdan