OpenCPN/OpenCPN #1551: Build OpenCPN 5.0.522+e49c898 has problems finding plugin images

transmitterdan, 2019-12-21:
Some image files are not included in the installation.

```
10:57:54 AM: Error: can't open file 'C:\Program Files (x86)\OpenCPN 5.0.522+e49c898\plugins\chartdldr_pi\data\folder.png' (error 2: The system cannot find the file specified.)
10:57:54 AM: Error: Failed to load image from file "C:\Program Files (x86)\OpenCPN 5.0.522+e49c898\plugins\chartdldr_pi\data\folder.png".
10:57:54 AM: Error: Couldn't add an image to the image list.
10:57:54 AM: Error: can't open file 'C:\Program Files (x86)\OpenCPN 5.0.522+e49c898\plugins\chartdldr_pi\data\file.png' (error 2: The system cannot find the file specified.)
10:57:54 AM: Error: Failed to load image from file "C:\Program Files (x86)\OpenCPN 5.0.522+e49c898\plugins\chartdldr_pi\data\file.png".
10:57:54 AM: Error: Couldn't add an image to the image list.
10:57:54 AM: Error: can't open file 'C:\Program Files (x86)\OpenCPN 5.0.522+e49c898\plugins\chartdldr_pi\data\button_right.png' (error 2: The system cannot find the file specified.)
10:57:54 AM: Error: Failed to load image from file "C:\Program Files (x86)\OpenCPN 5.0.522+e49c898\plugins\chartdldr_pi\data\button_right.png".
10:57:54 AM: Error: Couldn't add an image to the image list.
10:57:54 AM: Error: can't open file 'C:\Program Files (x86)\OpenCPN 5.0.522+e49c898\plugins\chartdldr_pi\data\button_right.png' (error 2: The system cannot find the file specified.)
10:57:54 AM: Error: Failed to load image from file "C:\Program Files (x86)\OpenCPN 5.0.522+e49c898\plugins\chartdldr_pi\data\button_right.png".
10:57:54 AM: Error: Couldn't add an image to the image list.
10:57:54 AM: Error: can't open file 'C:\Program Files (x86)\OpenCPN 5.0.522+e49c898\plugins\chartdldr_pi\data\button_down.png' (error 2: The system cannot find the file specified.)
10:57:54 AM: Error: Failed to load image from file "C:\Program Files (x86)\OpenCPN 5.0.522+e49c898\plugins\chartdldr_pi\data\button_down.png".
10:57:54 AM: Error: Couldn't add an image to the image list.
10:57:54 AM: Error: can't open file 'C:\Program Files (x86)\OpenCPN 5.0.522+e49c898\plugins\chartdldr_pi\data\button_down.png' (error 2: The system cannot find the file specified.)
10:57:54 AM: Error: Failed to load image from file "C:\Program Files (x86)\OpenCPN 5.0.522+e49c898\plugins\chartdldr_pi\data\button_down.png".
10:57:54 AM: Error: Couldn't add an image to the image list.
```

Rasbats, 2019-12-21:
I have been trying to update "otcurrent_pi". Exactly the same problem. No icons shown for panel or toolbar.

When I install the radar plugin from the plugin installer the panel icon shows but the toolbar icons do not.

Squiddio does not use svg so everything works correctly for the plugin installer.

Running Win 8.1/OpenCPN 5.0.522+5ea57e2.

transmitterdan, 2019-12-22:
> I have been trying to update "otcurrent_pi". Exactly the same problem. No icons shown for panel or toolbar.
>
> When I install the radar plugin from the plugin installer the panel icon shows but the [...]

I don't think these issues are related.  The radar plugin is responsible for putting the icons in the right folder.  OpenCPN cannot do anything about a plugin with missing icons.

transmitterdan, 2019-12-22:
So I looked into this.  Somewhere between V5.0.0 and current git master some image files in the folder opencpn/plugins/dashboard/data were deleted.  They are duplicated in the opencpn/plugins/dashboard/src folder but the installer does not know about those.  So now the installer does not copy these image files over.   I will work on a comprehensive PR on this unless someone else is working on it.  Bottom line, I know what is wrong and I can fix it.

transmitterdan, 2019-12-22:
The chartdldr_pi plug has a similar but different issue.  It tries to load several .png bitmap files but these files are nowhere in the source tree.  Thesse files are loaded by the code but not found:

folder.png
file.png
button_right.png
button_down.png

Rasbats, 2019-12-23:
The 'leamas' radar_pi was used as a template for otcurrent_pi, following the 'leamas' pi installer wiki.

The png for the panel icon works correctly:

` wxString shareLocn = GetPluginDataDir("otcurrent_pi");
	  shareLocn = shareLocn + wxFileName::GetPathSeparator() + _T("data") + wxFileName::GetPathSeparator();

	  wxImage panelIcon(shareLocn + _T("otcurrent_panel_icon.png"));
	  if (panelIcon.IsOk())
		  m_panelBitmap = new wxBitmap(panelIcon);
	  else
		  wxLogMessage(_T("    otcurrent panel icon NOT loaded"));`
And the log:

`9:56:44 AM: Checking plugin candidate: C:\Program Files (x86)\OpenCPN 5.0.522+5ea57e2\plugins\otcurrent_pi.dll
9:56:44 AM: PLugin is compatible: true
9:56:44 AM: PlugInManager: Loading PlugIn: C:\Program Files (x86)\OpenCPN 5.0.522+5ea57e2\plugins\otcurrent_pi.dll
9:56:44 AM: PlugInManager: Using data dirs from: C:\Users\Mike\AppData\Local\opencpn;C:\Program Files (x86)\OpenCPN 5.0.522+5ea57e2\plugins
9:56:44 AM: PlugInManager: using data dir: C:\Program Files (x86)\OpenCPN 5.0.522+5ea57e2\plugins\otcurrent_pi
9:56:44 AM: PlugInManager:  C:\Program Files (x86)\OpenCPN 5.0.522+5ea57e2\plugins\otcurrent_pi.dll
              API Version detected: 116
              PlugIn Version detected: 205`

So OpenCPN is finding the data folder and the icons but the SVG icons are not rendering.

leamas, 2019-12-25:
@Rasbats :  do you have otcurrent_pi available on an URL possibly on github? If so, I could try to have a look at it... this is most certainly another issue, even though the symptoms are similar.

leamas, 2019-12-25:
@transmitterdan : when I install the last build  5.0.522+1732.a9628... those icons seems to be in place at \\Program Files (x86)\\OpenCPN\\plugins\\chartdldr_pi\\data. Can you please check?

That said, this does not look right, se #1566

Rasbats, 2019-12-26:
The problem with the icons is that they are not declared as 'extern' in 'icons.h'.

Also I had to use the following code to find the data for the icons:

```
wxFileName fn;

  wxString m_dataLocn =
      GetPluginDataDir("radar_pi")  // with plugin installer folders this only produces "\data\"

      // from the log we see "Using plugin data path: C:\Users\Mike\AppData\Local\opencpn" so I added folders with hard code

      + wxFileName::GetPathSeparator() + _T("plugins") + wxFileName::GetPathSeparator() + _T("radar_pi") +
      wxFileName::GetPathSeparator() + _T("data") + wxFileName::GetPathSeparator();

  fn.SetPath(m_dataLocn);
  fn.SetFullName(_T("radar_standby.svg"));
  svg_normal = fn.GetFullPath();
  fn.SetFullName(_T("radar_searching.svg"));
  svg_rollover = fn.GetFullPath();
  fn.SetFullName(_T("radar_active.svg"));
  svg_toggled = fn.GetFullPath();

  fn.SetFullName("radar_hidden.svg");
  svg_hidden = fn.GetFullPath();

  fn.SetFullName("radar_seen.svg");
  svg_seen = fn.GetFullPath();
  
```
Some other edits were needed but the icons are now showing correctly. I will post the updated code on my Github shortly.

Rasbats, 2019-12-26:
Lost the icons again. Why?

closed 2019-12-27 by transmitterdan