OpenCPN/OpenCPN #1556: Windows 10 uninstall of OCPN leaves Plugin Manager plugins still installed

jongough, 2019-12-22:
Install OCPN 5.0.522+1738.5fd35c3 on Windows 10. Install Squiddio plugin vi the new Plugin Installer. Plugin is installed to c:\Users\userid\AppData\Local\opencpn\plugins. Uninstall OCPN via the uninstall option (right click the OCPN application icon) and it leaves the squiddio.dll still 'installed'. The AppData location is normally hidden from normal users so they would not know where to look to do a hand uninstall.

Plugins installed by the current process, i.e. from their own installation exe file, can be uninstalled by right clicking on the installed application icon or going to the Apps & features section of the settings and left click the application and select uninstall. This allows the user to remove the plugin and all its associated data. The new process should mimic this.

bdbcat, 2020-03-18:
John...
Good point, but judged not blocking.  Leftover dlls that no app ever accesses are not an effective problem, though they are un-tidy.  They are generally small, insignificant at the scale of a full Win10 installation.

jongough, 2020-03-18:
But leaving them around is sloppy none the less.

bdbcat, 2020-03-18:
No argument there.
But overall, to me the advantages of managed plugins as currently implemented outweighs the disadvantages.

jongough, 2020-03-19:
Is there any merit to looking at moving the installed files under ~/opencpn (c:\Users\xxxxx\opencpn)? Then they could easily be found and the directory deleted.

bdbcat, 2020-03-19:
Yes, we should discuss.  I did not follow some earlier discussion about choice of ".local"...
Alec?

leamas, 2021-04-13:
It could make sense to compare the situation with Linux, which has  more mature packaging system and, not least, a comprehensive set of rules.   In short, these rules when applied to this situation boils down to:

  - When a package is removed, all traces of it under the system  directories like /usr and /etc must be removed.
  - A packaging event like  install or uninstall must *not* change anything under $HOME.

As a consequence, anything downloaded to $HOME by a package will be left in place. This includes for example everything under  for example ~/.mozilla, where also browser plugins are stored,  exact analogies to our case.  

We cannot know for sure what a user has downloaded and stored while running OpenCPN, and simply removing it can cause all sorts of harm. It will also break the updating, since this is implemented by first removing OpenCPN and the installing the new version -- it we remove everything downloaded this will break.

To make my point clear: I think think this issue should be closed as NOTABUG for the following reasons:

  - Doing so would break the rather carefully crafted rules of all Linux distros I know, communities which respect in this context.
  - Removing everything downloaded might remove user personal stuff which  cannot be re-created.

What *could* be done is adding some kind of docs and/or info in runtime when unistalling about the left-overs and how to remove them.

leamas, 2021-04-13:
That is, I have basically the same position on this issue as #1539.

jongough, 2021-04-13:
Or provide a 'remove' and 'remove completely' options. But this still leaves stuff around in directories owned by OCPN which are not cleaned up. Most users will not know how to remove this or that it even exists. Just because disks are big now does not alleviate or remove the need to be neat and tidy and clean up after ourselves.

rgleason, 2021-04-13:
How about a "Clear Plugins Cache" command too?

jongough, 2021-04-13:
Clearing the cache does not always seem to be enough when trying to get new versions, without version number changes, installed. PluginManager seems to keep information elsewhere, but am not sure where.

bdbcat, 2021-04-14:
Rick...
Please see my note elsewhere.
Some sort of "Clear Plugins Cache" command has nothing to do with the "issue" surfaced.  Focus....

rgleason, 2021-04-14:
As it is uninstalled executable should ideally prompt with questions such as:
Remove all plugins?  Yes No
Remove plugin cache Yes No
With the Plugin Manager features making it easy to reinstall and update plugins when connected to te internet perhaps the uninstall should just remove all plugins, but that might leave off-line users without their plugins, if they are simply going to reinstall OpenCPN. again (for  some reason).

leamas, 2021-04-20:
Duplicate of #1539

jongough, 2021-04-25:
Yes, but relevant due to different platform.

leamas, 2021-04-26:
This issue has been discussed extensively. The basic conclusion is that while Jon's position is that this is a bug, my is that things work as they should.

This an example on a bug [beaten to death due to honest differences of engineering opinion, but without resolution](https://github.com/OpenCPN/OpenCPN/discussions/2224#discussion-3334592).  According to Dave (same link) such bugs should be closed, with status quo as the resolution (i. e., NOT-A-BUG). 

I concur.

bdbcat, 2021-04-28:
Closing, status quo

closed 2021-04-28 by bdbcat

jongough, 2021-04-28:
I don't, but then I feel for the user and want to improve their experience. But.....