OpenCPN/OpenCPN #1559: Options/Plugin 'Enable' button does not work in certain locations on macOS

jongough, 2019-12-23:
I an running macOS catalina clean install in a VM. I have installed OpenCPN 5.0.522+1.710d513.dmg by double clicking to mount the dmg file then dragging the OpenCPN icon to the Applications link in the displayed panel. I then start OCPN and goto 'Options/Plugins' and click the '+' (plus icon) to use the Plugin Installer. I installed BSB 1.5.0 and Squiddio 1.16.0. I clicked the 'Done' button to take me back to the Plugins window. I then clicked on the BSB4 plugin to get the information drop down and the 'Preferences' (greyed out), 'Uninstall' and 'Enable' buttons. I then tried to left click on the 'Enable' button for Squiddio, but nothing happened. I scrolled the window up a bit and then tried clicking on the 'Enable' button and it worked.

It would appear that the 'Enable' button being on the same line as the '+' icon does not work. When it is moved up a little it does. Further checking shows that in this position none of the three buttons works. The following image shows the location of the buttons relative to the '+' plus button.
[image]

rgleason, 2019-12-23:
Jon, It is great you can test on so many platforms. I wish other macos users would help test and debug too!

bdbcat, 2020-03-18:
Please verify in current code base.

jongough, 2020-03-20:
Is there a current build for macOS that I can use?

bdbcat, 2020-03-20:
Jon...
https://dl.cloudsmith.io/public/david-register/opencpn-unstable/raw/files/OpenCPN_5.0.522+1.4c3c06d.dmg

Built two hours ago

jongough, 2020-03-20:
I have downloaded that file and installed it, however, whenever I try to get to the plugin screen I get the spinning logo and the plugins screen never displays. Here is a screenshot:
[image]
Here is the log from the run:
[log-issue.txt](https://github.com/OpenCPN/OpenCPN/files/4357903/log-issue.txt)

bdbcat, 2020-03-21:
Jon...
You may have an old/bad plugin loading:
/Users/jon/Library/Application Support/OpenCPN/Contents/PlugIns/libsquiddio_pi.dylib

Please delete this, and try again.

bdbcat, 2020-03-21:
Jon...
Red herring.
A recent commit is causing the problem.  I have reverted it.
@tdan... 
712ad2a9546270c455b4fd8a9950f1c0fbe92141 is failing hard on MacOS.  Reverted.

transmitterdan, 2020-03-21:
I'm looking into it.  I think I know what is wrong.

bdbcat, 2020-03-21:
Dan...
There is really no need to do the cast comparison any more. The only children are the panels we need to delete.  At one point there was another child window that we needed to save.  But that was refactored in one of the GUI updates.
Might simplify things....

transmitterdan, 2020-03-21:
Dave,

I'm testing that idea now.  It works on Windows.  Will try Ubuntu and RPi next.  I don't have a Mac to test.

jongough, 2020-03-21:
I was in that situation a little while ago. I now have two virtual Macs, one running under qemu/kvm and the other under virtualbox. They do take a little work to setup, I think it may work better/easier under virtualbox, but for this sort of testing it is the only way. The biggest issue is finding a bootable image, there are pointers but it does take time.

rgleason, 2021-04-23:
@transmitterdan  @jongough  I think this was fixed, can either of you confirm and close?

nohal, 2021-04-23:
This dialog is completely different in the current code.

closed 2021-04-23 by nohal