OpenCPN/OpenCPN #1560: Plugin Manager does not display scroll bars in Ubuntu 16..04

jongough, 2019-12-23:
I installed OpenCPN 5.0.522+5ea57e2 on Ubuntu 16.04 in a VM. I started OCPN then went to "Options/Plugins/'+'" to get to the Plugin Manager panel. I now click on the 'more...' link for oesenc 3.3.0 then I did the same for radar 5.0.4+beta2 and got the following screen with no scrollbars. I tried the same on Windows 10 and macOS catalina and the scrollbars did display. I tried it on Linux Mint 19 and again no scrollbars.
[image]

bdbcat, 2020-03-18:
Please verify in current code base.

closed 2021-04-19 by bdbcat

jongough, 2021-04-20:
The problem does not show up in my current test environment.