OpenCPN/OpenCPN #1562: macOS 10.15.2 crashes with plug-ins; no built-in plug-ins in list to enable/disable

bcn2, 2019-12-23:
With standard plug-ins enabled the crash is at start-up. Used the stable version of OCPN to toggle off the build-in plug-ins and then restarting with the beta.
The dump is almost identical to the one using a config with the immediate crash.
There are three dumps a) standard conf file with plug-ins b) and c) w/o pi

`Process:               OpenCPN [5221]
Path:                  /Applications/OpenCPN 2.app/Contents/MacOS/OpenCPN
Identifier:            org.opencpn
Version:               5.0.522+9d58594 (???)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           OpenCPN [5221]
User ID:               501

Date/Time:             2019-12-22 09:11:45.941 +0100
OS Version:            Mac OS X 10.15.2 (19C57)
Report Version:        12
Anonymous UUID:        3B4E7B11-B4AD-07F5-C446-A58AF64462AC

Sleep/Wake UUID:       6EDEEAAE-7D7C-4B57-BF8D-39451EC3CC88

Time Awake Since Boot: 170000 seconds
Time Since Wake:       1500 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_ARITHMETIC (SIGFPE)
Exception Codes:       EXC_I386_DIV (divide by zero)
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Floating point exception: 8
Termination Reason:    Namespace SIGNAL, Code 0x8
Terminating Process:   exc handler [5221]

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libwx_osx_cocoau_core-3.1.1.0.0.dylib	0x0000000106284291 wxNonOwnedWindowImpl::FindFromWXWindow(NSWindow*) + 9
1   libwx_osx_cocoau_core-3.1.2.0.0.dylib	0x00000001010aef25 -[wxNonOwnedWindowController windowDidResize:] + 37
2   com.apple.CoreFoundation      	0x00007fff2c4efe4a __CFNOTIFICATIONCENTER_IS_CALLING_OUT_TO_AN_OBSERVER__ + 12
3   com.apple.CoreFoundation      	0x00007fff2c4efdde ___CFXRegistrationPost1_block_invoke + 63
4   com.apple.CoreFoundation      	0x00007fff2c4efd53 _CFXRegistrationPost1 + 372
5   com.apple.CoreFoundation      	0x00007fff2c4ef9a9 ___CFXNotificationPost_block_invoke + 97
6   com.apple.CoreFoundation      	0x00007fff2c4bf808 -[_CFXNotificationRegistrar find:object:observer:enumerator:] + 1773
7   com.apple.CoreFoundation      	0x00007fff2c4bebaf _CFXNotificationPost + 1586
8   com.apple.Foundation          	0x00007fff2eb68a02 -[NSNotificationCenter postNotificationName:object:userInfo:] + 59
9   com.apple.AppKit              	0x00007fff2970b185 -[NSWindow _setFrameCommon:display:fromServer:] + 3296
10  libwx_osx_cocoau_core-3.1.2.0.0.dylib	0x00000001010b024a wxNonOwnedWindowCocoaImpl::MoveWindow(int, int, int, int) + 298
11  libwx_osx_cocoau_core-3.1.2.0.0.dylib	0x0000000101004865 wxNonOwnedWindow::DoMoveWindow(int, int, int, int) + 37
12  libwx_osx_cocoau_core-3.1.2.0.0.dylib	0x000000010100b9fb wxWindow::DoSetSize(int, int, int, int, int) + 331
13  org.opencpn                   	0x000000010036819c options::RecalculateSize() + 140
14  org.opencpn                   	0x00000001003675ed options::CreateControls() + 2605
15  org.opencpn                   	0x000000010036620b options::options(MyFrame*, int, wxString const&, wxPoint const&, wxSize const&, long) + 747
16  org.opencpn                   	0x000000010036837d options::options(MyFrame*, int, wxString const&, wxPoint const&, wxSize const&, long) + 13
17  org.opencpn                   	0x00000001001c2542 MyFrame::OnInitTimer(wxTimerEvent&) + 322
18  libwx_baseu-3.1.2.0.0.dylib   	0x000000010175967f wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*) + 223
19  libwx_baseu-3.1.2.0.0.dylib   	0x000000010175a3ca wxEvtHandler::ProcessEventLocally(wxEvent&) + 90
20  libwx_baseu-3.1.2.0.0.dylib   	0x000000010175a2b0 wxEvtHandler::ProcessEvent(wxEvent&) + 96
21  libwx_baseu-3.1.2.0.0.dylib   	0x000000010175a74c wxEvtHandler::SafelyProcessEvent(wxEvent&) + 12
22  libwx_baseu-3.1.2.0.0.dylib   	0x0000000101705dba wxTimerImpl::SendEvent() + 90
23  com.apple.CoreFoundation      	0x00007fff2c515456 __CFRUNLOOP_IS_CALLING_OUT_TO_A_TIMER_CALLBACK_FUNCTION__ + 20
24  com.apple.CoreFoundation      	0x00007fff2c515010 __CFRunLoopDoTimer + 872
25  com.apple.CoreFoundation      	0x00007fff2c514ae3 __CFRunLoopDoTimers + 317
26  com.apple.CoreFoundation      	0x00007fff2c4f9660 __CFRunLoopRun + 2227
27  com.apple.CoreFoundation      	0x00007fff2c4f8738 CFRunLoopRunSpecific + 503
28  com.apple.HIToolbox           	0x00007fff2b03f65d RunCurrentEventLoopInMode + 292
29  com.apple.HIToolbox           	0x00007fff2b03f2a9 ReceiveNextEventCommon + 356
30  com.apple.HIToolbox           	0x00007fff2b03f127 _BlockUntilNextEventMatchingListInModeWithFilter + 64
31  com.apple.AppKit              	0x00007fff296b0eb4 _DPSNextEvent + 990
32  com.apple.AppKit              	0x00007fff296af690 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1352
33  com.apple.AppKit              	0x00007fff296a13ae -[NSApplication run] + 658
[... 247 lines cut ...]
145 com.apple.HIToolbox           	0x00007fff2b03f65d RunCurrentEventLoopInMode + 292
146 com.apple.HIToolbox           	0x00007fff2b03f2a9 ReceiveNextEventCommon + 356
147 com.apple.HIToolbox           	0x00007fff2b03f127 _BlockUntilNextEventMatchingListInModeWithFilter + 64
148 com.apple.AppKit              	0x00007fff296b0eb4 _DPSNextEvent + 990
149 com.apple.AppKit              	0x00007fff296af690 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1352
150 com.apple.AppKit              	0x00007fff296a13ae -[NSApplication run] + 658
151 libwx_osx_cocoau_core-3.1.2.0.0.dylib	0x000000010a7b669e wxGUIEventLoop::OSXDoRun() + 174
152 libwx_baseu-3.1.2.0.0.dylib   	0x000000010ae52e51 wxCFEventLoop::DoRun() + 49
153 libwx_baseu-3.1.2.0.0.dylib   	0x000000010adb7f75 wxEventLoopBase::Run() + 85
154 libwx_baseu-3.1.2.0.0.dylib   	0x000000010ad90a74 wxAppConsoleBase::MainLoop() + 116
155 libwx_osx_cocoau_core-3.1.2.0.0.dylib	0x000000010a75570a wxApp::OnRun() + 26
156 libwx_baseu-3.1.2.0.0.dylib   	0x000000010ade6bf8 wxEntry(int&, wchar_t**) + 56
157 org.opencpn                   	0x00000001098be7a4 main + 20
158 libdyld.dylib                 	0x00007fff63c067fd start + 1

`
===================================================================
Without plug-ins enabled the crash happens after checking for plug-in updates on "ok"

[image]

Second issue:
no standard, built-in plug-in listed. So how to enable or disable those (but editing the conf)?

[image]

bcn2, 2019-12-23:
Toggling the built-in PIs one by one, the only one which "survives" is ChartDL. The other three WMM, Dashb and GRIB are crashing the the program at start.

Switching PIs on/off is done with the stable OCPN.

bdbcat, 2019-12-24:
Hubert....
This may be Catalina problem, or other...
Let us try installing an older version of OCPN (before SignalK integration), and see if that changes anything.
Please try this:  
https://dl.cloudsmith.io/public/david-register/opencpn-unstable/raw/files/OpenCPN_5.0.522+1.6a6c4ab.dmg

bcn2, 2019-12-24:
Dave...
makes no difference. Same crashes.

CarCode, 2019-12-24:
The first 2 lines of the crash log shows different versions of libwx_osx_cocoa_core-...dylib.
Version 3.1.1 and 3.1.2
Why?

> Thread 0 Crashed:: Dispatch queue: com.apple.main-thread

> 0 libwx_osx_cocoau_core-3.1.1.0.0.dylib 0x000000011c30d291 wxNonOwnedWindowImpl::FindFromWXWindow(NSWindow*) + 9

> 1 libwx_osx_cocoau_core-3.1.2.0.0.dylib 0x000000010a7c105b -[wxNonOwnedWindowController windowDidResignKey:] + 43

Is this still the own patched wx version as done for Mac-OCPN 5.0.1 and not the official public wx version?

bdbcat, 2019-12-26:
@bcn2 
I would like to see the logfiles, at least the section relating to attempted loading of plugins.  There seems to be some missing library problems, and the built-in plugins do not actually load.  That is why they do not appear in the selection list.
Dave

bdbcat, 2019-12-26:
@bcn2 
Is it possible that you have another version of wxWidgets installed?
The OCPN bundle includes wx3.1.2. internally.  Linkage to these bundled libraries happens by softlink called e.g. libwx.....-3.1.dylib.  This could possibly collide with a pre-existing installation of wx3.1.1.

You can try to find an offending library set from the CLI:
$locate libwx
Pre-installed libraries could be anywhere, but probably something like /usr/lib/....
Best if you find nothing, since OCPN brings it's own libraries in the bundle.

BTW, I succeeded in booting a Catalina Virtual machine, and the cloudsmith OCPN images startup without fault.

bdbcat, 2019-12-26:
@bcn2 
You may check the OCPN linkage requirements by:
$otool -L /Applications/Opencpn.app/Contents/MacOS/OpenCPN

bcn2, 2019-12-26:
Made a clean-up, deleting all the former parallel versions and keeping just the actual stable 5.0.1 and  a fresh downloaded version from Cloudsmith.
Running now without the issues formerly reported.
Standard plug-ins visible and not conflicting. S-63, oeSENC and oeRNC up and running.

One more difference when installing: renamed the stable "OpenCPN" before installing the beta.
If you don't do that, macOS will ask if install in parallel and rename the new version to "OpenCPN 2" or "OpenCPN 3" if more than one instance exists.

Will close this issue.

closed 2019-12-26 by bcn2