OpenCPN/OpenCPN #1564: Plugin Manager use after free

taruti, 2019-12-24:
In `pluginmanager.cpp`:

```
                PluginMetadata* found = 0;
                for (auto p: PluginHandler::getInstance()->getInstalled()) {
                    if (p.name == pic->m_common_name.Lower()) {
                        found = &p;
                        break;
                    }
                }
                pic->m_version_str =
                    found && !found->readonly ? found->version : "";
```

getInstalled returns the vector, which will not live past the for-loop. Thus found references a temporary stack variable which is out of scope.

@leamas you might be most familiar with this code?

taruti, 2019-12-24:
Turning on `ENABLE_SANITIZER=address` in cmake helps for spotting stuff like this :)

closed 2019-12-24 by bdbcat

leamas, 2019-12-25:
Sorry, not that mush bandwidth during Christmas:  family, grandchildren, and pagan amounts of food...

Thanks for catching this silly mistake of mine, and for bringing ENABLE_SANITIZER. Merry Xmas!

taruti, 2019-12-25:
Happy Holidays :) Concentrating on family sounds like an excellent idea!

leamas, 2019-12-29:
My own patch which didn't make it, for the record:
```

@ -646,8 +646,9 @@ bool PlugInManager::LoadPlugInDirectory(const wxString& plugin_dir, bool load_en
 #endif
                     pic->m_bInitState = true;
                 }
-                PluginMetadata* found = 0;
-                for (auto p: PluginHandler::getInstance()->getInstalled()) {
+                const PluginMetadata* found = 0;
+                auto installed = PluginHandler::getInstance()->getInstalled();
+                for (auto& p: installed) {
                     if (p.name == pic->m_common_name.Lower()) {
                         found = &p;
                         break;

```

PR #1565 (2019-12-24): Plugin Manager: Avoid referencing dead stack variable
Fixes #1564