OpenCPN/OpenCPN #1566: PluginPaths::InitWindowsPaths

taruti, 2019-12-24:
I think this one has bugs in it. @leamas if you have bandwidth you might know the best?
```
void PluginPaths::initWindowsPaths()
{
    using namespace std;

    const string platform_dir = g_Platform->GetPluginDir().ToStdString();
    const string winPluginBaseDir =
        g_Platform->GetWinPluginBaseDir().ToStdString();
    m_userLibdir = winPluginBaseDir + "\\plugins";
    m_userBindir = winPluginBaseDir + "\\plugins";
    m_userDatadir = winPluginBaseDir;
    m_unknownPathDir = winPluginBaseDir + "\\unknown-prefix";

    m_libdirs.push_back(m_userLibdir);
    m_libdirs.push_back(g_Platform->GetPluginDir().ToStdString());
    m_bindirs = m_libdirs;

    m_datadirs.push_back(platform_dir + "\\plugins");
    m_datadirs.push_back(winPluginBaseDir + "\\plugins");
}

```

Why is `m_userDatadir` missing the `\\plugins` that is added elsewhere? Especially since m_datadirs has the version with `\\plugins` added. Should it read instead:
```
    m_userDatadir = winPluginBaseDir + "\\plugins";
```

leamas, 2019-12-25:
It should be OK. For historical reasone there is no \\\\plugins suffix in this case, See https://github.com/leamas/OpenCPN/wiki/Installation-paths

leamas, 2019-12-25:
Hm... or is that assumption actually wrong? See #1551. Need to make more checks

leamas, 2019-12-25:
Thanks for a really good catch (again!). The windows plugin installation path is wrong. This is not just about these lines of code, it's also about docs and the test plugins. This will take some time...

PR #1581 (2019-12-26): PluginPaths.cpp: Fix broken windows data path (#1566).
Closes: #1566

Untested, the chocolatey servers are down. However, it's already broken so it shouldn't get worse. Looks good to me, though.
commit 4ebf93bb (2019-12-26): PluginPaths.cpp: Fix broken windows data path (#1566).

Closes: #1566
closed 2019-12-26 by bdbcat