OpenCPN/OpenCPN #1567: Plugin install paths, non-regular file archive members, problems? UserUnknownPrefixDir purpose?

taruti, 2019-12-24:
This might be two in one.

First of do any *correct* plugins install stuff in `UserUnknownPrefixDir`? If not why does it exists? Should the files instead be dropped and warnings be logged?

As far as I understand everything in plugins should be under `/bin`, `/lib` (or `/lib64` they are the same), and `/share`? And Windows has different rules for .exe and .dll?

The second question is that can plugin tarballs sneak files in? Perhaps using `..` or special Windows paths ``\\?\`` or somesuch? What happens if I had a special file inside the tarball? Various AE_IFREG checks would fail etc, and the path would lack the install prefix?

rgleason, 2019-12-24:
I think this should have title beginning with  "PI Installer" or "Plugin Manager"  thanks.

leamas, 2019-12-25:
>First of do any correct plugins install stuff in UserUnknownPrefixDir?

No

> The second question is that can plugin tarballs sneak files in [...]

To be honest, I havn't thought much about this. The basic assumption is that we only install plugins  which are trusted.  If we install evil plugins we lose -- this is true for most application plugins I'm aware of besides browsers.

taruti, 2019-12-25:
Would you be ok with me doing a PR removing UserUnknownPrefixDir and just warn on invalid paths in tarballs?

Additionally I'll do some checks on the paths in the PR.

leamas, 2019-12-25:
The reasons to drop the files are obvious. The reason to keep them is basically a debugging issue: to see what has been dropped.

Perhaps dropping them if NDEBUG is set?  We don't like #ifdefs, but NDEBUG is used in a myriad of places anyway.

leamas, 2019-12-25:
> Additionally I'll do some checks on the paths in the PR.

That sounds actually great.

rgleason, 2021-04-14:
What happened?

leamas, 2021-04-23:
IMHO no. This is a bug which should be dealt with

commit ae85b261 (2021-05-03): Handle unknown destination paths in plugin tarballs.  Closes #1567
closed 2021-05-03 by bdbcat