OpenCPN/OpenCPN #1569: Plugin Manger - how to install plugins without internet access

jongough, 2019-12-24:
I am trying to use the Plugin Manager to install a plugin but I don't have internet access. I cannot find a way to pick up files that have been copied to the pc. I can download the files from Cloudsmith, but cannot use them in Plugin Manager.

tai-fun, 2019-12-24:
I think this is a very good point. We can't require users to have Internet access to install a plugin.
I should be possible for users to install a plugin from, for example, a USB drive.
Is the idea to keep the old way of installing plugins parallel with the new system?

jongough, 2019-12-24:
I am not really sure. I would hope that both methods are retained, but then 'interesting' things happen when you use both methods. The new method masks the old method of installation so you have to uninstall the plugin using the new method before the old method plugin will show up. There is currently no way to pick which one to use.

rgleason, 2019-12-24:
This might be a problem. Can the PI Installer be augmented to be able to install using the metadata file, but utilizing a local pathname  to the gz file?

leamas, 2019-12-25:
As it is now installing without internet access is not supported.

That said, this touches another thing  I have been thinking about: a tool which could install a tarball directly on he command line without the clumsy detour upload to a server - update metadata -  download and install.

It's IMHO well worth considering. The first step would be to write a new user story, see https://github.com/leamas/OpenCPN/wiki/End-user-stories

rgleason, 2019-12-25:
Such a tool, if flexible enough, might also solve the problem of reverting back and installing previous plugin rather than the most current.

leamas, 2019-12-25:
There is *no* limitation that you can only install the latest plugin, none. You can install any version listed on the catalog.

jongough, 2019-12-25:
The mechanics of the installation process do not limit the number of plugins as long as they have unique names, however, the current user interface does not appear to support, in a user friendly way, the use of multiple versions of the same plugin. This is difficult to test in the absence of multiple versions of a plugin.

jongough, 2019-12-25:
Have looked at https://github.com/leamas/OpenCPN/wiki/End-user-stories and it is too simplistic with too many assumptions made. It needs to be built from a 'user' perspective and address ease of use of the whole concept. Do you want other git hub users to update this wiki?

leamas, 2019-12-25:
Proposals are always welcome.

leamas, 2019-12-25:
> the current user interface does not appear to support, in a user friendly way, the use of multiple versions of the same plugin

It does, it's just that we haven't published more than one version of each plugin. A new metadata file => a new entry. It could be another plugin or a another version of the same plugin. The installer doesn't really care.

bdbcat, 2021-04-13:
"Import plugin" function is implemented in current code.
Please close.

closed 2021-04-13 by jongough