OpenCPN/OpenCPN #1570: OCPN crash after trying Plugin Manger reinstall without internet access

jongough, 2019-12-24:
On Ubuntu 16.04 I installed oesenc 3.3.0 using the Plugin Manager. I lost internet connectivity. I then tried to reinstall oesenc 3.3.0 using the 'Reinstall' button. I got a popup box saying 'Installation error':
[image]
I then click 'OK' on the popup and 'Done' in the Plugin Manger.  If I click on oesenc 3.3.0 in the list of plugins OCPN crashes. Attached is the log of the session that crashed.
[opencpn.log](https://github.com/OpenCPN/OpenCPN/files/3999012/opencpn.log)

rgleason, 2019-12-24:
I think it would be possible to modify the PI Installer to allow a user to download the source files (both tar or gz and metadata files to a local drive) and be able to install from the local drive.
The only thing that would need to change is to change metadata location,  the url 

 <tarball-url> **_https://dl.cloudsmith.io/public/alec-leamas/opencpn-plugins-stable/raw/names/bsb4-darwin-10.13.6-tarball/versions/v1.5.0.3/_**bsb4_pi-1.5.0.3_darwin-10.13.6.tar.gz </tarball-url>

to a pathname ending with bsb4_pi-1.5.0.3_darwin-10.13.6.tar.gz

If a simple way of doing this can be worked out, it should be done..

to have it install.

jongough, 2019-12-24:
I have tried this on Windows 10 and it crashes as well.

rgleason, 2019-12-24:
I think this should have a title beginning with "Plugin Manager" or "PI Installer"

jongough, 2019-12-24:
Done.

leamas, 2019-12-26:
The core issue here is that there  is no code to clean up partial downloads created after network failures. We will need to look into  that . 

Patches welcome.

commit 4268a9bc (2019-12-30): download: Enhanced error handling, part 2 (#1570, #1598).

Add cleanup code for failed installation attempts.

Closes: #1570

Leaving #1598 open until these things settles.
closed 2019-12-30 by bdbcat