OpenCPN/OpenCPN #1571: Plugin Manager - how to pick an older version?

jongough, 2019-12-24:
Using Plugin Manager to install plugins it is only possible to pick the latest version of a plugin from 'experimental', 'master', 'pr' and 'release'. If a version of a plugin does not work in the user environment there is no way to get back to a version that used to work. There should be a way of choosing which version of a plugin to install if the 'current' version does not work.

rgleason, 2019-12-24:
Agreed.

rgleason, 2019-12-24:
Perhaps being able to "revert" back to the manually installed version is impractical, and in that case, the user would have to download and install the previous version manually as before.

jongough, 2019-12-24:
That will only work if there is a manually installable version available and they uninstall the Plugin Installer version first or it will get masked.

leamas, 2019-12-25:
> Using Plugin Manager to install plugins it is only possible to pick the latest version of a plugin from 'experimental', 'master', 'pr' and 'release'

This is a misunderstanding. *experimental*, *master*, and *release* refers to the catalog. Each catalog can have any number of plugins with any number of versions available.

rgleason, 2019-12-25:
Nevertheless users should be able to go back to the previous version of the  PI without too much trouble. There is no easy way to do that.

leamas, 2019-12-25:
This is only about publishing more versions in the catalog. There is no limitation in the installer as such. As long as nothing is removed from the catalog, you can always go back to what you have installed before.

rgleason, 2019-12-25:
So if we publish the most current pi and the previous one, users will be able to roll back to a previous pi installer plugin quite easily.
Another way to look at it would be to have a rollback catalogue sinilar to experimental, but perhaps that is too much to maintain?

leamas, 2019-12-25:
>So if we publish the most current pi and the previous one, users will be able to roll back to a previous pi installer plugin quite easily.

Yes

>Another way to look at it would be to have a rollback catalogue similar to experimental, but perhaps that is too much to maintain?

I don't see the user value in that. What's the point having them in a separate catalog? 

Still:  the channel mechanism could be to achieve that if required.  A channel is just a name for a different catalog, updated separately. So adding  a new channel called "rollback" would achieve exactly this.

rgleason, 2019-12-25:
Thanks.  Should have written "Catalog" .  I have no strong opinion re the best way to handle going back to a previous version. The older rollback version could be included in opcn_plugins.xml catalog or in an separate rollback_plugins.xml catalog.  Alec points out there is flexibility here, and we will have to see what is the best way.

jongough, 2019-12-25:
Just because the mechanics of the plugin installation process does not restrict the number of plugins in the list as long as they have unique names does not mean that the list presented to the user has previous versions in it. In fact, if the list, as it is today, were to have old versions it would quickly become impossible to use. Think 40 plugins with 4 versions each, the user would now need to scroll through all the current and historic releases to find the one they wanted. The user interface would not be user friendly. There needs to be some way of picking not only the plugin the user wants but a simple way to pick which version they want. I would suspect that this will require some modification to the windows displayed.

leamas, 2019-12-25:
Right, as I said earlier, there will a need for tooling (sort, search etc...)  to support this.

rgleason, 2019-12-25:
Jon, Isn't the "view" restricted by the OS being used, automatically? I thought it was, if so the problem is about 1/4 the length, but still has some potential for user confusion.

jongough, 2019-12-25:
I have assumed that only valid plugins for the platform being used are shown, so if historic versions of the plugins are included in the list, I assumed 4, there would be, potentially, 160 plugins in the list. If each build of a plugin were to be added to the list the list could get very long. Hence the need for a user friendly method of picking a plugin and the version.

It also has a follow on issue which is the continual maintenance of the xml file which would need to be automated rather than human driven. Which ever solution for the xml file maintenance is chosen there are issues that would need to be addressed.

rgleason, 2019-12-25:
In reference to your point, maybe we need an ocpn_plugins.xml view selection function?
by Plugin Name
by Most recent version
by Second most recent version
by Third most recent version
by Fourth most recent version

leamas, 2019-12-25:
Already today, only plugins for actual platform is shown. The amount  of plugins *will* be a problem, but it will take a while.

closed 2021-04-26 by jongough