OpenCPN/OpenCPN #1574: Plugin Manager/Installer - display of plugins does not update when 'Update plugin catalog' is selected

jongough, 2019-12-24:
Test done on Windows 10. Disconnect network (or network not available), start Plugin Manger, click 'Update plugin catalog' button. Catalog Manger shown with failed messages. Cloick 'OK', the Plugin Manager has not changed. Click 'Done' to close window then click the '+' icon to open the Plugin Manager, a blank window is displayed.

Reconnect network, open Plugin Manager, click on 'Update plugin catalog', the Catalog Manager shows with OK message. Click 'OK', Plugin Manager windows does not show updates. Click 'Done' and reopen Plugin Manager and plugins show up.

leamas, 2019-12-26:
Yes, known issue: the list of plugins is not updated after catalog change. At a glance, this looked technically complicated in current code, so it's postponed.

bdbcat, 2021-04-19:
Corrected in current 5.5.0 code base.
Re-open if still a problem.

closed 2021-04-19 by bdbcat