OpenCPN/OpenCPN #1575: Plugin Manager - Catalog Manger window does not display correctly

jongough, 2019-12-24:
On both Windows 10 and Ubuntu 16.04 the Catalog Manager window does not size correctly. The buttons for 'OK' and 'Cancel' are not shown, the window has to be resized.
[image]
If errors are displayed the window has to be resized in two directions
[image]

rgleason, 2019-12-25:
I did not know that there were "Cancel" and "Ok" buttons in the Catalogue Manager menu. That is why I found the menu system so puzzling from the beginning!  
Will now try to find these commands, but I believe I found this window frozen and not resizeable.

bdbcat, 2021-04-19:
Closing.
Re-open if still relevant on OCPN 5.5.0 master.