OpenCPN/OpenCPN #1576: Plugin Manager - Catalog Manger does not consistently display the 'Custom URL'

jongough, 2019-12-24:
Under Ubuntu 16.04 click 'Update plugin catalog' to display the Catalog Manager:
[image]
Notice the Custom Url going off the right side of the window and no buttons.
Resize window to get buttons and the Custom URL becomes just `https://`
[image]

leamas, 2019-12-26:
hm... When I try this, it is just a question of pulling wide enough. Have you tried making it wide enough to accommodate the first part *raw.githubusercontent.com* ? For me, it looks fine then.

jongough, 2019-12-26:
There should be no need to manually adjust the size of the dialog, it should happen automatically. If needed the text could wrap. The fact that the buttons do not even show (top image) is even more of an issue. The dialog should always show the buttons at the bottom even if a scrollbar is needed for the content part of the dialog.

rgleason, 2020-01-05:
Alec, I agree with Jon about this. Lot's of time is spent dragging this menu out.

"There should be no need to manually adjust the size of the dialog, it should happen automatically. If needed the text could wrap. The fact that the buttons do not even show "

Also, since this is just a useful informational menu, why couldn't it just be closed by the upper right "X" and not have any [ "Ok" or "Cancel"] which should really be a "Close" button?

Additionally the phantom Catalog manager should be eliminated.

Also I've made a separate Issue, but when I select Channel "Alpha"  and Ultra Advanced and enter a URL: https://githubusercontent.com/rgleason/plugins/rg-alpha/ocpn-plugins.xml and "SAVE" it.
Then select "Update plugin Catalog"   (Which is just the same as  Catalog Manager, isn't it?)  Why not call it that?   -- I get this!

[image]

Channel: Experimental   <---Shouldn't this be Alpha?
URL: https://githubusercontent.com/rgleason/plugins/rg-alpha/ocpn-plugins.xml

jongough, 2021-04-26:
Dialog box has been removed so no longer relevant.

closed 2021-04-26 by jongough