OpenCPN/OpenCPN #1577: Plugin Manager - Catalog Manger inconsistent display between windows and ubuntu

jongough, 2019-12-24:
On Windows display the Catalog Manger window. When if finishes updating his looks like this:
[image]
On Ubuntu it looks like this:
[image]
The Ubuntu version shows the 'Custom URL' but Windows does not.

leamas, 2019-12-26:
hm... have you set the custom url in Windows? It is only shown if set (try to clear it on Ubuntu, it should lead to the same display).

jongough, 2019-12-31:
I think I must have opened the advanced catalog update dialog and selected default at some point on ubuntu. I also probably opened the Catalog Manger went to 'Advanced>>>' clicked clear and then done but did not click save, so the clear had no effect.

closed 2019-12-31 by jongough