OpenCPN/OpenCPN #1583: Plugin Manager - No cache of downloaded packages

jongough, 2019-12-25:
If a plugin is selected for install it will be downloaded from the internet. If this package is then uninstalled it will be removed. If the user then selects install for the package again it is again downloaded from the internet even if the version has not changed. There is no apparent cache of downloaded files that could be referenced for installing the plugin. This has an impact on users with slow or metered downloads.

rgleason, 2019-12-25:
Great point. 

This would be a very nice feature we should plan to implement somehow. There are times when you have too many plugins and have to remove and simplify, until the time you need the plugin again. 

This PI Installer promises to make installation and uninstallation easier, a local cache of metadata and plugins could be very useful to those without internet!

leamas, 2019-12-26:
Indeed. Patches welcome!  

The metadata is already cached, so it's only about the downloaded  tarballs.

jongough, 2019-12-26:
Nope, you have taught me not to try and help with coding. This is down to whoever developed/created/programmed these changes.

leamas, 2019-12-26:
Well... there are  other members of this project, some of which already making patches  to the installer code. 

So: "patches welcome"  :)

jongough, 2019-12-26:
Yep, but you have shown me that you don't care about my fixes so I am not going to go to the trouble of making patches you are going to ignore or overwrite. As I say, you trained me well when trying to get Squiddio going.

bdbcat, 2020-03-18:
Please check current code.  Caching implemented.

jongough, 2020-03-18:
Caching works, I just need to find a newer package for one of the plugins that is currently installed to be able to check that the cache process does check correctly for newer versions. I will try doing this with testplugin_pi when I get the deployment to cloudsmith working for armh systems.

rgleason, 2021-04-23:
@jongough   Have you been able to check testplugin with a newer package?  I believe in Windows it works.  Then you can close this and  we will be down to 127 Issues.  Thanks

jongough, 2021-04-26:
Caching is being used, but from a user perspective there is no way to flush the cache without going into the file system. There should be a capability to 'refresh' the cache such that a plugin without a new version number can be re-downloaded into the cache.

leamas, 2021-04-26:
> , but from a user perspective there is no way to flush the cache without going into the file system.

Correct, but another issue. The original issue filed in this bug is obviously  fixed, we all know that downloaded plugins are cached since long.

jongough, 2021-04-26:
No, it's a half implemented solution. There is caching but no cache management. Similar to Plugin Manager general implementation. Spread stuff around but don't clean up. Probably in the too hard basket. Shame!

bdbcat, 2021-04-29:
Closing, partially implimented, remainder is duplicated by #1829

closed 2021-04-29 by bdbcat